espressif/esp-idf #7877: no AP appears while esp_wifi_start() return ESP_OK

Dandjinh, 2021-11-11:
Device: ESP32-S2
ESP-IDF: release/v4.4

start AP as below

```c
	g_wifi_status = OB_WIFI_STATUS_AP;
	esp_wifi_disconnect();
	esp_wifi_stop();
	
    esp_wifi_get_mac(ESP_IF_WIFI_STA, ap_mac_addr);

    sprintf(tmp, "My_WiFi%02X%02X", ap_mac_addr[4], ap_mac_addr[5]);
    memcpy(wifi_config.ap.ssid, tmp, 11);
    
    wifi_config.ap.ssid_len = 11;
    wifi_config.ap.authmode = WIFI_AUTH_WPA2_PSK;
    wifi_config.ap.channel =  1;
    wifi_config.ap.ssid_hidden = 0;
    wifi_config.ap.max_connection = 3;
    wifi_config.ap.beacon_interval = 100;

    err = tcpip_adapter_dhcps_option(TCPIP_ADAPTER_OP_GET, REQUESTED_IP_ADDRESS, &lease, sizeof(lease));
    //ESP_ERROR_CHECK(err);
    printf("tcpip_adapter_dhcps_option=%d,DHCPS start=%d,end=%d\n", err,lease.start_ip.addr, lease.end_ip.addr);

    tcpip_adapter_ip_info_t ip_info = {0};
    err = tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_AP, &ip_info);
    //ESP_ERROR_CHECK(err);
    printf("tcpip_adapter_get_ip_info=%d,tcpip get ip ip=%d,gw=%d,net=%d\n", err,ip_info.ip.addr, ip_info.gw.addr, ip_info.netmask.addr);

    err = tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP);
    //ESP_ERROR_CHECK(err);

    ip_info.ip.addr = ipaddr_addr(OB_WIFI_AP_IP);
    ip_info.gw.addr = ipaddr_addr(OB_WIFI_AP_IP);
    ip_info.netmask.addr = ipaddr_addr(OB_WIFI_AP_MASK);
    err = tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_AP, &ip_info);
    //ESP_ERROR_CHECK(err);
    printf("tcpip_adapter_set_ip_info=%d,tcpip get ip ip=%d,gw=%d,net=%d\n", err,ip_info.ip.addr, ip_info.gw.addr, ip_info.netmask.addr);

    lease.enable = 1;
    lease.start_ip.addr = ipaddr_addr(OB_WIFI_AP_DHCPS_START_IP);
    lease.end_ip.addr = ipaddr_addr(OB_WIFI_AP_DHCPS_END_IP);

    err = tcpip_adapter_dhcps_option(TCPIP_ADAPTER_OP_SET, TCPIP_ADAPTER_REQUESTED_IP_ADDRESS, &lease, sizeof(lease));
    printf("tcpip_adapter_dhcps_option=%d,tcpip lease ip startip=%d,endif=%d\n", err,lease.start_ip.addr, lease.end_ip.addr);

    err = tcpip_adapter_dhcps_start(TCPIP_ADAPTER_IF_AP);
    printf("tcpip_adapter_dhcps_start=%d\n", err);

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI(TAG, "wifi_init_softap finished. SSID:%s password:%s channel:%d",
             wifi_config.ap.ssid, wifi_config.ap.password, wifi_config.ap.channel );
```

log show below

[image]

Then neither my phone nor my laptop can find the AP.
Does ```I (8049) phy: phy_filt_delay fail!!!, k=11``` mean some errors? 
And I cannot find any infomation related in esp-idf.

xueyunfei998, 2021-11-12:
hi @Dandjinh 
Please use idf's softap example to test. If there is no problem, configure softap according to this example.

location:
esp-idf/examples/wifi/getting_started/softAP

Alvin1Zhang, 2021-11-16:
@Dandjinh Thanks for reporting, would you please help share if any updates for the issue? Thanks.

zhangyanjiaoesp, 2022-05-23:
@Dandjinh Please update your IDF version  to the latest v4.4, the phy version has update to `I (459) phy_init: phy_version 2300,d67cf06,Feb 10 2022,10:03:07`

closed 2022-07-13 by Alvin1Zhang