espressif/esp-idf #7880: Some classic Bluetooth device names cannot be searched

wgcod, 2021-11-11:
I use Huawei mobile phone to connect Xiaomi headset and play music normally. At this time, the name of the headset can be scanned by another mobile phone, but esp32 cannot.

BetterJincheng, 2021-11-12:
Hi, @wgcod,

Thanks for reporting!

I tested all the speakers at hand and reach such a conclusion: When a speaker is in connection, it can be scanned by ESP32 as long as it can be scanned by any other device. 

Some speakers do not support multiple connections, so it will enter the non-discoverable state when it has been connected to a device.

Please check again the issue you post. (Maybe the other phone has remembered the speaker or just refresh scanning.)

Turn to us at any time.
Thanks

Alvin1Zhang, 2021-12-27:
@wgcod Thanks for reporting, will close due to short of feedback, feel free to reopen with more details.

closed 2021-12-27 by Alvin1Zhang