espressif/esp-idf #7882: What should the ULP power consumption be in deep sleep?

eyaleb, 2021-11-11:
Summary: the doco seems to suggest ULP uses 150uA but I see 450uA with the most minimal hwr and swr.

I want to run my project from battery so need to optimize power usage.

The hardware is a naked esp-wroom-32 module soldered to a carrier board. A low Iq LDO (RT9080) is used with a Li-Ion battery. No devices are attached yet. The sw is built with IDF 4.4.

When I go into deep sleep the current is 10-12uA, as expected. If I activate the empty ULP program it uses 450uA. The doco suggests it should be around 150uA.

I use [this](https://github.com/makehackvoid/ESP8266Lib/tree/master/examples/eyal/null) project. It now includes a hand drawn [sketch](https://github.com/makehackvoid/ESP8266Lib/tree/master/examples/eyal/null/20211111_213000.jpg) of the board components.

TIA

esp-lis, 2021-11-12:
try again after deleting these codes (https://github.com/makehackvoid/ESP8266Lib/blob/master/examples/eyal/null/main/ulp.c#L36-L37)

eyaleb, 2021-11-13:
No change. BTW, seeing all the commented lines in udp.c should tell that I tried many combinations with no success.
From the schematics one can see that only the EN pin has a 10k pull-up. Others are either open (most) or attached to 1m+1m voltage dividers which should draw very little current even if they are active.

I wonder if there is another esp_sleep_pd_config() to set.

To make it even clearer, I measure current at the battery rather than after the LDO. I did at one point measure at the esp32 pin and the difference was negligible. And anyway, the same project but with the ULP inactive draws only 10-12uA (depending on the meter used).

esp-lis, 2021-11-13:
I use the release/v4.4 version of the $IDF_PATH/examples/system/ulp_fsm/ulp example project to test on the ESP32-WROOM--32E module. When I delete the following two lines of code, the actual measurement results look normal.
```
rtc_gpio_isolate(GPIO_NUM_12);
rtc_gpio_isolate(GPIO_NUM_15);
```
The picture below is my measurement result:
[image]

I think the difference in our measurement results may come from the board or the measurement method. Perhaps you can provide a simple hardware schematic and measurement method to help troubleshoot the problem.

eyaleb, 2021-11-13:
To simplify the situation further I soldered [another ESP32](https://github.com/makehackvoid/ESP8266Lib/blob/master/examples/eyal/null/20211113_190015.jpg) on an empty board. Connected a LiFePO4 battery (measured as 3.32V) to GND and 3.3V and installed a 10k pull-up from EN to 3.3V. Nothing else.

To program it I temporarily attached the necessary wires but the test ran without anything else attached. The results are the same, measured from one of the battery wires. I used a number of meters to confirm the readings and they mostly agree.

I think this is as basic a hardware setup as I can use. I will appreciate all informed suggestions.

esp-lis, 2021-11-13:
I used your example project to test on the **ESP32-WROOM--32E** module, and the current is really large when deep sleep sleeps (refer to the figure below).
[image]
When I checked your code, I found that ULP is always running, so I modified some code and it looks OK (refer to the picture below)
[image]

You need to add a `halt` instruction [here](https://github.com/makehackvoid/ESP8266Lib/blob/master/examples/eyal/null/main/ulp/pulse.S#L4-L5), You need to add a halt instruction here, it will trigger ULP to sleep.

esp-lis, 2021-11-13:
Sorry, I got it wrong. As you mentioned, the current when ULP is working is higher than 150 uA (I measured 360 uA), I will confirm this problem.

eyaleb, 2021-11-13:
Regarding the "halt": the actual program is, naturally, more involved than this trimmed down test which, on purpose, establishes a specific situation. The full application uses the ULP to count pulses from a flow sensors which can arrive at a few hundred a second so it runs permanently. The main processor is in deep sleep , waking up as required to report the status (using wifi).

If the current usage of the ULP is higher than 150uA when it is active then this should probably be documented, and I would like to know what to expect.

esp-lis, 2021-11-15:
[image]
If the ADC, tsens sensor is not used in your project, you can modify the code according to the figure above, the current I tested is about 206 uA
[image]

eyaleb, 2021-11-15:
Thanks, a good start. Do you think we can go lower, as the doco says, to 150uA?

Both changes 1 (.../rtc_init.c) and 2 (.../ulp.c)  are required to drop the current to 250uA for me. The 3rd change has no effect.

Are you suggesting that this change will go into the repository? I would rather not have to patch ESP-IDF after every git pull.

I think it is now time for me to open my second issue, which is actually the first I encountered, The new issue is [to be inserted].

esp-lis, 2021-11-15:
[image]
You can try to modify it like this. The actual test current is as follows:
[image]

**The actual test ULP can run, but there is no guarantee that there will be no problems after such modification. This requires more time to test and verify.**

eyaleb, 2021-11-15:
Yes, very nice, I now see 130uA. Hope it does not break anything. Tomorrow I will test my full app with these changes.

I will wait for more verification, and will probably not use these changes until they land in github. This project still has some time to go before being commissioned, as I am tying loose ends.

eyaleb, 2021-11-23:
I see a status "Awaiting Response" but  I already responded "works OK". and it still does. My project uses the ULP just to count pulses on a digital pin so I cannot say how this change will affect other devices. Either accept the change or provide an API to control the feature (`RTC_CNTL_BIAS_*_FOLW_8M`) when required. Thanks.