espressif/esp-idf #7884: contextlib2 can't be installed

wushifu32, 2021-11-12:
Run ./install.sh got following error message:
```
Collecting contextlib2>=0.5.5
  Downloading https://dl.espressif.com/pypi/contextlib2/contextlib2-21.6.0-py2.py3-none-any.whl (13 kB)
ERROR: Package 'contextlib2' requires a different Python: 2.7.17 not in '>=3.6'
```
My PC env is: Ubuntu 18.04.4 LTS

dobairoland, 2021-11-12:
Hi @wushifu32.

What is the version of your ESP-IDF?

Python 2.7 is no longer supported by the Python community. Therefore, packages are dropping support for it.

If you use a recently patched version of ESP-IDF and install the required [package dependencies](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html) - `python3` and `python3-pip` in particular, then `install.sh` will detect and use Python 3 instead.

wushifu32, 2021-11-12:
My ESP-IDF has been updated to the commit  caa9e323cc9556855e122cabe289d61b49aaa479

The ./install.sh running message:
```
Detecting the Python interpreter
Checking "python" ...
Python 3.6.9
"python" has been detected
Installing ESP-IDF tools
WARNING: File /home/colin/.espressif/idf-env.json can not be created. 
Selected targets are: esp32s3, esp32s2, esp32c3, esp32
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r2-8.4.0 (already installed)
Skipping riscv32-esp-elf@esp-2021r2-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20210902 (already installed)
Installing Python environment and packages
===== here detected python2 =====
Python 2.7.17
pip 20.3.4 from /home/colin/.espressif/python_env/idf5.0_py3.6_env/local/lib/python2.7/site-packages/pip (python 2.7)
Installing Python packages from /home/colin/code/esp-idf/requirements.txt
```
But the default python & pip on My PC are allready version3+:
```bash
colin@T440:~/code/esp-idf$ pip --version
pip 21.3.1 from /usr/local/lib/python3.6/dist-packages/pip (python 3.6)
colin@T440:~/code/esp-idf$ python --version
Python 3.6.9
```

dobairoland, 2021-11-12:
Please try to remove `/home/colin/.espressif/python_env/idf5.0_py3.6_env/` and re-run the intall script.

closed 2021-11-12 by wushifu32