espressif/esp-idf #7885: ESP32S3 GPIO interrupts don't work if gpio_install_isr_service() executes on APP CPU

nonoo, 2021-11-12:
- Development Kit: ESP32-S3-DevKitC-1
- Kit version: v1
- Module or chip used: ESP32-S3-WROOM-1
- IDF version: v5.0-dev-308-g2c606c29b1
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

GPIO interrupts don't work if `gpio_install_isr_service()` executes on APP CPU. They work correctly if GPIO interrupts are handled on the PRO CPU. This issue is unrelated to #7594 as that is for the ESP32 (not the S3).

By comparing the SOC `gpio_struct.h` header files between ESP32 (`esp-idf/components/soc/esp32/include/soc/gpio_struct.h`) and ESP32S3 (`esp-idf/components/soc/esp32s3/include/soc/gpio_struct.h`), it seems that the ESP32S3 does not have the `acpu_int` struct.

I can also find the relevant separate core0/core1 registers in `esp-idf/components/soc/esp32s3/include/soc/interrupt_core0_reg.h` and `esp-idf/components/soc/esp32s3/include/soc/interrupt_core1_reg.h`, which should indicate that the APP CPU should be able to handle GPIO interrupts:

```cpp
interrupt_core0_reg.h: #define INTERRUPT_CORE0_GPIO_INTERRUPT_PRO_MAP_REG (DR_REG_INTERRUPT_CORE0_BASE + 0x040)
interrupt_core0_reg.h: #define INTERRUPT_CORE0_GPIO_INTERRUPT_APP_MAP_REG (DR_REG_INTERRUPT_CORE0_BASE + 0x048)

interrupt_core1_reg.h: #define INTERRUPT_CORE1_GPIO_INTERRUPT_PRO_MAP_REG (DR_REG_INTERRUPT_CORE1_BASE + 0x840)
interrupt_core1_reg.h: #define INTERRUPT_CORE1_GPIO_INTERRUPT_APP_MAP_REG (DR_REG_INTERRUPT_CORE1_BASE + 0x848)
```

### Expected Behavior

It should be possible to handle GPIO interrupts on the APP CPU (core 1).

### Actual Behavior

GPIO interrupts can only be handled on the PRO CPU (core 0).

### Steps to reproduce

Compile and run the source code below. It prints lines containing `0` every 500ms, and a line with `1` if a GPIO interrupt has received on GPIO0.

If the main task affinity in the `sdkconfig` is changed to core1 (set `CONFIG_ESP_MAIN_TASK_AFFINITY_CPU0=n` and `CONFIG_ESP_MAIN_TASK_AFFINITY_CPU1=y`) then the program will only display zeroes because the interrupt handler won't be triggered.

### Code to reproduce this issue

```cpp
#include <driver/gpio.h>

#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

#include <stdio.h>

static volatile uint8_t got_int = 0;

static void IRAM_ATTR main_isr_handler(void *arg) {
    got_int = 1;
}

void app_main(void) {
    int gpio_nr = 0;

    ESP_ERROR_CHECK(gpio_reset_pin(gpio_nr));
    ESP_ERROR_CHECK(gpio_set_direction(gpio_nr, GPIO_MODE_INPUT));
    ESP_ERROR_CHECK(gpio_set_pull_mode(gpio_nr, GPIO_PULLUP_ONLY));
    ESP_ERROR_CHECK(gpio_set_intr_type(gpio_nr, GPIO_INTR_NEGEDGE));
    ESP_ERROR_CHECK(gpio_install_isr_service(0));
    ESP_ERROR_CHECK(gpio_isr_handler_add(gpio_nr, main_isr_handler, NULL));

    while (1) {
        printf("%d\n", got_int);
        got_int = 0;
        vTaskDelay(pdMS_TO_TICKS(500));
    }
}
```

## Other items if possible

- [X] [sdkconfig file](https://github.com/espressif/esp-idf/files/7526173/sdkconfig.txt)
- [X] [elf file](https://github.com/espressif/esp-idf/files/7526128/esp-test.elf.gz)

AxelLin, 2021-11-12:
https://github.com/espressif/esp-idf/issues/7594

alexander-sorokin, 2021-11-14:
I have the same issue. GPIO interrupts are working only on GPIO 0. Also I tried GPIO 6, 8, 9, 47 and 21 (random). None of them worked with interrupts on any core. Is there some secret table with interrupt capable GPIOs?

nonoo, 2021-11-14:
@alexander-sorokin please read my issue description, as probably you have an issue which is not related to my ticket. GPIO interrupts work correctly for me regardless of the GPIO number on the ESP32S3 if the GPIO ISR service is started on the PRO CPU (core0). The issue is that they don't work if the ISR service is started on the APP CPU (core1). The example code should work for you in case the main task's affinity is set to core0.

nonoo, 2021-12-03:
Any follow ups?

closed 2021-12-09 by espressif-bot