espressif/esp-idf #7886: Freertos-addons - is contribution welcomed?

dsabala, 2021-11-12:
**Is your feature request related to a problem? Please describe.**

I would like to use michaelbecker's `freertos-addon` in ESP32 projects
https://github.com/michaelbecker/freertos-addons

**Describe the solution you'd like**
I would like to try to contribute to esp-idf project "porting" `freertos-addon` but first I want to ask you if this contribution is welcomed?
[Someone already integrated freertos-addon with esp32,](https://github.com/eagi223/freertos-addons-ESP32-Port) but it would be nice to make this module a formal part of esp-idf.

My workflow idea is to:
- make a `freertos-addon` fork and apply minimalistic fixes (mainly fix header include files and extend thread class constructors with CPU core id argument) so you guys can pull this fix into esspressif own repo fork
- integrate esp-idf ready examples as part of esp-idf repo pull request

What do you think guys? :) If I get the greenlight we can close this issue.

igrr, 2021-11-12:
@dsabala Thanks for the nice suggestion! Have you considered contributing the esp-related patches upstream to freertos-addons? I think an espressif fork of freertos-addons may become outdated/unmaintained. The upstream version can then be packaged as an ESP-IDF component, and published to the IDF component registry (https://components.espressif.com/). We can assist with setting up and automating the part related to the component packaging and publishing.

dsabala, 2021-11-12:
> Have you considered contributing the esp-related patches upstream to freertos-addons? I think an espressif fork of freertos-addons may become outdated/unmaintained. 

It could be made as a esp32 branch in freertos addons original repository, right?

> The upstream version can then be packaged as an ESP-IDF component, and published to the IDF component registry 

Honestly this doesnt seem rewarding - I noticed this repository dont work (click on package results in "fetch error". I guess you are proposing this registry due to limited espressif manpower and certain plans which esp-idf features gain the most attention from core developers. I will probably make standalone esp-idf component on my own. I recognise great chance to learn freertos. Anyway, if I succeed to make this component I will post a link  in this issue ;)

closed 2021-11-12 by dsabala