espressif/esp-idf #7887: When exporting datas：BrokenPipeError: [Errno 32] Broken pipe

YXW-123, 2021-11-12:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v2]
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

When using the command：idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA" > my-experiment-file.csv ,there is a error:BrokenPipeError: [Errno 32] Broken pipe

### Expected Behavior
Data exported successfully
### Actual Behavior
There is a error:BrokenPipeError: [Errno 32] Broken pipe

### Steps to reproduce

1. for ESP-CSI tool cd ./active_ap
2. Recording to the README.md use the commond: idf.py menuconfig
3. idf.py -p /dev/ttyUSB0 flash
4. use the commond :idf.py -p /dev/ttyUSB0 monitor ,then let my phone connect to the AP,it can collect CSI successfully.
[image]
5.use the commond:idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA" > my-experiment-file.csv or  "idf.py -p /dev/ttyUSB0 monitor | grep "CSI_DATA"", let my phone connect to the AP, then it printed the error like this :
[image]

dobairoland, 2021-11-12:
@YXW-123 There seems to be a guru meditation error before the broken pipe error. The pipe gets broken because the ESP device stops sending data through UART. Please try to solve the guru meditation error instead.

MaxwellAlan, 2021-11-15:
hi @YXW-123, can you help check the IDF version or upload a raw log of this issue, or you can try the new IDF release/v4.3 or master branch.

YXW-123, 2021-11-22:
> @YXW-123 There seems to be a guru meditation error before the broken pipe error. The pipe gets broken because the ESP device stops sending data through UART. Please try to solve the guru meditation error instead.

Thanks for your help!

YXW-123, 2021-11-22:
> hi @YXW-123, can you help check the IDF version or upload a raw log of this issue, or you can try the new IDF release/v4.3 or master branch.

Thanks for your reply! The IDF version is 4.2.1,when I use the v4.3 I find it isn't useful for CSI tools.I will try again.

MaxwellAlan, 2021-11-23:
hi @YXW-123 please try the master branch or the newest release/v4.3 and should call esp_err_t esp_wifi_config_11b_rate(wifi_interface_t ifx, bool disable); this api to disable 11b rate. since 11b rate does not contains csi info.

closed 2022-03-03 by Alvin1Zhang