espressif/esp-idf #7889: ESP32 S3, triggering watchdog without a crush

stdenits, 2021-11-13:
## Environment

- Development Kit:      [ESP32-S3-DevKitC-1]
- Module or chip used:  [ESP32-S3-WROOM-1]
- IDF version:    [v5.0-dev-329-g4a011f3183-dirty]
- Build System:         [CMake]
- Compiler version:     [???]
- Operating System:     [Windows]
- (Windows only) environment type: [???].
- Using an IDE?: [VSCode]
- Power Supply:         [USB]

## Problem Description
Application works before without any errors on ESP32C3.

After switching to ESP32 S3 it starts triggering watchdog at mqtt_task with IDLE.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

### Code to reproduce this issue

## Debug Logs

```
E (21838) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (21838) task_wdt:  - IDLE (CPU 0)
E (21838) task_wdt: Tasks currently running:
E (21838) task_wdt: CPU 0: mqtt_task
E (21838) task_wdt: CPU 1: IDLE
E (21838) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x42025FB2:0x3FC96CB00x40376DD2:0x3FC96CD0 0x42084840:0x600FF6A0 0x42063180:0x600FF6C0 0x420633FD:0x600FF700 0x420612E5:0x600FF740 0x42061431:0x600FF760 0x42061512:0x600FF780 0x4205F757:0x600FF7B0 
0x4205AE44:0x600FF8B0 0x4205AEEA:0x600FF920 0x4205AFB0:0x600FF9D0 0x42056952:0x600FFA30 0x420569F9:0x600FFBD0 0x4204833C:0x600FFC10 0x42046FA6:0x600FFC50 0x42047387:0x600FFC70 0x4204774F:0x600FFCF0 0x42049457:0x600FFD20 0x420B0869:0x600FFD50 0x4200CE8C:0x600FFD70 0x40380601:0x600FFDA0
0x42025fb2: task_wdt_isr at C:/Users/USER/esp-idf/components/esp_system/task_wdt.c:182 (discriminator 3)

0x40376dd2: _xt_lowint1 at C:/Users/USER/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1111

0x42084840: sha_ll_busy at C:/Users/USER/esp-idf/components/hal/esp32s3/include/hal/sha_ll.h:101 (discriminator 1)
 (inlined by) sha_hal_wait_idle at C:/Users/USER/esp-idf/components/hal/sha_hal.c:114 (discriminator 1)

0x42063180: esp_sha_dma_process at C:/Users/USER/esp-idf/components/mbedtls/port/sha/dma/sha.c:317

0x420633fd: esp_sha_dma at C:/Users/USER/esp-idf/components/mbedtls/port/sha/dma/sha.c:255

0x420612e5: esp_internal_sha512_dma_process at C:/Users/USER/esp-idf/components/mbedtls/port/sha/dma/esp_sha512.c:155

0x42061431: mbedtls_sha512_update_ret at C:/Users/USER/esp-idf/components/mbedtls/port/sha/dma/esp_sha512.c:232

0x42061512: mbedtls_sha512_finish_ret at C:/Users/USER/esp-idf/components/mbedtls/port/sha/dma/esp_sha512.c:297

0x4205f757: mbedtls_sha512_ret at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/sha512.c:503

0x4205ae44: entropy_update at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/entropy.c:222

0x4205aeea: entropy_gather_internal at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/entropy.c:318

0x4205afb0: mbedtls_entropy_func at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/entropy.c:393

0x42056952: mbedtls_ctr_drbg_reseed at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/ctr_drbg.c:363

0x420569f9: mbedtls_ctr_drbg_seed at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/ctr_drbg.c:441

0x4204833c: esp_create_mbedtls_handle at C:/Users/USER/esp-idf/components/esp-tls/esp_tls_mbedtls.c:102

0x42046fa6: create_ssl_handle at C:/Users/USER/esp-idf/components/esp-tls/esp_tls.c:74

0x42047387: esp_tls_low_level_conn at C:/Users/USER/esp-idf/components/esp-tls/esp_tls.c:413

0x4204774f: esp_tls_conn_new_sync at C:/Users/USER/esp-idf/components/esp-tls/esp_tls.c:488

0x42049457: ssl_connect at C:/Users/USER/esp-idf/components/tcp_transport/transport_ssl.c:119

0x420b0869: esp_transport_connect at C:/Users/USER/esp-idf/components/tcp_transport/transport.c:181

0x4200ce8c: esp_mqtt_task at C:/Users/USER/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1403

0x40380601: vPortTaskWrapper at C:/Users/USER/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:130

E (21838) task_wdt: Print CPU 1 backtrace

Backtrace:0x40378D35:0x3FC972B00x40376DD2:0x3FC972D0 0x400559DD:0x3FCF5360  |<-CORRUPTED
0x40378d35: esp_crosscore_isr at C:/Users/USER/esp-idf/components/esp_system/crosscore_int.c:84

0x40376dd2: _xt_lowint1 at C:/Users/USER/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1111
```

```
E (8858) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8858) task_wdt:  - IDLE (CPU 1)
E (8858) task_wdt: Tasks currently running:
E (8858) task_wdt: CPU 0: IDLE
E (8858) task_wdt: CPU 1: mqtt_task
E (8858) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x42025FB2:0x3FC96CB00x40376DD2:0x3FC96CD0 0x420AF74B:0x3FCF4C60 0x42003E06:0x3FCF4C80 0x4037E859:0x3FCF4CA0 0x40380601:0x3FCF4CC0 
0x42025fb2: task_wdt_isr at C:/Users/USER/esp-idf/components/esp_system/task_wdt.c:182 (discriminator 3)

0x40376dd2: _xt_lowint1 at C:/Users/USER/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1111

0x420af74b: cpu_ll_waiti at C:/Users/USER/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h:182
 (inlined by) esp_pm_impl_waiti at C:/Users/USER/esp-idf/components/esp_pm/pm_impl.c:850

0x42003e06: esp_vApplicationIdleHook at C:/Users/USER/esp-idf/components/esp_system/freertos_hooks.c:63

0x4037e859: prvIdleTask at C:/Users/USER/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3973 (discriminator 1)

0x40380601: vPortTaskWrapper at C:/Users/USER/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:130

E (8858) task_wdt: Print CPU 1 backtrace

Backtrace:0x40378D35:0x3FC972B00x40376DD2:0x3FC972D0 0x42084838:0x600FF6A0 0x42063180:0x600FF6C0 0x420633FD:0x600FF700 0x420612E5:0x600FF740 0x42061431:0x600FF760 0x42061512:0x600FF780 0x4205F757:0x600FF7B0 
0x4205AE44:0x600FF8B0 0x4205AEEA:0x600FF920 0x4205AFB0:0x600FF9D0 0x42056952:0x600FFA30 0x420569F9:0x600FFBD0 0x4204833C:0x600FFC10 0x42046FA6:0x600FFC50 0x42047387:0x600FFC70 0x4204774F:0x600FFCF0 0x42049457:0x600FFD20 0x420B0869:0x600FFD50 0x4200CE8C:0x600FFD70 0x40380601:0x600FFDA0
0x40378d35: esp_crosscore_isr at C:/Users/USER/esp-idf/components/esp_system/crosscore_int.c:84

0x40376dd2: _xt_lowint1 at C:/Users/USER/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1111

0x42084838: sha_hal_wait_idle at C:/Users/USER/esp-idf/components/hal/sha_hal.c:113

0x42063180: esp_sha_dma_process at C:/Users/USER/esp-idf/components/mbedtls/port/sha/dma/sha.c:317

0x420633fd: esp_sha_dma at C:/Users/USER/esp-idf/components/mbedtls/port/sha/dma/sha.c:255

0x420612e5: esp_internal_sha512_dma_process at C:/Users/USER/esp-idf/components/mbedtls/port/sha/dma/esp_sha512.c:155

0x42061431: mbedtls_sha512_update_ret at C:/Users/USER/esp-idf/components/mbedtls/port/sha/dma/esp_sha512.c:232

0x42061512: mbedtls_sha512_finish_ret at C:/Users/USER/esp-idf/components/mbedtls/port/sha/dma/esp_sha512.c:297

0x4205f757: mbedtls_sha512_ret at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/sha512.c:503

0x4205ae44: entropy_update at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/entropy.c:222

0x4205aeea: entropy_gather_internal at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/entropy.c:318

0x4205afb0: mbedtls_entropy_func at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/entropy.c:393

0x42056952: mbedtls_ctr_drbg_reseed at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/ctr_drbg.c:363

0x420569f9: mbedtls_ctr_drbg_seed at C:/Users/USER/esp-idf/components/mbedtls/mbedtls/library/ctr_drbg.c:441

0x4204833c: esp_create_mbedtls_handle at C:/Users/USER/esp-idf/components/esp-tls/esp_tls_mbedtls.c:102

0x42046fa6: create_ssl_handle at C:/Users/USER/esp-idf/components/esp-tls/esp_tls.c:74

0x42047387: esp_tls_low_level_conn at C:/Users/USER/esp-idf/components/esp-tls/esp_tls.c:413

0x4204774f: esp_tls_conn_new_sync at C:/Users/USER/esp-idf/components/esp-tls/esp_tls.c:488

0x42049457: ssl_connect at C:/Users/USER/esp-idf/components/tcp_transport/transport_ssl.c:119

0x420b0869: esp_transport_connect at C:/Users/USER/esp-idf/components/tcp_transport/transport.c:181

0x4200ce8c: esp_mqtt_task at C:/Users/USER/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1403

0x40380601: vPortTaskWrapper at C:/Users/USER/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:130
```

## Other items if possible

Dazza0, 2021-11-15:
Seems like it's getting stuck in either a length or infinite loop at `sha_hal.c:114` polling an SHA register.

```c
void sha_hal_wait_idle()
{
    while (sha_ll_busy()) {
    }
}
```

Could you try increasing the Task Watchdog timeout (e.g., doubling it) an dsee if the issue persists? It would be useful to determine whether the SHA peripheral is somehow stuck in a busy state or if the processing is actually taking that long.

stdenits, 2021-11-15:
>Could you try increasing the Task Watchdog timeout (e.g., doubling it) and see if the issue persists?
I did update to the last master branch.
I changed the CONFIG_ESP_TASK_WDT_TIMEOUT_S parameter from 5 to 15, but the problem persists: it just increase the occurrence interval. Moreover, the rest of the tasks continue to work normally, exclude mqtt_task.

Dazza0, 2021-11-15:
@stdenits My guess is that this is either a SHA hardware or driver problem (seems like the the busy register is never cleared). I'll forward this issue internally to the relevant code owner.

stdenits, 2021-11-15:
@Dazza0,
>If you have some sample code
Could you write to kaija78@gmail.com? I can send the whole project.
I don’t use Github and i don’t have enough experience to extract your e-mail from a commit.

Alvin1Zhang, 2022-01-05:
Thanks for reporting, and sorry for slow turnaround, fix is available https://github.com/espressif/esp-idf/commit/5a5fe48fcb46a4e04ae9af1b105b36ff69abd61c, feel free to reopen if the issue still happens.

closed 2022-01-05 by Alvin1Zhang