espressif/esp-idf #7890: Deprecation warning for mcpwm_isr_register()

gin66, 2021-11-13:
I am the author of [FastAccelStepper](https://github.com/gin66/FastAccelStepper/tree/espidf4.4). This provides for esp32 a high speed stepper motor driver (up to 200kStep/s) supporting up to six parallel active motors. This is enabled by the use of mcpwm and pcnt modules of the esp32. While upgrading to espidf 4.4 a deprecation warning has popped up: 
```
warning: 'esp_err_t mcpwm_isr_register(mcpwm_unit_t, void (*)(void*), void*, int, intr_handle_data_t**)' is deprecated: interrupt events are handled by driver, please use callback [-Wdeprecated-declarations]
         NULL, ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED, NULL);
```

Now I have checked for the proposed callback alternative. Apparently this callback only supports the capture events, but not the other great features of the mcpwm module. The library FastAccelStepper relies on interrupts linked to `op0_tea_int_st`, `op1_tea_int_st` and `op2_tea_int_st`.

That's why I request to either NOT deprecate this API call, or add a callback for interrupts related to these `opX_tea_int_st`.

BTW: I have noticed, that espressif has dropped the mcpwm modules in all new derivates. This means, espressif does not want its users to use the esp32-derivates for any kind of high performance stepper motor application ?

gin66, 2021-11-14:
Just have seen ESP32-S3 series has a comeback of mcpwm, but a smaller pulse counter.

suda-morris, 2021-11-15:
Hi @gin66 The MCPWM peripheral does have a lot more interrupts than just capture interrupts. But I remember the driver didn't ever exposed any API for users to use the events like "utez", "utez". I guess you enabled the interrupt by expanding the `mcpwm_intr_t` by yourself. If that's the case, we can remove those deprecations.

In ESP-IDF 5.0, the MCPWM driver will be totally re-design, exposing all abilities to user with much extensible APIs. We will provide all the callbacks to user and user don't have to handle the ISR low level stuffs by themselves.

> This means, espressif does not want its users to use the esp32-derivates for any kind of high performance stepper motor application.
MCPWM peripheral is resource consuming, so in some ESP32 derivates (like ESP32C3), it's not supported.

BTW, for stepper motor applications, you can also try the RMT peripheral, you can  use it to generate fixed number of pulses. See example https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/step_motor

suda-morris, 2021-11-16:
Yes, sure, the register file is always available.

Thanks for your trial with RMT, with regard to the RMT entry depth, it can be altered from https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/rmt.h#L75

Alvin1Zhang, 2021-11-22:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/e6ee8b2fcf0ba3d9f7ec4654ceb0d61fb4857da1, feel free to reopen.

closed 2021-11-22 by Alvin1Zhang