espressif/esp-idf #7892: No CMAKE_C_COMPILER (CMAKE_CXX_COMPILE,CMAKE_ASM_COMPILE) could be found.

dreamweaver90, 2021-11-13:
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40300.0 (4.3.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
WARNING: You are using pip version 21.2.4; however, version 21.3.1 is available.
You should consider upgrading via the 'C:\Users\Administrator\.platformio\penv\Scripts\python.exe -m pip install --upgrade pip' command.
Reading CMake configuration...
-- Could NOT find Git (missing: GIT_EXECUTABLE) 
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Didn't find assembler
-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "C:/Users/Administrator/Documents/PlatformIO/Projects/test202111093/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Administrator/Documents/PlatformIO/Projects/test202111093/.pio/build/esp32dev/CMakeFiles/CMakeError.log".

CMake Warning at C:/Users/Administrator/.platformio/packages/framework-espidf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  C:/Users/Administrator/.platformio/packages/framework-espidf/tools/cmake/idf.cmake:39 (include)
  C:/Users/Administrator/.platformio/packages/framework-espidf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:2 (include)

CMake Error at C:/Users/Administrator/.platformio/packages/framework-espidf/tools/cmake/project.cmake:308 (__project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

CMake Error at C:/Users/Administrator/.platformio/packages/framework-espidf/tools/cmake/project.cmake:308 (__project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

CMake Error at C:/Users/Administrator/.platformio/packages/framework-espidf/tools/cmake/project.cmake:308 (__project):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

dreamweaver90, 2021-11-13:
CMakeError.log file

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --target=arm-arm-none-eabi;-mcpu=cortex-m3 

Checking whether the ASM compiler is GNU using "--version" did not match "(GNU assembler)|(GCC)|(Free Software Foundation)":
Checking whether the ASM compiler is Clang using "--version" did not match "(clang version)":
Checking whether the ASM compiler is AppleClang using "--version" did not match "(Apple LLVM version)":
Checking whether the ASM compiler is ARMClang using "--version" did not match "armclang":
Checking whether the ASM compiler is HP using "-V" did not match "HP C":
Checking whether the ASM compiler is Intel using "--version" did not match "(ICC)":
Checking whether the ASM compiler is SunPro using "-V" did not match "Sun C":
Checking whether the ASM compiler is XL using "-qversion" did not match "XL C":
Checking whether the ASM compiler is MSVC using "-?" did not match "Microsoft":
Checking whether the ASM compiler is TI using "-h" did not match "Texas Instruments":
Checking whether the ASM compiler is IAR using "" did not match "IAR Assembler":
Checking whether the ASM compiler is ARMCC using "" did not match "(ARM Compiler)|(ARM Assembler)":
Checking whether the ASM compiler is NASM using "-v" did not match "(NASM version)":
Checking whether the ASM compiler is YASM using "--version" did not match "(yasm)":
Checking whether the ASM compiler is ADSP using "-version" did not match "Analog Devices":
Checking whether the ASM compiler is QCC using "-V" did not match "gcc_nto":
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags:

0xjakob, 2021-11-15:
@dreamweaver90 Did you do execute the [getting started instructions](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html) successfully?

Alvin1Zhang, 2022-03-17:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2022-03-17 by Alvin1Zhang

brad-colbert, 2024-04-08:
I have the same problem.

I installed esp-idf-tools-setup-offline-4.4.7.exe.

I launched the ESP-IDF 4.4 PowerShell and executed the following:

PS C:\Users\bradc\Projects\esp32-wifi-manager> idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory 'C:\Users\bradc\Dropbox\Projects\esp32-wifi-manager\build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
PS C:\Users\bradc\Projects\esp32-wifi-manager> idf.py menuconfig

But when I execute the menuconfig I get:

```
PS C:\Users\bradc\Projects\esp32-wifi-manager> idf.py menuconfig
Executing action: menuconfig
Running cmake in directory C:\Users\bradc\Dropbox\Projects\esp32-wifi-manager\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Users\bradc\Dropbox\Projects\esp32-wifi-manager"...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:1 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:1 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.23)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "C:/Users/bradc/Dropbox/Projects/esp32-wifi-manager/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/bradc/Dropbox/Projects/esp32-wifi-manager/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
```

KonstantinKondrashov, 2024-04-09:
Hi @brad-colbert!
You need to set up the tools before - https://docs.espressif.com/projects/esp-idf/en/v4.4.7/esp32/get-started/index.html#step-3-set-up-the-tools.

For Linux:
```bash
cd ~/esp/esp-idf
./install.sh
. ./export.sh
```

voidzero, 2024-07-16:
```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Install the latest PowerShell for new features and improvements! https://aka.ms/PSWindows

Using Python in C:\Espressif\python_env\idf5.2_py3.11_env\Scripts
Python 3.11.2
Using Git in C:\Espressif\tools\idf-git\2.44.0\cmd
git version 2.44.0.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v5.2.2
Checking Python compatibility
Adding ESP-IDF tools to PATH...
WARNING: Error while accessing the ESP-IDF version file in the Python environment: [Errno 2] No such file or directory: 'C:\\Espressif\\python_env\\idf5.2_py3.11_env\\idf_version.txt'

Name                           Value
----                           -----
OPENOCD_SCRIPTS                C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\s...
IDF_CCACHE_ENABLE              1
ESP_ROM_ELF_DIR                C:\Espressif\tools\esp-rom-elfs\20230320\
IDF_PYTHON_ENV_PATH            C:\Espressif\python_env\idf5.2_py3.11_env
ESP_IDF_VERSION                5.2
IDF_DEACTIVATE_FILE_PATH       C:\Users\user\AppData\Local\Temp\tmpem4lrg0xidf_26048

Added to PATH
-------------
C:\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin
C:\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin
C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin
C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin
C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
C:\Espressif\tools\cmake\3.24.0\bin
C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin
C:\Espressif\tools\ninja\1.11.1\
C:\Espressif\tools\idf-exe\1.0.3\
C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
C:\Espressif\frameworks\esp-idf-v5.2.2\tools
Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.2.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-v5.2.2\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of amazon-corretto-11-x64-windows-jdk, espressif-ide use command 'python.exe C:\Espressif\frameworks\esp-idf-v5.2.2\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\Espressif\frameworks\esp-idf-v5.2.2\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build

PS C:\Espressif\frameworks\esp-idf-v5.2.2> idf.py menuconfig
Executing action: menuconfig
Running cmake in directory C:\Espressif\frameworks\esp-idf-v5.2.2\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Espressif\frameworks\esp-idf-v5.2.2"...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Didn't find assembler
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.
[... 33 lines cut ...]
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-idf-v5.2.2/build/CMakeFiles/CMakeOutput.log".
See also "C:/Espressif/frameworks/esp-idf-v5.2.2/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1, output of the command is in the C:\Espressif\frameworks\esp-idf-v5.2.2\build\log\idf_py_stderr_output_11216 and C:\Espressif\frameworks\esp-idf-v5.2.2\build\log\idf_py_stdout_output_11216
PS C:\Espressif\frameworks\esp-idf-v5.2.2>

```

voidzero, 2024-07-17:
So the trick is to do this in a project directory. I have done the following in the esp-idf command window:

```
cd %userprofile%\esp
xcopy /e /i %IDF_PATH%\examples\get-started\hello_world hello_world
cd hello_world
idf.py set-target esp32c6
```

This works. So it was a bit more involved but apparently the trick is to do it in a project directory.

Zitrax, 2024-08-13:
If it helps someone - I had the same error and in my case the issue was that I had used:

```
include($ENV{IDF_PATH}/tools/cmake/idf.cmake)
```

instead of:

```
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
```