espressif/esp-idf #7893: "No such file or directory" when trying to merge examples

olsonap, 2021-11-13:
## Environment

- Development Kit:      None
- Kit version (for WroverKit/PicoKit/DevKitC): NA
- Module or chip used:  ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it): ESP-IDF v4.3.1-dirty
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes, Visual Studio Code
- Power Supply:        

## Problem Description

I think what I'm dealing with is a user-error, but I can't find the proper way to do what I'm trying to do.
Long story short, I'm taking two examples from esp-idf/examples. One is a2dp_gatts_coex, the other is blehr in the nimble folder.
I've copied the entire example directory (C:\esp\esp-idf\examples\bluetooth\nimble\blehr) into a different location (C:\esp\myProjects\blehr).
My main.c file uses aspects of both examples, and there are #includes from both. The problem is, I can't build it. The debug log section shows the terminal window contents.

I've tried adding the missing inclusions directly into my project, but it's a fools errand that eventually leads me to an inclusion that isn't discoverable (e.g. event/callback.h). It's a bad solution anyway.

### Expected Behavior

Project to be built.

### Actual Behavior

No such file or directory found.

### Steps to reproduce

1. Copy an example folder into a location outside the esp-idf directory.
2. Attempt to add requirements (e.g. #include "esp_a2dp_api.h")
3. Run "idf.py build"

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

[GIST](https://github.com/olsonap/blehr.git)

## Debug Logs

C:\esp\myProjects\blehr>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\esp\myprojects\blehr\build
Executing "ninja all"...
[1/9] Performing build step for 'bootloader'
ninja: no work to do.
[2/7] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj 
ccache C:\Users\aman_\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -IC:/esp/esp-idf/components/newlib/platform_include -IC:/esp/esp-idf/components/freertos/include -IC:/esp/esp-idf/components/freertos/port/xtensa/include -IC:/esp/esp-idf/components/esp_hw_support/include -IC:/esp/esp-idf/components/esp_hw_support/port/esp32/. -IC:/esp/esp-idf/components/heap/include -IC:/esp/esp-idf/components/log/include -IC:/esp/esp-idf/components/lwip/include/apps -IC:/esp/esp-idf/components/lwip/include/apps/sntp -IC:/esp/esp-idf/components/lwip/lwip/src/include -IC:/esp/esp-idf/components/lwip/port/esp32/include -IC:/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/esp/esp-idf/components/soc/include -IC:/esp/esp-idf/components/soc/esp32/. -IC:/esp/esp-idf/components/soc/esp32/include -IC:/esp/esp-idf/components/hal/esp32/include -IC:/esp/esp-idf/components/hal/include -IC:/esp/esp-idf/components/esp_rom/include -IC:/esp/esp-idf/components/esp_rom/esp32 -IC:/esp/esp-idf/components/esp_rom/include/esp32 -IC:/esp/esp-idf/components/esp_common/include -IC:/esp/esp-idf/components/esp_system/include -IC:/esp/esp-idf/components/esp32/include -IC:/esp/esp-idf/components/driver/include -IC:/esp/esp-idf/components/driver/esp32/include -IC:/esp/esp-idf/components/esp_ringbuf/include -IC:/esp/esp-idf/components/efuse/include -IC:/esp/esp-idf/components/efuse/esp32/include -IC:/esp/esp-idf/components/xtensa/include -IC:/esp/esp-idf/components/xtensa/esp32/include -IC:/esp/esp-idf/components/espcoredump/include -IC:/esp/esp-idf/components/esp_timer/include -IC:/esp/esp-idf/components/esp_ipc/include -IC:/esp/esp-idf/components/esp_pm/include -IC:/esp/esp-idf/components/vfs/include -IC:/esp/esp-idf/components/esp_wifi/include -IC:/esp/esp-idf/components/esp_wifi/esp32/include -IC:/esp/esp-idf/components/esp_event/include -IC:/esp/esp-idf/components/esp_netif/include -IC:/esp/esp-idf/components/esp_eth/include -IC:/esp/esp-idf/components/tcpip_adapter/include -IC:/esp/esp-idf/components/app_trace/include -IC:/esp/esp-idf/components/mbedtls/port/include -IC:/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/esp/esp-idf/components/app_update/include -IC:/esp/esp-idf/components/spi_flash/include -IC:/esp/esp-idf/components/bootloader_support/include -IC:/esp/esp-idf/components/nvs_flash/include -IC:/esp/esp-idf/components/pthread/include -IC:/esp/esp-idf/components/esp_gdbstub/include -IC:/esp/esp-idf/components/esp_gdbstub/xtensa -IC:/esp/esp-idf/components/esp_gdbstub/esp32 -IC:/esp/esp-idf/components/wpa_supplicant/include -IC:/esp/esp-idf/components/wpa_supplicant/port/include -IC:/esp/esp-idf/components/wpa_supplicant/include/esp_supplicant -IC:/esp/esp-idf/components/perfmon/include -IC:/esp/esp-idf/components/asio/asio/asio/include -IC:/esp/esp-idf/components/asio/port/include -IC:/esp/esp-idf/components/bt/common/osi/include -IC:/esp/esp-idf/components/bt/include/esp32/include -IC:/esp/esp-idf/components/bt/common/api/include/api -IC:/esp/esp-idf/components/bt/common/btc/profile/esp/blufi/include -IC:/esp/esp-idf/components/bt/common/btc/profile/esp/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include -IC:/esp/esp-idf/components/bt/host/nimble/port/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/util/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/include -IC:/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include -IC:/esp/esp-idf/components/bt/host/nimble/esp-hci/include -IC:/esp/esp-idf/components/cbor/port/include -IC:/esp/esp-idf/components/unity/include -IC:/esp/esp-idf/components/unity/unity/src -IC:/esp/esp-idf/components/cmock/CMock/src -IC:/esp/esp-idf/components/coap/port/include -IC:/esp/esp-idf/components/coap/port/include/coap -IC:/esp/esp-idf/components/coap/libcoap/include -IC:/esp/esp-idf/components/coap/libcoap/include/coap2 -IC:/esp/esp-idf/components/console -IC:/esp/esp-idf/components/nghttp/port/include -IC:/esp/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/esp/esp-idf/components/esp-tls -IC:/esp/esp-idf/components/esp-tls/esp-tls-crypto -IC:/esp/esp-idf/components/esp_adc_cal/include -IC:/esp/esp-idf/components/esp_hid/include -IC:/esp/esp-idf/components/tcp_transport/include -IC:/esp/esp-idf/components/esp_http_client/include -IC:/esp/esp-idf/components/esp_http_server/include -IC:/esp/esp-idf/components/esp_https_ota/include -IC:/esp/esp-idf/components/protobuf-c/protobuf-c -IC:/esp/esp-idf/components/protocomm/include/common -IC:/esp/esp-idf/components/protocomm/include/security -IC:/esp/esp-idf/components/protocomm/include/transports -IC:/esp/esp-idf/components/mdns/include -IC:/esp/esp-idf/components/esp_local_ctrl/include -IC:/esp/esp-idf/components/sdmmc/include -IC:/esp/esp-idf/components/esp_serial_slave_link/include -IC:/esp/esp-idf/components/esp_websocket_client/include -IC:/esp/esp-idf/components/expat/expat/expat/lib -IC:/esp/esp-idf/components/expat/port/include -IC:/esp/esp-idf/components/wear_levelling/include -IC:/esp/esp-idf/components/fatfs/diskio -IC:/esp/esp-idf/components/fatfs/vfs -IC:/esp/esp-idf/components/fatfs/src -IC:/esp/esp-idf/components/freemodbus/common/include -IC:/esp/esp-idf/components/idf_test/in [... cut ...] -DIDF_VER=\"v4.3.1-dirty\" -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj   -c ../main/main.c
In file included from ../main/main.c:21:
../main/bt_app_av.h:13:10: fatal error: esp_a2dp_api.h: No such file or directory
 #include "esp_a2dp_api.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

negativekelvin, 2021-11-14:
You're trying to use two different Bluetooth host stacks at the same time which is not supported

olsonap, 2021-11-14:
Would that be Bluedroid and NimBLE at the same time?

negativekelvin, 2021-11-14:
Correct

olsonap, 2021-11-14:
I'm not trying to use Bluedroid, but I am trying to have Bluetooth Classic and NimBLE coexist. Do you happen to see where I could "turn Bluedroid off" for lack of better wording?

negativekelvin, 2021-11-14:
Bluetooth classic features require bluedroid

olsonap, 2021-11-14:
Got it. I'm way off then. Thank you.

closed 2021-11-14 by olsonap