espressif/esp-idf #7894: Project does not compile after installing esp-idf v4.4-dev-3657-gdf0929009a

baldhead69, 2021-11-14:
Hi,

I installed esp-idf version v4.4-dev-3657-gdf0929009a following this link:
https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/get-started/windows-setup.html

When i compile the esp-idf project this error appear:
Cannot parse CMake requirements line: __component_set_property(__idf.global_conf REQUIRES "cxx;newlib;freertos;esp_hw_support;heap;log;lwip;soc;hal;esp_rom;esp_common;esp_system;esp32")

i tried to follow this:
#388

but when i put "ESP-IDF: Open ESP-IDF Terminal" in cmd or esp-idf cmd

'ESP-IDF:' is not recognized as an internal command
or external, an operable program or a batch file.

How to solve this problem ???

baldhead69, 2021-11-14:
I tried this, but not solved the problem.

[image]

baldhead69, 2021-11-14:
Some suggestion on how to solve this ?

Thank's.

baldhead69, 2021-11-15:
@brianignacio5 and @0xjakob and @igrr

baldhead69, 2021-11-15:
I am trying in 2 computers this installation and same problem on 2 computers.
I am stuck in this problem.
Whenever i go to install a new idf is this annoyance.
Ever.
For this reason i go months without updating esp idf (6 months or more).

Espressif should choose a single ide and focus all efforts on it.

brianignacio5, 2021-11-15:
@baldhead69 The command `ESP-IDF: Open ESP-IDF Terminal` is for the esp-idf extension in vscode. In terminal this doesn't work because it is not part of it.

If you are using the terminal itself, after updating `idd-component-manager`, delete your project's build folder and try again. Check if there is an environment variable `IDF_COMPONENT_MANAGER` enabled in your terminal. Delete it, delete the build folder and try to build again.

baldhead69, 2021-11-15:
@brianignacio5 

How to update idd-component-manager ?

I understand very little about the esp-idf build tools.

baldhead69, 2021-11-15:
@brianignacio5 

Did you mean idf-component-manager ?

If so, i already have installed.

[code]
IDF_TOOLS_PATH not set. Setting to C:\.espressif\
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\.espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\.espressif\tools\idf-git\2.30.1\cmd\
git version 2.30.1.windows.1
Setting IDF_PATH: C:\esp-idf

Adding ESP-IDF tools to PATH...
    C:\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin
    C:\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\bin
    C:\.espressif\tools\riscv32-esp-elf\esp-2021r2-8.4.0\riscv32-esp-elf\bin
    C:\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\.espressif\tools\cmake\3.20.3\bin
    C:\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210902\openocd-esp32\bin
    C:\.espressif\tools\ninja\1.10.2\
    C:\.espressif\tools\idf-exe\1.0.2\
    C:\.espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\esp-idf>pip

Usage:
  pip <command> [options]

Commands:
  install                     Install packages.
  download                    Download packages.
  uninstall                   Uninstall packages.
  freeze                      Output installed packages in requirements format.
  list                        List installed packages.
  show                        Show information about installed packages.
  check                       Verify installed packages have compatible dependencies.
  config                      Manage local and global configuration.
  search                      Search PyPI for packages.
  cache                       Inspect and manage pip's wheel cache.
  index                       Inspect information available from package indexes.
  wheel                       Build wheels from your requirements.
  hash                        Compute hashes of package archives.
  completion                  A helper command used for command completion.
  debug                       Show information useful for debugging.
  help                        Show help for commands.

General Options:
  -h, --help                  Show help.
  --debug                     Let unhandled exceptions propagate outside the main subroutine, instead of logging them
                              to stderr.
  --isolated                  Run pip in an isolated mode, ignoring environment variables and user configuration.
  -v, --verbose               Give more output. Option is additive, and can be used up to 3 times.
  -V, --version               Show version and exit.
  -q, --quiet                 Give less output. Option is additive, and can be used up to 3 times (corresponding to
                              WARNING, ERROR, and CRITICAL logging levels).
  --log <path>                Path to a verbose appending log.
[... 124 lines cut ...]
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Cannot parse CMake requirements line: __component_set_property(___idf_.global_conf REQUIRES "lvgl;cxx;newlib;freertos;esp_hw_support;heap;log;lwip;soc;hal;esp_rom;esp_common;esp_system;esp32")

CMake Error at C:/esp-idf/tools/cmake/component.cmake:238 (message):
Call Stack (most recent call first):
  C:/esp-idf/tools/cmake/build.cmake:491 (__component_get_requirements)
  C:/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:15 (project)

-- Configuring incomplete, errors occurred!
See also "C:/esp32Projects/WSS_SERVER_AND_UI_INTEGRATION/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

c:\esp32Projects\WSS_SERVER_AND_UI_INTEGRATION>
[/code]

baldhead69, 2021-11-15:
[image]
[image]

baldhead69, 2021-11-15:
> If you are using the terminal itself, after updating idd-component-manager, delete your project's build folder and try again. Check if there is an environment variable IDF_COMPONENT_MANAGER enabled in your terminal. Delete it, delete the build folder and try to build again.

This not solve.

0xjakob, 2021-11-15:
@baldhead69 Have you tried to delete the build folder as @brianignacio5 suggested [above](https://github.com/espressif/esp-idf/issues/7894#issuecomment-968516243)?

baldhead69, 2021-11-15:
> @baldhead69 Have you tried to delete the build folder as @brianignacio5 suggested [above](https://github.com/espressif/esp-idf/issues/7894#issuecomment-968516243)?
> 

Yes.
In all attempts.

igrr, 2021-11-15:
@baldhead69 do you happen to have a component directory named `.global_conf` inside your project? If yes, what is the contents of that directory? It looks like idf-component-manager considers it to be a component and fails because of the `.` symbol in the directory name.

baldhead69, 2021-11-15:
in esp-idf version "v4.4-dev-2359-g58022f859" the program compiled ok.
And anothers versions compiled ok too.

igrr, 2021-11-15:
I understand that, it's entirely possible that the same project worked before. — the error is related to the component manager, which is a new feature recently enabled in the master branch. I'm asking about that directory to try to identify the root cause of the issue. Once we know the cause, we can fix it.or give you a workaround so that you can move on with your project.

baldhead69, 2021-11-15:
The ".global_conf" is only a folder that i put lvgl component configurations like lv_conf.h, lv_conf_checker.h, lv_version.h and lvgl.h and a CMakeLists.txt file.

CMakeLists.txt file:
idf_component_register( INCLUDE_DIRS .							
						REQUIRES lvgl						
					  )

baldhead69, 2021-11-15:
@igrr 

I took the dot from the folder and the respective CMakeLists.txt files that referenced this folder and worked.

Will espressif guys try to fix this ?

Thank you very much.

Example:
CMakeLists.txt
```
file(GLOB littlevgl_srcs 	src/lv_core/*.c 
							src/lv_draw/*.c 
							src/lv_font/*.c 
							src/lv_hal/*.c 
							src/lv_misc/*.c
							src/lv_objx/*.c
							src/lv_themes/*.c
	)

idf_component_register(	SRCS ${littlevgl_srcs}
						INCLUDE_DIRS 	src/lv_core/include
										src/lv_draw/include
										src/lv_font/include
										src/lv_hal/include
										src/lv_misc/include
										src/lv_objx/include
										src/lv_themes/include
														
						REQUIRES global_conf				
						#REQUIRES fatfs sdmmc						
					  )
```

baldhead69, 2021-11-15:
Taking the opportunity to talk to you, what is the difference between these two files ?
I noticed that both esp32 and esp32-s3 have these 2 files.

xtensa-esp32-elf-gcc.exe
xtensa-esp32-elf-gcc-8.4.0.exe

baldhead69, 2021-11-15:
Hi,

Now i installed "ESP-IDF VS Code Extension" and all is working.

Nice work guys.

Thank's all.

Alvin1Zhang, 2021-11-16:
Thanks for reporting and sharing the updates, good to hear all is working now, feel free to reopen.

closed 2021-11-16 by Alvin1Zhang

baldhead69, 2021-11-16:
> Taking the opportunity to talk to you, what is the difference between these two files ?
> I noticed that both esp32 and esp32-s3 have these 2 files.
> 
> xtensa-esp32-elf-gcc.exe
> xtensa-esp32-elf-gcc-8.4.0.exe

I only would like to know this because i use this configuration for configure intellisense on visual studio code.

baldhead69, 2021-11-16:
A strange thing happened.

I started a installation from scratch on another computer and the dot folder problem go away.

Did you guys update anything already ?

kumekay, 2021-11-18:
> Did you guys update anything already?

No, support for `.` in component names is not published yet, 
but if the component manager is disabled, it shouldn't be a problem. Or, for example the issue won't appear if some CMake-based IDE is used, like Jetbrains CLion

kumekay, 2021-12-17:
@baldhead69  Just a quick question, did you ever have your project compiled successfully with `.global_conf` component (started with the leading dot)? Components with leading dots in the name were always ignored in the IDF: https://github.com/espressif/esp-idf/blob/master/tools/cmake/component.cmake#L112

baldhead69, 2021-12-17:
@kumekay 

yes, always compiled with esp32(i believe that the chip does not influence).

"Root" CMakeLists.txt
```c
# The following lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# (Not part of the boilerplate)
set(EXTRA_COMPONENT_DIRS "${CMAKE_SOURCE_DIR}/components/ui" "${CMAKE_SOURCE_DIR}/components/ui/global_conf" "${CMAKE_SOURCE_DIR}/components/communication")   # add components "drv" "lvgl" and ".global_conf". ".global_conf" only *.h global files. 
#  ${CMAKE_SOURCE_DIR} -> project root folder.						 

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(wss_server)
```

The .global_conf folder was inside the ui folder

kumekay, 2021-12-17:
Thank you for the quick reply, I thought you had `components/.global_conf` and wondered why it ever worked.

At the same time, the [CMake error](https://github.com/espressif/esp-idf/issues/7894#issuecomment-968546559) in the component manager you mentioned is fixed now and will be included in the next release of it.  But the support of components with leading dots like  `components/.global_conf` won't be enabled in IDF (at the same time  `components/ui/.global_conf` should be fine)