espressif/esp-idf #7896: Bluetooth always disconnects after 10 seconds

expresspotato, 2021-11-14:
## Environment

- Development Kit: ESP32-LyraT v4.3
- Module or chip used:  ESP32-WROVER-B
- IDF version: commit 2e74914051d14ec2290fc751a8486fb51d73a31e (HEAD, tag: v4.3.1)
- Build System: idf.py
- Operating System: Linux (Ubuntu)
- (Windows only) environment type: Command Prompt
- Using an IDE?: Yes, Eclipse
- Power Supply: USB @ 5v

## Problem Description

1. Bluetooth always disconnects after 10 seconds, when light sleep options are enabled (32 Khz External clock source)

### Expected Behavior

1. Bluetooth remains stable

### Actual Behavior

1. Disconnects and crashes

### Steps to reproduce

1. Download esp-idf v4.3.1
2. Go into esp-idf/examples/bluetooth/bluedroid/classic_bt/a2dp_sink
3. idf.py menuconfig with:
 - RTC Clock Source -> External 32 Khz Crystal
 - Bluetooth Sleep -> Use external 32 Khz Crystal
 - FreeRTOS -> Enable Tickless Idle Support
4. Compile and run with
5. Pair and connect with Samsung Galaxy S21 (or other devices)
6. Wait 10 seconds
7. Disconnected
8. Crashes

### Code to reproduce this issue
Compile and run the a2dp_sink example
Project code with HFP

## Debug Logs

```
I (31753) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: e96fbffc mode:0x2
D (31763) BT_BTM: btm_handle_to_acl_index: 128

**D (31763) BT_BTM: btm_pm_snd_md_req switching from ACTIVE to SNIFF.**
D (31773) BT_HCI: HCI Enqueue Comamnd opcode=0x803

D (31783) BT_HCI: Receive packet event_code=0xf

D (31783) BT_BTM: btm_handle_to_acl_index: 128

D (31793) BT_BTC: btc_thread_handler msg 1 11 19 0x3ffd3798

D (31793) BT_BTC: btc_av_state_opened_handler event: BTA_AV_RC_FEAT_EVT flags 0

D (31803) BT_BTC: btc_rc_handler event:BTA_AV_RC_FEAT_EVT
D (31803) BT_BTC: Peer_features:25b
D (31803) BT_HCI: Receive packet event_code=0x14

D (31813) BT_BTM: btm_handle_to_acl_index: 128

**D (31823) BT_BTM: btm_pm_proc_mode_change switched from UNKNOWN to SNIFF.**
D (31823) BT_BTM: btm mode change to active; check l2c_link for outgoing packets
D (31833) BT_L2CAP: l2c_link_check_send_pkts
D (31843) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (31843) BT_L2CAP: RR scan pri=0, lcid=0x0041, q_cout=0
D (31853) BT_L2CAP: RR scan pri=0, lcid=0x0040, q_cout=0
[... 300 lines cut ...]
D (39783) BT_L2CAP: l2cu_release_ccb: cid 0x0041  in_use: 1
D (39783) BT_HCI: btu_free_quick_timer Unable to find expected alarm in hashmap
D (39793) BT_HCI: btu_free_quick_timer Unable to find expected alarm in hashmap
D (39803) BT_L2CAP: l2cu_dequeue_ccb  CID: 0x0041
D (39803) BT_L2CAP: L2CA_GetDisconnectReason=8 
I (39813) BT_L2CAP: L2CA_DisconnectReq()  CID: 0x0042
D (39813) BT_L2CAP: L2CAP - st: OPEN evt: 27
I (39823) BT_BTM: BTM_SetPowerMode: pm_id 128 BDA: e96fbffc mode:0x0
D (39823) BT_BTM: btm_handle_to_acl_index: 128

**D (39833) BT_BTM: btm_pm_snd_md_req switching from SNIFF to ACTIVE.**

```

[bluetooth-disconnect.txt](https://github.com/espressif/esp-idf/files/7534191/bluetooth-disconnect.txt)

## Notes
* Even with light sleep not enabled at run time, the examples will still disconnect with unmodified code
* With light sleep enabled at runtime the cpu can enter light sleep when the current step is still on advertising. You can monitor the entry / exit of light sleep using the onboard LED (or another of your choosing) with the following code:

```
	gpio_set_direction(GPIO_NUM_22, GPIO_MODE_OUTPUT);
	gpio_set_level(GPIO_NUM_22, 1);
	gpio_sleep_sel_en(GPIO_NUM_22);
```

xiongweichao, 2021-11-15:
Hi @expresspotato 

Can you provide a demo and sdkconfig file that reproduces this issue?

Thanks

expresspotato, 2021-11-15:
Yes, see attached. Found the issue, but not the cause... If you enable tickless idle, external crystal, etc for light sleep but don't call the related pm_ functions before bluetooth to enable light sleep on power on, it will disconnect and crash. 

[a2dp_sink.zip](https://github.com/espressif/esp-idf/files/7542022/a2dp_sink.zip)

```idf.py build && sudo chmod 777 /dev/ttyUSB0 && idf.py flash && idf.py monitor```

xiongweichao, 2021-11-18:
Hi @expresspotato 

I have reproduced this problem and I am debugging. In order not to affect the work, you can temporarily set the CPU/APB MAX to 160M.
```
pm_config.max_freq_mhz = 160;
```

Thanks

xiongweichao, 2023-04-24:
Hi @expresspotato ,

Please try the latest [release/v4.3](https://github.com/espressif/esp-idf/tree/release/v4.3). Please let me know if the problem still exist.

Thanks

expresspotato, 2025-02-10:
This can be closed

closed 2025-02-10 by expresspotato