espressif/esp-idf #7897: crashed name: 'wifi' after running for a couple of  days

achao1104, 2021-11-15:
Hello,

IDF version:v4.3.1 and module is esp32s2 wrover

we used the sample of pppos and enabled wifi and NAT, it can run smoothly but occasionally crashed. We dumped the log but it looks nothing related to our application codes. please help.
 
==================== THREAD 1 (TCB: 0x3ffe3958, name: 'wifi') =====================
#0  0x4002eae1 in panic_abort (details=0x3ffcd920 \"abort() was called at PC 0x40027357 on core 0\") at D:/project/esp32/idf-v4.3.1/components/esp_system/panic.c:379
#1  0x4002f25c in esp_system_abort (details=0x3ffcd920 \"abort() was called at PC 0x40027357 on core 0\") at D:/project/esp32/idf-v4.3.1/components/esp_system/system_api.c:112
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

xueyunfei998, 2021-11-16:
HI @achao1104 
Please provide the commit id of idf and the recurring demo.

achao1104, 2021-11-19:
E (187517575) wifi:sched is null, if=0 tid=0 flags=0
E (187517575) wifi:trc.c rcReachRetryLimit 1741

E (187522565) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (187522565) task_wdt:  - IDLE (CPU 0)
E (187522565) task_wdt: Tasks currently running:
E (187522565) task_wdt: CPU 0: wifi

abort() was called at PC 0x40027357 on core 0

Backtrace:0x4002eade:0x3ffcd8e0 0x4002f259:0x3ffcd900 0x40038972:0x3ffcd920 0x40027357:0x3ffcd990 0x4002748d:0x3ffcd9c0 0x4002751a:0x3ffcd9e0 0x400f8af2:0x3ffcda10 0x400fbb25:0x3ffcdd20 0x40105c72:0x3ffcdd50 0x40034b4d:0x3ffcdd80 0x400c6ae1:0x3ffcddd0 0x400d950d:0x3ffcde40 0x400c86c6:0x3ffcde90 0x400cbf55:0x3ffcdeb0 0x400ccb6d:0x3ffcded0 0x400cbbae:0x3ffcdef0 0x400cd802:0x3ffcdf10 0x400c5d42:0x3ffcdf30 0x4002895d:0x3ffcdf60 0x4008abf7:0x3ffcdf80 0x400c3a13:0x3ffcdfa0 0x40027132:0x3ffcdfc0 0x400df6c9:0x3ffe3830 0x400db865:0x3ffe3860 0x400dbce5:0x3ffe38a0 0x400dbf47:0x3ffe38d0 0x4003a199:0x3ffe3910

ELF file SHA256: b017d89a7614548a

achao1104, 2021-11-19:
we used idf4.3.1 and pppos sample+nat, and used two camera to connect esp32s2.
from dump, you can not find any helpinfo, but we keep it running and display hte log with monitor, we catched some helpful log as above.
It looks wifi too busy to let idle task to execute?

xueyunfei998, 2021-11-25:
> E (187517575) wifi:sched is null, if=0 tid=0 flags=0 E (187517575) wifi:trc.c rcReachRetryLimit 1741
>
> E (187522565) task_wdt: Task watchdog got triggered. The following tasks did not reset the [...]

HI @achao1104 

Please provide the elf file and backtrace when watchdog appear. There is no useful information provided above.

xueyunfei998, 2021-11-25:
This problem is the same as https://github.com/espressif/esp-idf/issues/7936. 

I suggest closing This problem is the same as https://github.com/espressif/esp-idf/issues/7936. 

We will follow up this problem with https://github.com/espressif/esp-idf/issues/7897

jack0c, 2022-05-20:
@achao1104 did you use ADC in your project?

Alvin1Zhang, 2022-05-23:
As suggested in previous post, let us track in https://github.com/espressif/esp-idf/issues/7897

AxelLin, 2022-12-18:
> we used idf4.3.1 and pppos sample+nat, and used two camera to connect esp32s2. 

@achao1104 
Do you still hit the issue with latest 4.3.4 release?

jack0c, 2023-06-02:
Close since no reply for long

closed 2023-06-02 by jack0c