espressif/esp-idf #7898: ble_hid_device_demo 例程相关问题

SEASKY-Master, 2021-11-15:
ble_hid_device_demo 无法获取LED状态设动态设置电池电量

xiewenxiang, 2021-12-03:
需要动态的设置电池电量这个 characteristic 的 value 的值

Weijian-Espressif, 2022-03-30:
@SEASKY-Master 如果没有问题, 请关闭这个 issue

yuanyihan, 2022-10-28:
没有解决啊？

xiewenxiang, 2022-10-31:
@yuanyihan 
Battery power depends on the battery level characteristic. You can try adding the battery service and send notification to the remote side when the temperature changes.

[battery service](https://www.bluetooth.com/zh-cn/specifications/specs/battery-service-1-0/)

yuanyihan, 2022-10-31:
这个issue重点也是说明：
ble_hid_device_demo 无法获取LED状态，即PC无法向esp设备发送output，传送LED状态：
```
    /* Output 5-bit LED Indicator Kana | Compose | ScrollLock | CapsLock | NumLock */ \
    HID_USAGE_PAGE  ( HID_USAGE_PAGE_LED                   )       ,\
      HID_USAGE_MIN    ( 1                                       ) ,\
      HID_USAGE_MAX    ( 5                                       ) ,\
      HID_REPORT_COUNT ( 5                                       ) ,\
      HID_REPORT_SIZE  ( 1                                       ) ,\
      HID_OUTPUT       ( HID_DATA | HID_VARIABLE | HID_ABSOLUTE  ) ,\
      /* led padding */ \
      HID_REPORT_COUNT ( 1                                       ) ,\
      HID_REPORT_SIZE  ( 3                                       ) ,\
      HID_OUTPUT       ( HID_CONSTANT                            ) ,\
```
至于电池电量，，，我还没有到达大佬的那一步，我还有很多路要走~~~

esp-cjh, 2023-11-27:
The fixed has been merge to github: a86bc3c75aa8745b25b78ad8d4626c89c8a82934.
This issue will be closed and that they can reopen anytime.