espressif/esp-idf #7899: wifi long running connection break

achao1104, 2021-11-15:
Hello,

idf:4.3.1 and module:esp32s2

We used a camera to connect esp32s2 by wifi, it worked, but after a couple of hours, the connection will break, but if the camera connects huawei wifi, it can run for long time without any issue.

Can you think of any reason about it?
Thanks,
Peter

TianaESP, 2021-11-15:
Hi @achao1104, we cannot get enough info from your current description.
Please provide more info, for example, but not limited to:
1. The recurrence steps and the rate of the issue
2. logs, best to be as complete as possible

achao1104, 2021-11-16:
we only have dump log, plese take a look below.
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffe3958, name: 'wifi', GDB name: 'process 1073625432'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1d (StoreProhibitedCause)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x40089314
pc             0x4002eae1       0x4002eae1 <panic_abort+21>
lbeg           0x0      0
lend           0x0      0
lcount         0x0      0
sar            0x8      8
ps             0x60823  395299
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8002f25c       -2147290532
a1             0x3ffcd8e0       1073535200
a2             0x3ffcd920       1073535264
a3             0x3ffcd96d       1073535341
a4             0xa      10
a5             0x14     20
a6             0x3      3
a7             0x3ffd2ca4       1073556644
a8             0x0      0
a9             0x1      1
a10            0x3ffcd96d       1073535341
a11            0x3ffcd96d       1073535341
a12            0xa      10
a13            0xc      12
a14            0x3      3
a15            0x1c     28

==================== CURRENT THREAD STACK =====================
#0  0x4002eae1 in panic_abort (details=0x3ffcd920 \"abort() was called at PC 0x40027357 on core 0\") at D:/project/esp32/idf-v4.3.1/components/esp_system/panic.c:379
#1  0x4002f25c in esp_system_abort (details=0x3ffcd920 \"abort() was called at PC 0x40027357 on core 0\") at D:/project/esp32/idf-v4.3.1/components/esp_system/system_api.c:112
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

======================== THREADS INFO =========================
  Id   Target Id         Frame
* 1    process 1073625432 0x4002eae1 in panic_abort (details=0x3ffcd920 \"abort() was called at PC 0x40027357 on core 0\") at D:/project/esp32/idf-v4.3.1/components/esp_system/panic.c:379
  2    process 1073344660 0x4001a8d4 in ?? ()
  3    process 1073598248 0x4002fd12 in xQueueReceive (xQueue=0x3fffd7c0, pvBuffer=0x3ffdcef0, xTicksToWait=<optimized out>) at D:/project/esp32/idf-v4.3.1/components/hal/esp32s2/include/hal/cpu_ll.h:33
[... 80 lines cut ...]
.coredump.tasks.data 0x3ffe3958 0x154 RW
.coredump.tasks.data 0x3ffe37b0 0x1a0 RW
.coredump.tasks.data 0x3ff9f094 0x154 RW
.coredump.tasks.data 0x3ffd9a50 0x110 RW
.coredump.tasks.data 0x3ffdcf28 0x154 RW
.coredump.tasks.data 0x3ffdcdf0 0x130 RW
.coredump.tasks.data 0x3ffe2304 0x154 RW
.coredump.tasks.data 0x3ffe21f0 0x10c RW
.coredump.tasks.data 0x3ffe11a8 0x154 RW
.coredump.tasks.data 0x3ffe1090 0x110 RW
.coredump.tasks.data 0x3ff9fac8 0x154 RW
.coredump.tasks.data 0x3ff9f9d0 0xf0 RW
.coredump.tasks.data 0x3ff9f36c 0x154 RW
.coredump.tasks.data 0x3ffda9f0 0x174 RW
.coredump.tasks.data 0x3ff9fc20 0x154 RW
.coredump.tasks.data 0x3ffdb2a0 0xc8 RW
.coredump.tasks.data 0x3ffddb84 0x154 RW
.coredump.tasks.data 0x3ffdda30 0x14c RW
.coredump.tasks.data 0x3ffdff28 0x154 RW
.coredump.tasks.data 0x3ffdfe00 0x120 RW
.coredump.tasks.data 0x3ffdedcc 0x154 RW
.coredump.tasks.data 0x3ffdecc0 0x104 RW

===================== ESP32 CORE DUMP END =====================
===============================================================

MaxwellAlan, 2022-03-29:
Hi @achao1104 
From your core dump log, we only get that there are some operation in WiFi will cause the stack corruption, but lack of more details in the backtrace, it's hard for us to debug. Can you config the IDF Log level to Debug level, it will show more logs.
And what the camera is? And what operation when the camera connect to ESP32-S2 SoftAP?
Thanks.

jack0c, 2022-05-20:
@achao1104 again have you use ADC in your project? please use latest v4.3 and try again.

closed 2022-07-13 by Alvin1Zhang