espressif/esp-idf #7900: Post build scripts automation in ESP-IDF eclipse IDE

vim-git, 2021-11-15:
## Environment

- Development Kit:       ESP32-Mini-1
- Kit version :
- Module or chip used: ESP32
- IDF version : v4.2.1
- Build System: Make|CMake|idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:   Windows.
- (Windows only) environment type:ESP Command Prompt.
- Using an IDE?: Yes, Eclipse
- Power Supply: USB

## Problem Description

We want to execute the script automatically post-build for ESP32 in eclipse. Not getting any option with ESP-IDF plugins to execute the post-build script option in eclipse. 

### Expected Behavior

In a Normal eclipse without ESP-IDF plugin, There is a build option in the eclipse IDE. But the same option is not available with ESP-IDF plugin in eclipse. 

### Actual Behavior
[image]

### Steps to reproduce

1. Import any project of esp-idf v4.2.1 in eclipse
2. After building the code successfully try to execute any batch script automatically through Cmake or any eclipse option.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

0xjakob, 2021-11-15:
It looks like this is an [idf-eclipse-plugin](https://github.com/espressif/idf-eclipse-plugin) issue? @kolipakakondal.

igrr, 2021-11-15:
@vim-git  Aside from implementing this as an IDE feature there is also an option to add use `add_custom_command` with `POST_BUILD` inside project CMakeLists.txt file. The advantage of this approach is that the post-build step will be executed the same way for a command line build, should you need it at some point.

kolipakakondal, 2021-11-16:
Post-build steps - option is missing for CMake projects, will check further on this

Darktidelulz, 2022-12-17:
Has there been any progress on this?

igrr, 2022-12-18:
@Darktidelulz you can use CMake add_custom_target and add_custom_command functions to implement post-build steps. Here's a example: https://github.com/espressif/esp-idf/blob/edd815af2e7d541b25ff3a74c59a7fe0612df42d/tools/test_apps/system/bootloader_sections/CMakeLists.txt#L8

Darktidelulz, 2022-12-18:
@igrr  Thanks!
Was struggeling with getting it to work kept getting `No TARGET '(' has been created in this directory.`
Eventually found out that it was the commands in `add_custom_command` that had wrong syntax.

Found comments you made in the past about post_build and now settled on the following.

```
idf_build_get_property(elf EXECUTABLE)
add_custom_command(
TARGET ${elf}
POST_BUILD
COMMAND echo "This is post build" 
COMMAND powershell -ExecutionPolicy Unrestricted ${PROJECT_SOURCE_DIR}/post_build_ESP.ps1 ${PROJECT_BINARY_DIR} ${ProjectId} ${IDF_TARGET} 
)
```