espressif/esp-idf #7901: Where is v4.4?

olsonap, 2021-11-15:
I see lots of references to a v4.4 of the idf, but I don't see it on the release page. Arduino as a component requires v4.4 or greater for use, yet it's apparently not available.

igrr, 2021-11-15:
Hi @olsonap, apologies for the confusion. Arduino-esp32 project is currently using *release/v4.4 branch* of ESP-IDF. v4.4 release hasn't been made yet. At this time we are working on fixing the issues found by QA in release/v4.4 branch and preparing v4.4-beta1 pre-release. That will be followed by -rc and eventually the final release.
We'll look into making Arduino-esp32 documentation more clear about this.

olsonap, 2021-11-15:
That was an incredibly fast response at this hour. Thanks!

closed 2021-11-15 by olsonap