espressif/esp-idf #7904: How to tell a udp packet was sent?

eyaleb, 2021-11-15:
Maybe a question, maybe a feature request, maybe even an API deficiency.

A udp packet is lost if the send is followed by entering deep sleep.
However, a delay of 15ms before the sleep ensures all packets are sent. 10ms and you lose some, 5ms and none are sent. This is ugly.

Is there a reliable way to check that the transmit queue is empty? Maybe `wifi_tx_wait_idle()` similar to `uart_tx_wait_idle()`?

I asked this [same question on the forum](https://esp32.com/viewtopic.php?t=2037) over 4 years ago when there was no reliable solution but one hopes...

xueyunfei998, 2021-11-16:
hi @eyaleb 
There is no function you said for the time being.

But it can be replaced by a new scheme.

When you finish sending a udp packet, you can call wifi_stop and wifi_deinit, and then enter deepsleep.

Because when wifi_stop is executed, all data in the queue will be sent out.

eyaleb, 2021-11-16:
Thanks @xueyunfei998  but this does not work for me.

Doing the `stop+deinit` does not seem to flush the queue. Doing a small delay (20-30ms) instead allows the packet to go.
In the past I tried `wifi_disconnect()`+wait for status but it also did not wait for the packet to go.

I am looking for a nicer solution that does not use any magic delay but properly enters deep sleep once the packet is out.

xueyunfei998, 2021-11-17:
[read_phy.txt](https://github.com/espressif/esp-idf/files/7553339/read_phy.txt)

    esp_wifi_set_ps(WIFI_PS_MIN_MODEM); //  **Enter modesleep first.**   

    udp_client_send(); // **send udp data,** 

**If the udp packet is sent, enter deepsleep and execute the following code**
    while(1)
    {
        if (read_phy_state()==0){
            deepsleep();
            break;
        } 
    }

hi @eyaleb  I added an interface, see the attachment for details.

Then you can follow my steps above to realize the functions you want, and I passed the test here.

eyaleb, 2021-11-17:
Thanks @xueyunfei998, this seems to work. I needed to understand the order of the events, and I ended up with this:
```
send udp packet
while (0 != read_phy_state()) {}
esp_wifi_stop();
esp_wifi_deinit();
esp_deep_sleep();
```
Disappointingly, this process is rather lengthy, averaging 125ms, rarely less than 100ms but at times up to 250ms. This is against the app as a whole taking only a few 100s of milliseconds.

BTW, is there a need to issue `esp_wifi_set_ps()` when the doco says `WIFI_PS_MIN_MODEM` is already the default?

xueyunfei998, 2022-06-07:
HI @eyaleb 

You can also use another method,

Included in this header file in your main.c file，#include "esp_private/wifi.h"

Then, after obtaining the ip address, call this function to register a data sending completion callback function.

`esp_wifi_set_tx_done_cb`

closed 2022-07-13 by Alvin1Zhang