espressif/esp-idf #7907: How do I fill unused code memory using linker in esp-idf eclipse?

dimpalsharm, 2021-11-15:
## Environment

- Development Kit:  ESP32-Mini-1
- Kit version 
- Module or chip used: ESP32
- IDF version :v4.2.1
   
- Build System:   Make|CMake|idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool NCG esp-2020r3)8.4.0
- Operating System:    Windows
- (Windows only) environment type: Command Prompt
- Using an IDE?: Yes (Eclipse)
- Power Supply:  USB

## Problem Description

I am looking for option, on how to fill unused code memory using some specific data like 0xFF using linker script.
I am trying to fill my binary file with 0xFF up to some address manually and looking for automatic option.
My binary file size is up to 0x1000 and after that I am trying to fill 0x1000 to 0x1100.

We tried this [as shown in below snip],but didn't work.
[image]

Could you please help with this.

Thanks!!

0xjakob, 2021-11-16:
@dimpalsharm Are you trying to create memory checks aka "heap poisoning"? If yes, there should be an option in somewhere in menuconfig. Heap poisoning is activated in the IDF unit tests per default.

dimpalsharm, 2021-11-16:
Nope, we are trying for memory check. My requirement is to calculate checksum of binary starting from 0x00 to ox1000.
However, my binary size is 0x890 only. So I wanted to fill remaining part upto  0x100 with 0xFF.

o-marshmallow, 2021-11-16:
Hi @dimpalsharm ,
Can you give a bit more details on why it did not work? Was the section created? Was the address in the binary incorrect? Was the load address incorrect? Was the size incorrect?

From a linker script point of view, you can create a section that will be filled with bytes that you want. Let's take the ESP32  linker script and add an empty section of size 0x100 after `.iram0.data` section:
```
  .iram0.data :
  {
    . = ALIGN(4);
    _iram_data_start = ABSOLUTE(.);

    mapping[iram0_data]

    _iram_data_end = ABSOLUTE(.);
  } > iram0_0_seg
```
We can add another section right after that will be filled with 0xFF, to do so, here is what to add after `.iram0.data`:
```
  .iram0.data :
  {
    . = ALIGN(4);
    _iram_data_start = ABSOLUTE(.);

    mapping[iram0_data]

    _iram_data_end = ABSOLUTE(.);
  } > iram0_0_seg

  .iram0.fill :
  {
    FILL(0xFF);
    _fill_start = .;
    BYTE(0xFF);
    . = . + 0x100 - 1;
  } > iram0_0_seg
```
We need to subtract 1 to the latest declaration because `BYTE` already adds a byte to the section. After executing `readelf` on the generated binary, we get:
```
  [...]
                                         LoadAddr Offset Size
  [19] .iram0.data       PROGBITS        4008b104 02d010 000000 00   W  0   0  1
  [20] .iram0.fill       PROGBITS        4008b104 017104 000100 00  WA  0   0  1
  [...]
```
After analyzing the generated binary, we can see the FF bytes at the offset pointed by `readelf`:
```
$ xxd -s +0x17104 -l 0x110 build/your_bin.elf
00017104: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00017114: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00017124: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00017134: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00017144: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00017154: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00017164: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00017174: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00017184: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00017194: ffff ffff ffff ffff ffff ffff ffff ffff  ................
000171a4: ffff ffff ffff ffff ffff ffff ffff ffff  ................
000171b4: ffff ffff ffff ffff ffff ffff ffff ffff  ................
000171c4: ffff ffff ffff ffff ffff ffff ffff ffff  ................
000171d4: ffff ffff ffff ffff ffff ffff ffff ffff  ................
000171e4: ffff ffff ffff ffff ffff ffff ffff ffff  ................
000171f4: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00017204: 0000 0000 0000 0000 0000 0000 0000 0000  ................
```

So this will create a filled section right after an existing one. According to GNU Linker documentation, you can still provide a VMA/LoadAddress to the section and also a hardcoded offset in the ELF file. (Here is the documentation: https://ftp.gnu.org/old-gnu/Manuals/ld-2.9.1/html_chapter/ld_3.html#SEC21)

dimpalsharm, 2021-11-17:
Hi @o-marshmallow,

Thank you so much for the response!

I am able to fill the memory up to plus(+)100 from the end of my bin file,
but , what I want to fill is exactly upto a  some specific location (upto 0x17F000...in my case) from the end of he bin file.

Could you suggest any modification for that.

o-marshmallow, 2021-11-18:
@dimpalsharm Unfortunately, there are no way to control the offset of a section with a linker script file. In fact, we can't retrieve the offset of a section in the ELF. Thus, in the generated binary, we won't have this possibility either.

However, from what you said earlier, what you are trying to do is make your binary file up to a certain size by filling it out, the goal is to perform a checksum on it (on the host, right?). So, I would say that you don't even need a section to do this as you won't really use any of the symbols. In that case, you can use the Linux commands to perform the following operations:
- Generate a large amount of 0xFF (0377 in octal)
- Append them to your binary
- Truncate your binary to the desired size (0x1000)
All of this can be done with:
```
yes '' | tr "\n" "\377" | head -c 1024 >> your_binary.bin && truncate -s 4096 your_binary.bin
```
Thus, you can perform this in a script. But keep in mind that this padding is not part of the system image: it does not belong to any segment or section, it is just file padding.

o-marshmallow, 2022-08-16:
Closing this issue

closed 2022-08-16 by o-marshmallow