espressif/esp-idf #7908: ULP uses high current in deep sleep if the app uses wifi

eyaleb, 2021-11-15:
Testing a project which uses the ULP and is expected to run on batteries I encountered this problem.

The app wakes up every period and uses wifi to send one udp packet. It uses all static config.

Normally, in deep sleep, the project uses 4uA, which is expected.
However, if the ULP is enabled (but is in a simple "jump entry" loop) the app uses about 1.87mA. This is tested with a minimal board where I soldered an ESP-WROOM-32 on an empty board with a 10k pull-up on the EN pin. Nothing else. Power is directly from a LiFePO4 battery showing 3.32V. Current checked with a few meters.

So, with ULP active: enabling wifi leads to 1.87mA current in deep sleep, disabling wifi uses 220uA. Disabling both uses 4uA.
Expected (as the doco says) is 150uA with ULP active, 4uA without.

Along the way I logged another [issue](https://github.com/espressif/esp-idf/issues/7882) which is making good progress. There, with no wifi and with the ULP enabled, the app used 420uA (expected is 150uA) and now, with some suggested changes, uses 220uA.

esp-lis, 2021-11-15:
It may be that tsens has opened xpd_sar in the PHY, and I am checking and fixing this problem.

esp-lis, 2021-11-16:
What is mode of WiFi? **AP** or **station**

eyaleb, 2021-11-16:
The app is a wifi station. Very basic: do some readings and send a udp packet. For example [this](https://github.com/makehackvoid/ESP8266Lib/tree/master/examples/eyal/wifi-v4.3) is a cut down version of my app, but I tested with an even simpler version.

esp-lis, 2021-11-16:
Add the following code [here](https://github.com/makehackvoid/ESP8266Lib/blob/master/examples/eyal/wifi-v4.3/main/udp.c#L160) and test it, my test is ok.
```
esp_wifi_stop();
esp_wifi_deinit();
```
I try to support deep sleep without this restriction (wifi stop and deinit, in fact it is a friendly way).

chegewara, 2021-11-16:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#wifi-bt-and-sleep-modes

> Before entering deep sleep or light sleep modes, applications must disable WiFi and BT using appropriate calls (esp_bluedroid_disable(), esp_bt_controller_disable(), esp_wifi_stop()).

eyaleb, 2021-11-16:
Thanks, I should have read the doco more closely... It now drops to 140uA, very reasonable, as good as when there is no wifi active. This is the same as in my other [issue](https://github.com/espressif/esp-idf/issues/7882).

The down side is that it adds 50ms to the app active time (380ms->430ms) which will hurt the battery energy budget. Is this time coming from a slow shutdown, or is it due to a longer startup now that the wifi was properly shut?

Now all I need is a solution to this final [issue](https://github.com/espressif/esp-idf/issues/7904) and I am good.

chegewara, 2021-11-16:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/index.html

cmsedore, 2022-01-03:
[...discussion noting the requirement to call esp_wifi_stop() before deep sleep...]

> The down side is that it adds 50ms to the app active time (380ms->430ms) which will hurt the battery energy budget. Is this time coming from a slow shutdown, or is it due to a longer startup now that the wifi was properly shut?
> 
> Nevertheless, this slowdown is worth it. Assuming the app wakes up every 1 minute, the extra 50ms at, say 50mA, consumes (50ms*50mA=) **2.5mAs/m**. The savings are (1.87-.013=1.74mA * 60s=) **104.4mAs/m**.

I've run into this same issue -- is there a faster way to shut down WiFi? (can I poke a register to shut off the power directly? an internal _wifi_off() function?) I don't care about graceful disconnect, etc, I want to save the ~50ms.  While ~50ms isn't that much, it is representing a third of the time that the system is awake in my application--this is a couple years of battery life...

jack0c, 2022-04-27:
The speed to run `esp_wifi_stop()` depends on the Wi-Fi MAC status, which means if you still have packets  in transmitting, we can stop immediately, and if we are connected to an AP, we need to disconnect from it(otherwise you can't connect to it easily next time). For your projects, if there is not a packet left when you `esp_wifi_stop()` and no need to disconnect from AP, it will be very fast.
In which case you will not have packets to transmit? That is you have already check your packets transmitted before you run `esp_wifi_stop()`, which in most case is required. @cmsedore Can you check about this?

esp-wzh, 2023-05-22:
Hi all, after this [commit](https://github.com/espressif/esp-idf/commit/94034246b28f5dfde6184a1346b3a7a935137e11), it is no longer necessary to call `disable_wifi/bt` before deep sleep in user code. We have added the shutdown of RF to the process of deep_sleep. This will greatly speed up the speed of entering sleep. But similarly, you need to guarantee the transmission of the WiFi/BT has been completed in the application.

jack0c, 2023-06-02:
Close since no reply for long.

closed 2023-06-02 by jack0c