espressif/esp-idf #7909: esp_netif_create_ip6_linklocal fails when called from ETHERNET_EVENT_CONNECTED handler

RJPoelstra, 2021-11-15:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  [ESP32-SOLO-1]
- IDF version (run ``git describe --tags`` to find it):
    v4.4-dev-3657-gdf0929009a
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    esp-2021r2-8.4.0 
- Operating System:     [Windows]
- (Windows only) environment type: [Eclipse].
- Using an IDE?: [Yes Eclipse]
- Power Supply:         [external 5V]

## Problem Description

I call `esp_netif_create_ip6_linklocal()` in my handler for the `ETHERNET_EVENT_CONNECTED` event.
This call fails in the release/4.4 branch (i.e. I do not get an IPv6 address assigned).

If I call it in the handler for the `IP_EVENT_ETH_GOT_IP` event, the call succeeds (and I get an `IP_EVENT_GOT_IP6` event).

Before I switched to the release/4.4 branch I used to be on master, and there it all worked from the `ETHERNET_EVENT_CONNECTED` handler.

### Expected Behavior
I get an IPv6 address

### Actual Behavior
No IPv6 address is assigned.

### Steps to reproduce

On release/4.4 call `esp_netif_create_ip6_linklocal()` from a `ETHERNET_EVENT_CONNECTED` handler.

### Code to reproduce this issue

In the initialisation code:
```cpp
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL));
```

The handlers:
```cpp
static void eth_event_handler(void *arg, esp_event_base_t event_base,
                              int32_t event_id, void *event_data)
{
    /* we can get the ethernet driver handle from event data */
//    esp_eth_handle_t eth_handle = *(esp_eth_handle_t *)event_data;

    switch (event_id) {
    case ETHERNET_EVENT_CONNECTED:
        ESP_LOGI(TAG, "Ethernet Link Up");
        esp_netif_create_ip6_linklocal(eth_netif); //This call fails
        break;
    case ETHERNET_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "Ethernet Link Down");
        break;
    case ETHERNET_EVENT_START:
        ESP_LOGI(TAG, "Ethernet Started");
        break;
    case ETHERNET_EVENT_STOP:
        ESP_LOGI(TAG, "Ethernet Stopped");
        break;
    default:
        break;
    }
}

/** Event handler for IP_EVENT_ETH_GOT_IP */
static void got_ip_event_handler(void *arg, esp_event_base_t event_base,
                                 int32_t event_id, void *event_data)
{
//    ip_event_got_ip_t *event = (ip_event_got_ip_t *) event_data;
//    const esp_netif_ip_info_t *ip_info = &event->ip_info;
    esp_netif_create_ip6_linklocal(eth_netif); //This call succeeds
}
```

I noticed that in  `esp_netif_lwip.c:1618`:
```cpp
static esp_err_t esp_netif_create_ip6_linklocal_api(esp_netif_api_msg_t *msg)
{
    esp_netif_t *esp_netif = msg->esp_netif;

    ESP_LOGV(TAG, "%s esp-netif:%p", __func__, esp_netif);

    struct netif *p_netif = esp_netif->lwip_netif;
    if (p_netif != NULL && netif_is_up(p_netif)) {
        netif_create_ip6_linklocal_address(p_netif, 1);
        nd6_set_cb(p_netif, esp_netif_nd6_cb);
        return ESP_OK;
    } else {
        return ESP_FAIL;
    }
}
```

The call to netif_is_up(p_netif) returns `false`. This seems odd as the event should be send when the netif is up.

david-cermak, 2021-11-16:
Hi @RJPoelstra 

Was your initialization code:
```cpp
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL));
```
executed after the ethernet driver had been initialized?

Please note the warning in the [ethernet docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_eth.html#connect-driver-to-tcp-ip-stack) about registering user handlers.

This works on `v4.3` because the system handlers are registered explicitly in the example.

> The call to netif_is_up(p_netif) returns false. This seems odd as the event should be send when the netif is up.

The reason is that your user handler was registered first and thus was run before the system one which initializes TCPIP connection and sets the netif up.

RJPoelstra, 2021-11-16:
Thanks!
I had totally missed that warning (at least after I switched to release/4.4).
It's working as expected now.

closed 2021-11-16 by RJPoelstra