espressif/esp-idf #7912: WPA2 enterprise reset IP each 12 hours

godofwars1111, 2021-11-15:
## Environment

- Development Kit:   ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32U
- IDF version (run ``git describe --tags`` to find it): v4.2.2-375-gf370d5089f
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description
When we connect  with the WPA2 enterprise authentification we add this issue 
         Each twelve hours the ip adress is reset. That cause a reconnection
         We got this in our softwars and the  wpa2_enterprise example
When we connect with wpa2 authentification we don't have this issue

### Expected Behavior

It supposed to not lose is IP each 12 hours and keep the wifi connection,

### Actual Behavior

the IP adress is reseting after twelve hours. That cause a reconnection 

### Steps to reproduce

Just excute the example about the wpa2 enterprise autenthification 
(esp-idf\examples\wifi\wpa2_enterprise) and wait twelves hours

### Code to reproduce this issue

The bug append with the example about the wpa2 enterprise autenthification 
(esp-idf\examples\wifi\wpa2_enterprise)

## Debug Logs

`I (42650205) example: IP:192.168.30.158
I (42650205) example: MASK:255.255.255.0
I (42650205) example: GW:192.168.30.1
I (42650205) example: ~~~~~~~~~~~
I (42652215) example: ~~~~~~~~~~~
I (42652215) example: IP:0.0.0.0
I (42652215) example: MASK:0.0.0.0
I (42652215) example: GW:0.0.0.0
I (42652215) example: ~~~~~~~~~~~
I (42654215) example: ~~~~~~~~~~~
I (42654215) example: IP:192.168.30.158
I (42654215) example: MASK:255.255.255.0
I (42654215) example: GW:192.168.30.1

I (85853135) example: IP:192.168.30.158
I (85853135) example: MASK:255.255.255.0
I (85853135) example: GW:192.168.30.1
I (85853135) example: ~~~~~~~~~~~
I (85855145) example: ~~~~~~~~~~~
I (85855145) example: IP:0.0.0.0
I (85855145) example: MASK:0.0.0.0
I (85855145) example: GW:0.0.0.0
I (85855145) example: ~~~~~~~~~~~
I (85857145) example: ~~~~~~~~~~~
I (85857145) example: IP:192.168.30.158
I (85857145) example: MASK:255.255.255.0
I (85857145) example: GW:192.168.30.1
`

## Other items if possible
[teraterm-log-wpa2-ent-peap.txt](https://github.com/espressif/esp-idf/files/7540005/teraterm-log-wpa2-ent-peap.txt)

negativekelvin, 2021-11-15:
There are no useful logs there, you will have to turn on debug logging for wifi and dhcp

sagb2015, 2021-11-16:
@godofwars1111 Can you please provide radius server configuration as well ? Is there any Session-Timeout set?

godofwars1111, 2021-11-19:
@negativekelvin we turn on the wifi log in menuconfig and  dhcp in the code with esp_log_level_set("dhcpc", ESP_LOG_INFO); but we got the same l
[teraterm-2021-11-18-kitesp32-verbose.txt](https://github.com/espressif/esp-idf/files/7571014/teraterm-2021-11-18-kitesp32-verbose.txt)
og

negativekelvin, 2021-11-19:
You have to rebuild the project with higher log levels in menuconfig

negativekelvin, 2021-11-19:
Do you get other logs at startup to confirm logging is working for those components?

godofwars1111, 2021-11-22:
@negativekelvi
This is the new log

D (42963972) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdd60
I (42963972) wifi: ~~~~~~~~~~~
I (42963972) wifi: IP:192.168.30.158
I (42963972) wifi: MASK:255.255.255.0
I (42963972) wifi: GW:192.168.30.1
I (42963982) wifi: ~~~~~~~~~~~
D (42965982) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdd60
I (42965982) wifi: ~~~~~~~~~~~
I (42965982) wifi: IP:192.168.30.158
I (42965982) wifi: MASK:255.255.255.0
I (42965982) wifi: GW:192.168.30.1
I (42965992) wifi: ~~~~~~~~~~~
D (42967592) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffbdde4
D (42967592) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbdd60
D (42967602) esp_netif_lwip: if0x3ffbdd60 start ip lost tmr: interval=120
D (42967992) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdd60
I (42967992) wifi: ~~~~~~~~~~~
I (42967992) wifi: IP:0.0.0.0
I (42967992) wifi: MASK:0.0.0.0
I (42967992) wifi: GW:0.0.0.0
I (42968002) wifi: ~~~~~~~~~~~
D (42969492) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffbdde4
D (42969492) esp_netif_lwip: if0x3ffbdd60 ip unchanged
D (42970002) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdd60

[log.txt](https://github.com/espressif/esp-idf/files/7582387/log.txt)
n

negativekelvin, 2021-11-22:
I think you need to turn on dhcp debug logging within lwip component too to find out where esp_netif_dhcpc_cb called from

godofwars1111, 2021-11-22:
@negativekelvin could you gave me the path where i have to add code to active the log in the lwip component

negativekelvin, 2021-11-22:
https://github.com/espressif/esp-idf/blob/a18cb7090511d235ebe3c8b833180630f4f8bb33/components/lwip/Kconfig#L1060

godofwars1111, 2021-11-22:
@negativekelvin this config seem to be not available on the branch:  release/v4.2

negativekelvin, 2021-11-23:
https://github.com/espressif/esp-idf/blob/ccef7c17120ec8944119e0b05c44e9bc040f5044/components/lwip/port/esp32/include/lwipopts.h#L901-LL917

godofwars1111, 2021-11-24:
dhcp_coarse_tmr()
D (43130805) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdcdc
I (43130805) wifi: ~~~~~~~~~~~
I (43130805) wifi: IP:192.168.30.158
I (43130805) wifi: MASK:255.255.255.0
I (43130805) wifi: GW:192.168.30.1
I (43130815) wifi: ~~~~~~~~~~~
dhcp_coarse_tmr()
dhcp_coarse_tmr()
D (43132815) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdcdc
I (43132815) wifi: ~~~~~~~~~~~
I (43132815) wifi: IP:192.168.30.158
I (43132815) wifi: MASK:255.255.255.0
I (43132815) wifi: GW:192.168.30.1
I (43132825) wifi: ~~~~~~~~~~~
dhcp_coarse_tmr()
dhcp_coarse_tmr(): t1 timeout
dhcp_t1_timeout()
dhcp_t1_timeout(): must renew
dhcp_renew()
dhcp_renew(): dhcp state is RENEWING
transaction id xid(325318f7)
dhcp_renew: RENEWING
dhcp_renew(): set request timeout 2000 msecs
dhcp_recv(pbuf = 0x3ffb84c8) from DHCP server 192.168.30.1 port 67
pbuf->len = 300
pbuf->tot_len = 300
skipping option 56 in options
searching DHCP_OPTION_MESSAGE_TYPE
DHCP_NAK received
dhcp_handle_nak(netif=0x3ffbdd60) st1
dhcp_handle_nak(): dhcp state is BACKING_OFF
D (43133565) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffbdd60
D (43133575) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbdcdc
D (43133585) esp_netif_lwip: if0x3ffbdcdc start ip lost tmr: interval=120
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(b362aa1f)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 500 msecs
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(b362aa1f)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 1000 msecs
dhcp_coarse_tmr()
dhcp_recv(pbuf = 0x3ffb84c8) from DHCP server 192.168.30.1 port 67
pbuf->len = 300
pbuf->tot_len = 300
searching DHCP_OPTION_MESSAGE_TYPE
DHCP_OFFER received in DHCP_STATE_SELECTING state
dhcp_handle_offer(netif=0x3ffbdd60) st1
dhcp_handle_offer(): server 0x011ea8c0
dhcp_handle_offer(): offer for 0x9e1ea8c0
dhcp_select(netif=0x3ffbdd60) st1
dhcp_select(): dhcp state is REQUESTING
transaction id xid(b362aa1f)
dhcp_select: REQUESTING
dhcp_select(): set request timeout 2000 msecs
dhcp_recv(pbuf = 0x3ffb84c8) from DHCP server 192.168.30.1 port 67
pbuf->len = 300
pbuf->tot_len = 300
searching DHCP_OPTION_MESSAGE_TYPE
DHCP_ACK received
dhcp_check(netif=0x3ffbdd60) st
dhcp_check(): dhcp state is CHECKING
dhcp_check(): set request timeout 500 msecs
D (43134825) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdcdc
I (43134825) wifi: ~~~~~~~~~~~
I (43134825) wifi: IP:0.0.0.0
I (43134825) wifi: MASK:0.0.0.0
I (43134825) wifi: GW:0.0.0.0
I (43134835) wifi: ~~~~~~~~~~~
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): CHECKING, ARP request timed out
dhcp_check(netif=0x3ffbdd60) st
dhcp_check(): dhcp state is CHECKING
dhcp_check(): set request timeout 500 msecs
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): CHECKING, ARP request timed out
dhcp_bind(netif=0x3ffbdd60) st1
dhcp_bind(): t0 renewal timer 86400 secs
dhcp_bind(): set request timeout 86400000 msecs
dhcp_bind(): t1 renewal timer 43200 secs
dhcp_bind(): set request timeout 43200000 msecs
dhcp_bind(): t2 rebind timer 75600 secs
dhcp_bind(): set request timeout 75600000 msecs
dhcp_bind(): IP: 0x9e1ea8c0 SN: 0x00ffffff GW: 0x011ea8c0
dhcp_bind(): dhcp state is BOUND
D (43135525) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffbdd60
D (43135535) esp_netif_lwip: if0x3ffbdcdc ip unchanged
dhcp_coarse_tmr()
D (43136835) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdcdc
I (43136835) wifi: ~~~~~~~~~~~
I (43136835) wifi: IP:192.168.30.158
I (43136835) wifi: MASK:255.255.255.0
I (43136835) wifi: GW:192.168.30.1
I (43136845) wifi: ~~~~~~~~~~~
dhcp_coarse_tmr()
dhcp_coarse_tmr()
D (43138845) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdcdc
I (43138845) wifi: ~~~~~~~~~~~
I (43138845) wifi: IP:192.168.30.158
I (43138845) wifi: MASK:255.255.255.0
I (43138845) wifi: GW:192.168.30.1
I (43138855) wifi: ~~~~~~~~~~~

[teraterm-2021-11-23-with-dhcp-debug.txt](https://github.com/espressif/esp-idf/files/7596559/teraterm-2021-11-23-with-dhcp-debug.txt)

godofwars1111, 2021-11-24:
we test the exemple with esp 4.1.1 and we don't get this issue

godofwars1111, 2021-11-30:
@negativekelvin were you able to reproduce the issue ?

negativekelvin, 2021-11-30:
No, someone from espressif will be assigned.

You should compare dhcp packets to see cause of `DHCP_NAK received` in 4.2

andrelac-umano, 2021-12-03:
@negativekelvin   
a) I will try to setup wireshark to sniff DHCP packets.
b) Do you know what is the purpose of this timer ''dhcp_coarse_tmr(): t1 timeout''  is this timer set with a value coming from the DHCP server ?

@other
c) did anyone has reproduced this issue at espressif ?

andrelac-umano, 2021-12-06:
Following the info above (and more research on DHCP on my side) I have requested my IT department to increase the DHCP timer from 24 hours to 1 week and did not get any reset for more than 48 hours of testing.  So I'm guessing the esp lib V4.1.1 did not handle the timer the same way that the V4.2 does.

@negativekelvin    Thank you.

andrelac-umano, 2021-12-08:
It did fail after 3.5 days as shown in the debug output below
I will reduce the timer and sniff the DHCP packets as recommended.

**Does anyone from Espressif is looking at this issue ?  For us losing the connection and reconnecting is not very good because of our configuration.**

<<<<
I (302133555) wifi: ~~~~~~~~~~~
dhcp_coarse_tmr()
dhcp_coarse_tmr()
D (302135555) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdcdc
I (302135555) wifi: ~~~~~~~~~~~
I (302135555) wifi: IP:192.168.30.158
I (302135555) wifi: MASK:255.255.255.0
I (302135555) wifi: GW:192.168.30.1
I (302135565) wifi: ~~~~~~~~~~~
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr(): t1 timeout
dhcp_t1_timeout()
dhcp_t1_timeout(): must renew
dhcp_renew()
dhcp_renew(): dhcp state is RENEWING
transaction id xid(b83c0b61)
dhcp_renew: RENEWING
dhcp_renew(): set request timeout 2000 msecs
dhcp_recv(pbuf = 0x3ffb8a68) from DHCP server 192.168.30.1 port 67
pbuf->len = 300
pbuf->tot_len = 300
skipping option 56 in options
searching DHCP_OPTION_MESSAGE_TYPE
DHCP_NAK received
dhcp_handle_nak(netif=0x3ffbdd60) st1
dhcp_handle_nak(): dhcp state is BACKING_OFF
D (302137535) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffbdd60
D (302137535) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbdcdc
D (302137545) esp_netif_lwip: if0x3ffbdcdc start ip lost tmr: interval=120
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(c1a1028d)
D (302137565) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdcdc
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
I (302137575) wifi: ~~~~~~~~~~~
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
I (302137585) wifi: IP:0.0.0.0
dhcp_discover(): set request timeout 500 msecs
I (302137595) wifi: MASK:0.0.0.0
I (302137595) wifi: GW:0.0.0.0
I (302137605) wifi: ~~~~~~~~~~~
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(c1a1028d)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 1000 msecs
dhcp_coarse_tmr()
dhcp_recv(pbuf = 0x3ffbbc84) from DHCP server 192.168.30.1 port 67
pbuf->len = 300
pbuf->tot_len = 300
searching DHCP_OPTION_MESSAGE_TYPE
DHCP_OFFER received in DHCP_STATE_SELECTING state
dhcp_handle_offer(netif=0x3ffbdd60) st1
dhcp_handle_offer(): server 0x011ea8c0
dhcp_handle_offer(): offer for 0x9e1ea8c0
dhcp_select(netif=0x3ffbdd60) st1
dhcp_select(): dhcp state is REQUESTING
transaction id xid(c1a1028d)
dhcp_select: REQUESTING
dhcp_select(): set request timeout 2000 msecs
dhcp_recv(pbuf = 0x3ffbbc84) from DHCP server 192.168.30.1 port 67
pbuf->len = 300
pbuf->tot_len = 300
searching DHCP_OPTION_MESSAGE_TYPE
DHCP_ACK received
dhcp_check(netif=0x3ffbdd60) st
dhcp_check(): dhcp state is CHECKING
dhcp_check(): set request timeout 500 msecs
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): CHECKING, ARP request timed out
dhcp_check(netif=0x3ffbdd60) st
dhcp_check(): dhcp state is CHECKING
dhcp_check(): set request timeout 500 msecs
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): CHECKING, ARP request timed out
dhcp_bind(netif=0x3ffbdd60) st1
dhcp_bind(): t0 renewal timer 604800 secs
dhcp_bind(): set request timeout 604800000 msecs
dhcp_bind(): t1 renewal timer 302400 secs
dhcp_bind(): set request timeout 302400000 msecs
dhcp_bind(): t2 rebind timer 529200 secs
dhcp_bind(): set request timeout 529200000 msecs
dhcp_bind(): IP: 0x9e1ea8c0 SN: 0x00ffffff GW: 0x011ea8c0
dhcp_bind(): dhcp state is BOUND
D (302139525) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffbdd60
D (302139535) esp_netif_lwip: if0x3ffbdcdc ip unchanged
D (302139605) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdcdc
I (302139605) wifi: ~~~~~~~~~~~
I (302139605) wifi: IP:192.168.30.158
I (302139605) wifi: MASK:255.255.255.0
I (302139605) wifi: GW:192.168.30.1
I (302139615) wifi: ~~~~~~~~~~~
>>>

sagb2015, 2021-12-09:
@andrelac-umano- Apologies for late reply! Yes we were trying to reproduce the issue, but could not. Prima-facie, it does not seem to related to WPA2-enterprise. In addition to DHCP, please turn on the supplicant logs. Also, it will help to have wireless sniffer capture.

david-cermak, 2021-12-09:
Hi @andrelac-umano 

Could you please try to cherry-pick this lwip commit? https://github.com/espressif/esp-lwip/commit/2195f7416fb3136831babf3e96c027a73075bd4f

I can see that the [IDF version](https://github.com/espressif/esp-idf/tree/f370d5089f61ac39f183109b6b2908700cfe3b0a/components/lwip) you use references lwip in https://github.com/espressif/esp-lwip/commits/2c9c531f0a7e0ee536db9de4f9dc54e453712087 that may still cause issues with renewal of DHCP leases (as reported in https://github.com/espressif/esp-lwip/issues/30)

andrelac-umano, 2021-12-13:
Hi @david-cermak,   I've tested with the code you provided (above) and it solve the issues (there is no reset of IP).   I ran a test for more than 60 hours and I did previously make sure the system show the error before testing with the modified code.

Following are the setting of DHCP timer for the test (extracted from the test recording):
dhcp_bind(netif=0x3ffbdd60) st1
dhcp_bind(): t0 renewal timer 86129 secs
dhcp_bind(): set request timeout 86129000 msecs
dhcp_bind(): t1 renewal timer 43064 secs
dhcp_bind(): set request timeout 43064000 msecs
dhcp_bind(): t2 rebind timer 75362 secs
dhcp_bind(): set request timeout 75362000 msecs
dhcp_bind(): IP: 0x9e1ea8c0 SN: 0x00ffffff GW: 0x011ea8c0
dhcp_bind(): dhcp state is BOUND

---
D (243642495) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdcdc
I (243642495) wifi: ~~~~~~~~~~~
I (243642495) wifi: IP:192.168.30.158
I (243642495) wifi: MASK:255.255.255.0
I (243642505) wifi: GW:192.168.30.1
I (243642505) wifi: ~~~~~~~~~~~
dhcp_coarse_tmr()
dhcp_coarse_tmr()
D (243644505) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbdcdc
I (243644505) wifi: ~~~~~~~~~~~
I (243644505) wifi: IP:192.168.30.158
I (243644505) wifi: MASK:255.255.255.0
I (243644505) wifi: GW:192.168.30.1
I (243644515) wifi: ~~~~~~~~~~~

david-cermak, 2021-12-15:
@andrelac-umano Thank you for testing this and sorry for the inconvenience.
We will close this issue once the fixing commit (I mentioned above) is referenced from `release/v4.2` branch.

AxelLin, 2022-02-13:
> @andrelac-umano Thank you for testing this and sorry for the inconvenience. We will close this issue once the fixing commit (I mentioned above) is referenced from `release/v4.2` branch.

Fix for v4.2: 0eb18951be5c35369661193cef53d4c464f8ab5d

closed 2022-03-04 by Alvin1Zhang