espressif/esp-idf #7913: ESP-NOW security discussion

Razi91, 2021-11-15:
I found an issue while playing with ESP-NOW that wasn't mentioned anywhere. AFAIR it uses 'WiFi' to communicate, to broadcast messages that should not interfere with WiFi. But I found it interfering, quite seriously with my trackball Elecom HUGE, which uses 2.4GHz USB dongle, probably some custom protocol (it's not BT, or at least it doesn't broadcast BSID).

When I start broadcasting messages via ESP-NOW, especially (does it matter?) to a device that is not turned on, my cursor starts stuttering, like it was losing some packages. ESP (ESP32-C3) was powered by a power bank, so it had no connection with my USB BUS. Anyway, a wired mouse connected to the same HUB works fine.

Probably ESP-NOW wasn't designed to send 150 messages per second, but I'm seriously worried about that issue, not wanting to have anyone around having a problem with using their wireless devices. 

I'm 100% sure it's the ESP causing this problem. 

Code: https://gist.github.com/Razi91/4191c04784870f8ed33e6481ff466f09 . I know it's a mess, I was just trying out the protocol and it's limits. It's hard to show on movie, but I tried: https://www.youtube.com/watch?v=9ECiNnqZ7YA. Sorry for the mess on the desk, I tried to make circular moves, you can see it acting weird sometimes.

ljy770, 2021-11-16:
which channel is ESP-NOW working on ？ The default channel is ch1 if you not set   `esp_wifi_set_channel`. 
Please  change the ESP-NOW working channel and see if your cursor is  affected. 
I tried esp-now with my Logitech MX anywhere 2S and it seemed not be affected.
ESP-NOW  and wireless mouse all work on 2.4G, maybe some mouse be affected by WiFi.

Razi91, 2021-11-18:
I set channels 12, 6, 1, on each, it acts the same. I'm not sure if it's specific problem with Elecom's receiver, but I don't want my device to cause someone trouble. I also test my WiFi connection, it goes down a lot (from 30ms ping 20Mb/s down 14Mb/s up to 300ms ping, 3Mb/s down 6Mb/s up). It really shouldn't be used that way, it works like a WiFi jammer. I'll try to do some more tests with ESP32 (now I tested ESP32-C3)