espressif/esp-idf #7915: TWAI receive overrun when twai_receive() is called with 0 ticks_to_wait

kiranbhosale2, 2021-11-16:
## Environment

- Development Kit:  ESP32-Wrover-Kit
- Kit version WroverKit
- Module or chip used: ESP32-WROVER-E
- IDF version  - v4.3.1-3-gc5f29dd524
- Build System:    Make
- Compiler version  xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System: Linux
- Using an IDE?: Yes VSCODE
- Power Supply:  USB

## Problem Description
TWAI receive overrun when twai_receive() is called with 0 ticks_to_wait

//Detailed problem description goes here.
We have been trying run the TWAI driver for UDS based OTA download configuration, we have our own wrapper functions around twai APIs.   we use the twai_receive() inside these wrapper like below 

`twai_receive(&rx_msg, 0U);`

we carried out below tests 
1. Create the separate task and receive can messages in loop. Out of 50K messages, there was overrun of 500 messages 
we printed the statistics every 1 second
## Debug Logs
[20:07:00:889] can_msg_receive_count  = 50304 ␍␊
[20:07:00:889] status_info->tx_error_counter= 0 ␍␊
[20:07:00:889] status_info.rx_error_counter,= 0 ␍␊
[20:07:00:889] status_info->msgs_to_tx= 0 ␍␊
[20:07:00:889] status_info->msgs_to_rx= 0 ␍␊
[20:07:00:905] status_info->tx_failed_count= 0 ␍␊
[20:07:00:905] status_info->tx_failed_count= 0 ␍␊
[20:07:00:905] status_info->rx_missed_count= 0 ␍␊
**[20:07:00:905] status_info->rx_overrun_count= 585 ␍␊**
[20:07:00:905] status_info->arb_lost_count= 0 ␍␊
[20:07:00:905] status_info->bus_error_count= 0 ␍␊
[20:07:00:921] status_info->state= 1 ␍␊

2. Changing the `twai_receive(&rx_msg, 0U);`  from 

```cpp
twai_receive(&rx_msg, 0U); 
```
even 
```cpp
twai_receive(&rx_msg, 10U); 
```
and 
```cpp
twai_receive(&rx_msg, 10U); 
``` 
**There was no single overrun of the messages** 
## Debug Logs
[20:12:32:058] can_msg_receive_count  = 124708 ␍␊
[20:12:32:058] status_info->tx_error_counter= 0 ␍␊
[20:12:32:074] status_info.rx_error_counter,= 0 ␍␊
[20:12:32:074] status_info->msgs_to_tx= 0 ␍␊
[20:12:32:074] status_info->msgs_to_rx= 0 ␍␊
[20:12:32:074] status_info->tx_failed_count= 0 ␍␊
[20:12:32:074] status_info->tx_failed_count= 0 ␍␊
[20:12:32:074] status_info->rx_missed_count= 0 ␍␊
[20:12:32:090] status_info->rx_overrun_count= 0 ␍␊
[20:12:32:090] status_info->arb_lost_count= 0 ␍␊
[20:12:32:090] status_info->bus_error_count= 0 ␍␊
[20:12:32:090] status_info->state= 1 ␍␊

**please help us understand/resolve this issue**
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/7547088/sdkconfig.zip)

Dazza0, 2021-11-17:
It means the TWAI peripheral's RX FIFO being filled up, which means the TWAI driver's ISR is not responding quick enough to clear the RX FIFO (usually because its being disabled for extended periods of time). Are you running OTA at the same time?

kiranbhosale2, 2021-11-17:
Thanks a lot @Dazza0  for qucik reply. this issue was observed during TWAI testing, where we just kept sending 100K  messages using PCAN and there was an overrun, did you have any observation like this before, so what can we do to make it work under polling and not blocking ?

Dazza0, 2021-11-17:
@kiranbhosale2 Might depend on what you're doing in the task calling `twai_receive()`. Again, are interrupts disabled at any point?

kiranbhosale2, 2021-11-17:
@Dazza0 no the interrupts are not disabled at all, also i tried to capture the ISR execution time specifically for function `twai_handle_rx_buffer_frames()` by toggling the GPIOs and the ISR execution time is in 5ms range in the worst case attached the timing 
[image]

kiranbhosale2, 2021-11-18:
an Updated Observation 
e calculated the ISR execution time and ISR hit time (time between consecutive ISR triggers) by toggling the GPIOs. Below were the observations.

**1. ISR execution time (was fairly consistent)**
[image]

**2. When twai_receive(&rx_msg, 0U); is called without blocking**
  2.1 the ISR is scheduled after approximately 40 ms and that's when the overrun occurs
[image]

3.When **twai_receive()** is called with portMAX_DELAY `ticks_to_wait`
 3.1 All the messages were received properly 

```cpp 
twai_receive(&rx_msg, portMAX_DELAY);
```
[image]

kiranbhosale2, 2021-11-18:
@Dazza0  can you please help us understand because we are blocked by this

Dazza0, 2021-11-23:
@kiranbhosale2 A 40 ms latency of the ISR is definitely not normal (usually it's only a couple of microseconds if there are no other pending interrupts) so something is definitely affect its latency. OTA was not running during the CAN test correct?

kiranbhosale2, 2021-11-23:
@Dazza0 we successfully resolved this issue. the problem was that we were not kicking the watchdog in and wdt_isr was called which had lot of prints, resulting in 40ms delay. 
thanks for your inputs

Dazza0, 2021-11-23:
@kiranbhosale2  Ahh that's good to hear. I'm closing this issue then, Please feel free to reopen/create a new ticket if you face any more issues.

closed 2021-11-23 by Dazza0