espressif/esp-idf #7916: Memory not being released by the http client

sagar448, 2021-11-16:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32E|]
- IDF version (run ``git describe --tags`` to find it):  v3.3.5
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- (Windows only) environment type: [Plain Command Prompt].
- Power Supply:         [USB]

## Problem Description

As described by the http client documentation, the esp_http_client_cleanup is meant to release any memory. But it does not.
I ran the same function 50 times, and I always lost memory.

### Expected Behavior
Free up memory
### Actual Behavior
Losing about 88b every run

### Code to reproduce this issue

```cpp
void meshUpgrade(MeshStruct_t *structRootWrite)
{
    static uint8_t ubycount = 0;
    mdf_err_t ret = MDF_OK;
    uint8_t *data = MDF_MALLOC(MWIFI_PAYLOAD_LEN);
    char name[32] = {0x0};
    size_t total_size = 0;
    int start_time = 0;
    mupgrade_result_t upgrade_result = {0};
    //ESP_LOGI(TAG, "This is the string %s", structRootWrite->cReceivedData);
    esp_http_client_config_t config = {
        .url = structRootWrite->cReceivedData,
        .transport_type = HTTP_TRANSPORT_UNKNOWN,
    };
    //connect to  server
    esp_http_client_handle_t client = esp_http_client_init(&config);
    MDF_ERROR_GOTO(!client, EXIT, "Initialise HTTP connection");
    start_time = xTaskGetTickCount();
    ESP_LOGI(TAG, "Open HTTPS connection: %s", structRootWrite->cReceivedData);

    //establish connection and download the updated firmware
    do
    {
        ret = esp_http_client_open(client, 0);
        if (client == NULL)
        {
            ESP_LOGE(TAG, "Failed to initialise HTTP connection");
            goto EXIT;
        }
        if (ret != MDF_OK)
        {
            MDF_LOGW("<%s> Connection service failed", mdf_err_to_name(ret));
            // Report - connection fail
            if (ubycount == 0)
            {
                // UpdateReportCode(0, AWS_HTTP_CONNECTION_FAILED_WDF);
            }
            ubycount++;
            if (ubycount == 5)
            {
                goto EXIT;
            }
        }
    } while (ret != MDF_OK);
    ESP_LOGI(TAG, "Connected with Server");

    total_size = esp_http_client_fetch_headers(client);
    sscanf(structRootWrite->cReceivedData, "%*[^//]//%*[^/]/%[^.bin]", name);

EXIT:
    MDF_FREE(data);
    mupgrade_result_free(&upgrade_result);
    esp_http_client_close(client);
    esp_http_client_cleanup(client);
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

Even though im using ESP-MDF, the function there are ESP-IDF functions. Related to the issue https://github.com/espressif/esp-mdf/issues/251

ESP-YJM, 2021-11-18:
@sagar448 When you call `esp_http_client_close`, the lower layer will `close` socket. The socket will enter TIME-WAIT status and the memory will not be recycled immediately, maybe need wait for about 1-2 minutes. You can do a test, create a task to print the memory period to check whether the memory be recycled.

sagar448, 2021-11-19:
@ESP-YJM Thanks for the reply. I waited almost 5 minutes, the memory was not released. I did read the whole waiting thing before and so I did test it before I put up an issue. Unfortunately memory is not coming back. If someone could run this code and verify for themselves, that would be great!

sagar448, 2021-11-22:
Anyone else that can help with this?

ESP-YJM, 2021-11-22:
@sagar448 I think memory is occupied in here https://github.com/espressif/esp-lwip/blob/2acf959a2bb559313cd2bf9306c24612ba3d0e19/src/api/sockets.c#L597,

jiaxieya, 2021-12-02:
> @sagar448 When you call `esp_http_client_close`, the lower layer will `close` socket. The socket will enter TIME-WAIT status and the memory will not be recycled immediately, maybe need wait for [...]
If I enable the websocket function and connect to a certain server, will it be affected?

ESP-YJM, 2021-12-02:
@jiaxieya If you send FIN to peer and peer response the FIN or RST, it will no affected.

sagar448, 2021-12-02:
@jiaxieya I waited for 10 minutes, no memory released. Its still the same. There's definitely an issue I don't understand why no one has solved this from Espressif. Maybe its a problem on IDF 3.3.5? Or has it been patched?

ESP-YJM, 2021-12-03:
@sagar448 The memory will not be released. This is a design flaw below IDF v4.x in LWIP module. If you use the IDF v4.x version, the issue will disappear.

sagar448, 2021-12-03:
@ESP-YJM Is this going to be fixed in 3.3? We cannot move to a higher version as the production version for the ESP-MDF use 3.3.5. Please let me know a work around if possible. Much appreciated!

ESP-YJM, 2021-12-09:
@sagar448 No, this memory occuiped fix other serious problems and i think this memory is very small and can be ignored. For example, if you configured the max socket fd number to 10, it most occupied about 880 bytes.

Alvin1Zhang, 2021-12-27:
@sagar448 Thanks for reporting, would you please help share if any more updates for the issue? Thanks.

Alvin1Zhang, 2021-12-31:
Thanks for reporting and sharing the updates, will close due to short of feedback, feel free to reopen with more details.

Jishnujang, 2023-03-03:
I am using version IDF 4.4.1 still I got the memory issues when post request executed. I used esp_get_free_heap_size() to analyse the free memory after posting. if anyone solved the issue please help me to find the solution

GerryBriggs, 2024-08-23:
I am having this problem. The heap is becoming fragmented and appears to be a problem with http_client. Has there been a fix?  For me I am seeing more like 64 bytes (perhaps this is on a per socket basis)

ESP-YJM, 2024-08-23:
@GerryBriggs Which IDF version you used?

ESP-YJM, 2024-08-23:
> I am using version IDF 4.4.1 still I got the memory issues when post request executed. I used esp_get_free_heap_size() to analyse the free memory after posting. if anyone solved the issue please help me to find the solution

Do you  follow this https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks to debug memory leak? I think the issue is like this comments https://github.com/espressif/esp-idf/issues/7916#issuecomment-972501412

GerryBriggs, 2024-08-23:
> > I am using version IDF 4.4.1 still I got the memory issues when post request executed. I used esp_get_free_heap_size() to analyse the free memory after posting. if anyone solved the issue please help me to find the solution
> 
> Do you follow this https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks to debug memory leak? I think the issue is like this comments [#7916 (comment)](https://github.com/espressif/esp-idf/issues/7916#issuecomment-972501412)

I'm using IDF-5.1   ... I am doing a esp_http_client_cleanup() after closing the http handle. Is it possible that there is some TLS artifact getting stuck after resources are given back to the heap?  I was using https links

ESP-YJM, 2024-08-23:
> > > I am using version IDF 4.4.1 still I got the memory issues when post request executed. I used esp_get_free_heap_size() to analyse the free memory after posting. if anyone solved the issue please help me to find the solution
> > 
> > 
> > Do you follow this https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks to debug memory leak? I think the issue is like this comments [#7916 (comment)](https://github.com/espressif/esp-idf/issues/7916#issuecomment-972501412)
> 
> I'm using IDF-5.1 ... I am doing a esp_http_client_cleanup() after closing the http handle. Is it possible that there is some TLS artifact getting stuck after resources are given back to the heap? I was using https links
> 
> EDIT - no I am also getting the problem for http not just https. So both secure and non-secure

You can try set socket SO_LINGER option and when the socket close, it will free the memory immediately。

GerryBriggs, 2024-08-23:
OK I will try your suggestion of SO_LINGER changes.  It is possible sockets are just staying open. I noticed that with some httpd sockets as well, even though they should have been closed.  Here is a visual representation of my heap fragmentation from http_client __
[image]

GerryBriggs, 2024-08-23:
All of my sockets were closed. I wrote a piece of code which lists any open active or listening sockets. They are closed

Here is the code for listing all sockets, which I call directly from tcp.c - it works and I can see that the sockets are closed

void list_existing_TCP_sockets(){
  struct tcp_pcb_listen *lpcb;
  for (lpcb = tcp_listen_pcbs.listen_pcbs; lpcb != NULL; lpcb = lpcb->next) {
    ESP_LOGE("TCP","listen on local port: %"U16_F" with state: %s ", lpcb->local_port, tcp_state_str[(lpcb->state)%11]);
  }
  struct tcp_pcb *apcb;
  for (apcb = tcp_active_pcbs; apcb != NULL; apcb = apcb->next) {
    ESP_LOGE("TCP","active IP: %s on port: %"U16_F" with state: %d and snd_nxt: %"U32_F" and rcv_nxt: %"U32_F" ", ip_ntoa(&apcb->remote_ip), apcb->local_port, apcb->state, apcb->snd_nxt, apcb->rcv_nxt);
  }
  for (apcb = tcp_tw_pcbs; apcb != NULL; apcb = apcb->next) {
    ESP_LOGE("TCP","time-wait IP: %s on port: %"U16_F" with state: %d and snd_nxt: %"U32_F" and rcv_nxt: %"U32_F" ", ip_ntoa(&apcb->remote_ip), apcb->local_port, apcb->state, apcb->snd_nxt, apcb->rcv_nxt);
  }
  for (apcb = tcp_bound_pcbs; apcb != NULL; apcb = apcb->next) {
    ESP_LOGE("TCP","bound IP: %s on port: %"U16_F" with state: %d ", ip_ntoa(&apcb->remote_ip), apcb->local_port, apcb->state);
  }
}

ESP-YJM, 2024-08-23:
@GerryBriggs ` it works and I can see that the sockets are closed`---> Do you mean the issue solved?

GerryBriggs, 2024-08-23:
> @GerryBriggs ` it works and I can see that the sockets are closed`---> Do you mean the issue solved?

No, unfortunately. Not resolved. The 64bytes and another 64bytes are still there, even after all sockets are closed. They make the heap badly fragmented.

ESP-YJM, 2024-08-26:
@GerryBriggs You can use this method to debug who allocate this memory. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks

thaddeus35, 2024-08-30:
@GerryBriggs In menuconfig, there is a TCP option named "Maximum segment lifetime (MSL)" (LWIP_TCP_MSL), which defaults to 60000 ms. Try changing this to 0.

GerryBriggs, 2024-08-30:
> @GerryBriggs In menuconfig, there is a TCP option named "Maximum segment lifetime (MSL)" (LWIP_TCP_MSL), which defaults to 60000 ms. Try changing this to 0.

I have been using the following for the last several months, but I could go even lower, as you suggest:

CONFIG_LWIP_TCP_MSL=4000
CONFIG_LWIP_TCP_FIN_WAIT_TIMEOUT=1000
CONFIG_LWIP_TCP_TMR_INTERVAL=50

GerryBriggs, 2024-09-09:
Sadly after a week of hacking away at this I still cannot determine what is creating these 84-byte blocks that fragment the heap.  I've been through the TCP IP stack on the internal side and marked up every place where some kind of alloc (malloc / calloc / memp_alloc or whatever it was ... realloc) then I marked up the ESP_WIFI to show me when the heap was being called for wifi. Then also TLS (security) and the TRANSPORT stack (netconn, sockets, etc).  Then I went into TLSF (where the mallocs call to) and created a little buffer to watch anything that was between about 80 and 88 bytes and mark down the precise location in memory and the time from startup (in microseconds) to maybe give me a clue. Attached you can see the awful thing I am looking at now to try to track down this fragmentation (I dont even know if these 84 byte stragglers are used after I close the HTTP CLIENT and do cleanup). There is a 120 byte fragment which is in fact created by a WiFi Zalloc, but I dont care about that one as much. The 84 bytes is a problem because they seem to spawn more and more of them as I open then close ESP HTTP CLIENTS in my audio streaming app that I am building (ADF has a convenient heap tracker which I used - and no it isnt being created by the ADF).

[mem_dump_showing_wifi_allocsNaudio5.txt](https://github.com/user-attachments/files/16937495/mem_dump_showing_wifi_allocsNaudio5.txt)

So there is fragmentation from ESP HTTP CLIENT

GerryBriggs, 2024-09-10:
I note that my problem is probably the same as the original poster's (sagar448's) problem. He notices that 88 bytes are missing whereas my problem is 84 bytes. But my 84bytes plus the 4 bytes overhead for that block is also 88 bytes. I am having the same problem 3 years later

GerryBriggs, 2024-09-10:
ESP-YJM --- You mentioned something about setting MAX Socket FD Number to 10 which would mean a maximum of 880 bytes?  Is this 84+4 bytes something that happens for every socket that is opened? Could I trick my app into opening 4 sockets early in memory (I have a maximum of 4 sockets) and then not have the fragmentation problem?

GerryBriggs, 2024-09-10:
I have solved the first problem of the mystery wifi zalloc that leaves a 120 byte-block stuck on the heap after ESP HTTP CLIENT is cleaned up - the way I solve that is reserving 256 bytes in a malloc() at the beginning of the app, then releasing it with free() in the zalloc in adapter.c in ESP_WIFI in ESP32C3.

The bigger problem is the 84 bytes which is not even being picked up by calls to MALLOC, CALLOC, or REALLOC. I am trying to hijack in there in components/newlib/heap.c but somehow it is getting into the heap table without the use of a malloc, calloc, realloc, malloc_r or realloc_r ... which begs the question - how DOES this 84 bytes (88 bytes with overhead) enter into the heap table ?  how?

GerryBriggs, 2024-09-13:
The problem was always in the TLS security certificates. when they are either exchanged or verified, it causes heap fragmentation with 84 bytes two times or more, and something like 12 bytes or 16 bytes.

hyansuper, 2025-03-07:
Any progress on this issue?

I notice not only https client but also http server can cause free heap decrease after every connection

GerryBriggs, 2025-03-07:
For heap problems on the HTTPD (http server) I started doing this to kill the sockets entirely and get my memory back (and also smooth non-defragmented memory:

    httpd_sess_trigger_close(my_httpd_server,httpd_socket_fd[0]);
    httpd_sess_trigger_close(my_httpd_server,httpd_socket_fd[1]);
    httpd_sess_trigger_close(my_httpd_server,httpd_socket_fd[2]);
    httpd_sess_trigger_close(my_httpd_server,httpd_socket_fd[3]);

knowing that I had a maximum of 4 sockets based on my SDK_CONFIG settings 

and each time I had a callback for a handler for a get or post or whetever, I would start them with:

        httpd_socket_fd[httpd_socket_fd_pointer] = httpd_req_to_sockfd((httpd_req_t *)req);
        httpd_socket_fd_pointer=(httpd_socket_fd_pointer+1)%4;

and that httpd_socket_fd_pointer was just a rotating number from 0 to 3 then back to 0, representing the 4 sockets

hyansuper, 2025-03-08:
Thank you!
Do you put httpd_sess_trigger_close in some callback or in a task to periodically check and collect garbage?

GerryBriggs, 2025-03-08:
I put it in a task that should only run when all the HTTPD connections are either closed or not needed. I need to avoid memory fragmentation because I'm running an MP3 live stream receiver on an ESP32C3 and there isn't much RAM so I'm always juggling RAM. That's also why I limit things to 4 sockets since I'm doing high bandwidth HTTPS client and server at the same time

hyansuper, 2025-03-08:
It doesn't fix the problem for me. If I print httpd_req_to_sockfd(req) in the handler function and got the same value every time i refresh the page in browser. 
Does that mean multiple malloc was called, but only one of them gets freed when I call httpd_sess_trigger_close

GerryBriggs, 2025-03-08:
That is possible.  Try doing a heap_caps_dump_all(); both before and after you do the refresh to see exactly WHERE the memory is  being lost, and how much of it

hyansuper, 2025-03-09:
Hi @GerryBriggs , thanks for your help.

while I am still looking for server solution. I found my problem with http client, but it's not the client to blame. 
I am using current master branch, so maybe my case is different than your. Here's my application in case youre interested:

I previously use a timer to periodically start a task, in which a http client poll data from web. and every time after the task ends, I observe 100+ bytes of heap loss. it's probably the frequent task create/delete that cause mem leak/fragment. but I use `xTaskCreateStatic` with static task statck, I am still very confused.

Now, I keep the task alive all the time running a while-delay loop, I even reused the same http_client without clean it up, only call close after every http get. now free heap size is very stable.

GerryBriggs, 2025-03-09:
OK sounds like that worked for you. For me the problem was always in the HTTPS (secure) client not the HTTP client. I believe there is some kind of appendix to a certificate that is not erased for many new HTTPS sessions, possibly up to the total number of sockets but I dont know exactly, even now