espressif/esp-idf #7917: Undefined Reference in main.o

JsMailhot, 2021-11-16:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [No Clue]
- Module or chip used:  [No Clue]
- IDF version (run ``git describe --tags`` to find it) [not a git directory]: 4.3.1
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it) [invalid command]: espressif32 3.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [PlatformIO].
- Using an IDE?: [Visual Studio Code]
- Power Supply:         [N/A]

## Problem Description

Attempted re-creation of PPPoS utility results in failure to compile due to undefined reference.

[Here is the single line of code causing this issue.](https://github.com/espressif/esp-idf/blob/release/v4.3/examples/protocols/pppos_client/main/pppos_client_main.c#L251)

Here is the error message:
`c:/users/user/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.o:(.literal.app_main+0x40): undefined reference to `_g_esp_netif_netstack_default_ppp'
collect2.exe: error: ld returned 1 exit status`

Changing the single line of code causing this issue to the following allows for full compilation of code:
`esp_netif_config_t cfg;`

### Expected Behavior

Copying 99% of example code (replacing placeholders) compiles with minimal modification (12 total modified lines).

### Actual Behavior

Copying 99% of example code (replacing placeholders) causes compilation error unless further modified.

### Steps to reproduce

1. Copy code from [this file](https://github.com/espressif/esp-idf/blob/release/v4.3/examples/protocols/pppos_client/main/pppos_client_main.c) into main.c
2. Replace _EXAMPLE_ variables and other placeholders with static data (int, string, etc.)
3. Compile project

// If possible, attach a picture of your setup/wiring here.

Don't think I can do that

### Code to reproduce this issue is found [here](https://github.com/espressif/esp-idf/blob/release/v4.3/examples/protocols/pppos_client/main/pppos_client_main.c)

## Debug Logs

```
Compiling .pio\build\esp32dev\src\main.o
Linking .pio\build\esp32dev\firmware.elf
c:/users/user/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.o:(.literal.app_main+0x40): undefined reference to `_g_esp_netif_netstack_default_ppp'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder) [sdkconfig.txt](https://github.com/espressif/esp-idf/files/7550013/sdkconfig.txt)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**) Can't do that
- [ ] coredump (This provides stacks of tasks.) Don't think I can do that

david-cermak, 2021-11-18:
Hi @JsMailhot 

Thanks for reporting this issue.

> Changing the single line of code causing this issue to the following allows for full compilation of code: `esp_netif_config_t cfg;`

So, if I understand correctly this example: https://github.com/espressif/esp-idf/blob/release/v4.3/examples/protocols/pppos_client/main/pppos_client_main.c compiles okay, until you change [this line](https://github.com/espressif/esp-idf/blob/release/v4.3/examples/protocols/pppos_client/main/pppos_client_main.c#L251), correct?
Would you mind sharing that line of code, your modification, that causes this linker problem, so we can reproduce the problem?

Please note that even the default config should reference the `_g_esp_netif_netstack_default_ppp` symbol via:

https://github.com/espressif/esp-idf/blob/98d34e5f6db8904c31bd1e372d9896e4931d7a9b/components/esp_netif/include/esp_netif_defaults.h#L121-L126

https://github.com/espressif/esp-idf/blob/98d34e5f6db8904c31bd1e372d9896e4931d7a9b/components/esp_netif/include/esp_netif_defaults.h#L169

JsMailhot, 2021-11-18:
> So, if I understand correctly this example: https://github.com/espressif/esp-idf/blob/release/v4.3/examples/protocols/pppos_client/main/pppos_client_main.c compiles okay, until you change [this [...]

Reading that back, my wording was a little confusing, let me rephrase. [This](https://github.com/espressif/esp-idf/blob/release/v4.3/examples/protocols/pppos_client/main/pppos_client_main.c#L251) line of code causes a compile error for me. When I run my code, I get the following error message:

`c:/users/user/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.o:(.literal.app_main+0x40): undefined reference to _g_esp_netif_netstack_default_ppp`

I figure that this is the line of code causing the compilation error, because when I change it from this:

`esp_netif_config_t cfg = ESP_NETIF_DEFAULT_PPP();`

To this:

`esp_netif_config_t cfg;`

It compiles! I have not tested the functionality of the upload however, as I suspected there I wouldn't have to change that line to get it to compile. Rather my issue had something to do with the `ESP_NETIF_DEFAULT_PPP();` function. The only other lines of code that I have changed to the [example implementation](https://github.com/espressif/esp-idf/blob/release/v4.3/examples/protocols/pppos_client/main/pppos_client_main.c) are some of the static example variables:

`config.tx_io_num = 3;`
`config.rx_io_num = 2;`
`config.rts_io_num = 1;`
`config.cts_io_num = 0;`
`config.rx_buffer_size = 1024;`
`config.tx_buffer_size = 1024;`
`config.pattern_queue_size = 30;`
`config.event_queue_size = 30;`
`config.event_task_stack_size = 2048;`
`config.event_task_priority = 5;`
`config.line_buffer_size = 512;`

But I don't think these changes have anything to do with the issue.

I did find these definitions for what my compiler is claiming to be an undefined reference, so it is clearly there but it appears as though it is not being seen. I thought it might be a versioning issue, maybe my compiler is trying to use a different version installed, but even after uninstalling other versions and setting my PlatformIO environment to use specifically this one and it still appears to be unable to find it. I am not certain how to proceed further to diagnose the reason for my compiler not seeing this code that I would think to be available to it, or if that is even really what is going on. As I have never come across an issue like this, and I haven't found much success in following other similar issues to undefined references.

Thank you for the feedback and assistance!

david-cermak, 2021-11-19:
Oh, I see, so in fact you're not able to compile the default example with PlatformIO, correct?

Could you please post the linker command that links the application? It's usually the very long, last line that ends with `-o pppos_client.elf`, just before the error message.

This error typically happens when linking multiple components with unclear or mutually related dependency tree. I believe however, that this should never exhibit in IDF build system as the `main.o` is treated as the most top-level component. Understanding how the components are built on PlatformIO would help.

JsMailhot, 2021-11-22:
Thank you for your suggestion, I am going to do more looking into that today. That's my current situation, yes. I have never dealt directly with the linker commands because they are generated automatically, but I was able to collect the following by using the `pio run -v` command in my visual studio terminal.

`Using tempfile C:\Users\user\Documents\GitHub\project\esp-idf\.pio\build\esp32dev\tmp_mzmmbv9.tmp for command line:
xtensa-esp32-elf-gcc -o .pio/build/esp32dev/src/main.o -c -std=gnu99 -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -Wno-old-style-declaration -DPLATFORMIO=50203 -DARDUINO_ESP32_DEV -D_GNU_SOURCE -DIDF_VER=\"4.3.1\" -DESP_PLATFORM -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_pm/include -I.pio/build/esp32dev/config -IC:/Users/user/.platformio/packages/framework-espidf/components/newlib/platform_include -IC:/Users/user/.platformio/packages/framework-espidf/components/freertos/include -IC:/Users/user/.platformio/packages/framework-espidf/components/freertos/port/xtensa/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_hw_support/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_hw_support/port/esp32 -IC:/Users/user/.platformio/packages/framework-espidf/components/heap/include -IC:/Users/user/.platformio/packages/framework-espidf/components/log/include -IC:/Users/user/.platformio/packages/framework-espidf/components/lwip/include/apps -IC:/Users/user/.platformio/packages/framework-espidf/components/lwip/include/apps/sntp -IC:/Users/user/.platformio/packages/framework-espidf/components/lwip/lwip/src/include -IC:/Users/user/.platformio/packages/framework-espidf/components/lwip/port/esp32/include -IC:/Users/user/.platformio/packages/framework-espidf/components/lwip/port/esp32/include/arch -IC:/Users/user/.platformio/packages/framework-espidf/components/soc/include -IC:/Users/user/.platformio/packages/framework-espidf/components/soc/esp32 -IC:/Users/user/.platformio/packages/framework-espidf/components/soc/esp32/include -IC:/Users/user/.platformio/packages/framework-espidf/components/hal/esp32/include -IC:/Users/user/.platformio/packages/framework-espidf/components/hal/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_rom/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_rom/esp32 -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_rom/include/esp32 -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_common/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_system/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp32/include -IC:/Users/user/.platformio/packages/framework-espidf/components/driver/include -IC:/Users/user/.platformio/packages/framework-espidf/components/driver/esp32/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_ringbuf/include -IC:/Users/user/.platformio/packages/framework-espidf/components/efuse/include -IC:/Users/user/.platformio/packages/framework-espidf/components/efuse/esp32/include -IC:/Users/user/.platformio/packages/framework-espidf/components/xtensa/include -IC:/Users/user/.platformio/packages/framework-espidf/components/xtensa/esp32/include -IC:/Users/user/.platformio/packages/framework-espidf/components/espcoredump/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_timer/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_ipc/include -IC:/Users/user/.platformio/packages/framework-espidf/components/vfs/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_wifi/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_wifi/esp32/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_event/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_netif/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_eth/include -IC:/Users/user/.platformio/packages/framework-espidf/components/tcpip_adapter/include -IC:/Users/user/.platformio/packages/framework-espidf/components/app_trace/include -IC:/Users/user/.platformio/packages/framework-espidf/components/mbedtls/port/include -IC:/Users/user/.platformio/packages/framework-espidf/components/mbedtls/mbedtls/include -IC:/Users/user/.platformio/packages/framework-espidf/components/mbedtls/esp_crt_bundle/include -IC:/Users/user/.platformio/packages/framework-espidf/components/app_update/include -IC:/Users/user/.platformio/packages/framework-espidf/components/spi_flash/include -IC:/Users/user/.platformio/packages/framework-espidf/components/bootloader_support/include -IC:/Users/user/.platformio/packages/framework-espidf/components/nvs_flash/include -IC:/Users/user/.platformio/packages/framework-espidf/components/pthread/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_gdbstub/include -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_gdbstub/xtensa -IC:/Users/user/.platformio/packages/framework-espidf/components/esp_gdbstub/esp32 -IC:/Users/user/.platformio/packages/framework-espidf/components/wpa_supplicant/include -IC:/Users/user/.platformi [... cut ...] at there were some circular dependencies with the libraries provided in the example. Specifically the esp_modem_netif and esp_modem files, but this also occurs when I modify that one line as mentioned earlier, and does not prevent compilation. So I do not think that this has anything to do with the issue either.

I will have to do some reading up on the linker command context to understand it better. But I have a feeling that it is the correct direction to be looking for the issue. Thanks again!

david-cermak, 2021-11-22:
Thanks for sharing the command line, yes (part of it) shows the linking phase. There's no `-o pppos_client.elf`, as it's apparently built a bit differently under PlatformIO and the linker output is ` -o .pio\build\esp32dev\firmware.elf ` and (as we suspected) the main component gets also built differently, not as a static library, but linked directly as the object file: `main.o`.

Can you please try to add this line to your main component makefile?
```CMake
target_link_libraries(${COMPONENT_LIB} INTERFACE "-u _g_esp_netif_netstack_default_ppp")
```

just after the `idf_component_register()` in your project's `main/CMakeLists.txt` ?

JsMailhot, 2021-11-22:
Most certainly, Here is what my CMakeLists.txt file looks like:

`set(app_sources`
`	"main.c"`
`)`
`idf_component_register(SRCS ${app_sources})`
`target_link_libraries(${COMPONENT_LIB} INTERFACE "-u _g_esp_netif_netstack_default_ppp")`

After cleaning and compiling again, this is what I get from running the `pio run -v` command:

`xtensa-esp32-elf-gcc -o .pio\build\esp32dev\firmware.elf -T esp32_out.ld -u esp_app_desc -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_include_pthread_local_storage_impl -u ld_include_panic_highint_hdl -u start_app -u start_app_other_cores -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T esp32.rom.newlib-time.ld -u vfs_include_syscalls_impl -T esp32.project.ld -T esp32.peripherals.ld -u call_user_start_cpu0 -u app_main -u newlib_include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u __cxa_guard_dummy -u __cxx_fatal_exception -u _g_esp_netif_netstack_default_ppp -Wl,--Map=C:/Users/user/Documents/GitHub/project/esp-idf/.pio/build/esp32dev/esp-idf.map -Wl,--cref -Wl,--gc-sections -Wl,--undefined=uxTopUsedPriority -Wl,--wrap=_Unwind_Backtrace -Wl,--wrap=_Unwind_DeleteException -Wl,--wrap=_Unwind_FindEnclosingFunction -Wl,--wrap=_Unwind_Find_FDE -Wl,--wrap=_Unwind_ForcedUnwind -Wl,--wrap=_Unwind_GetCFA -Wl,--wrap=_Unwind_GetDataRelBase -Wl,--wrap=_Unwind_GetGR -Wl,--wrap=_Unwind_GetIP -Wl,--wrap=_Unwind_GetIPInfo -Wl,--wrap=_Unwind_GetLanguageSpecificData -Wl,--wrap=_Unwind_GetRegionStart -Wl,--wrap=_Unwind_GetTextRelBase -Wl,--wrap=_Unwind_RaiseException -Wl,--wrap=_Unwind_Resume -Wl,--wrap=_Unwind_Resume_or_Rethrow -Wl,--wrap=_Unwind_SetEnableExceptionFdeSorting -Wl,--wrap=_Unwind_SetGR -Wl,--wrap=_Unwind_SetIP -Wl,--wrap=__cxa_call_unexpected -Wl,--wrap=__deregister_frame_info -Wl,--wrap=__deregister_frame_info_bases -Wl,--wrap=__gxx_personality_v0 -Wl,--wrap=__register_frame -Wl,--wrap=__register_frame_info -Wl,--wrap=__register_frame_info_bases -Wl,--wrap=__register_frame_info_table -Wl,--wrap=__register_frame_info_table_bases -Wl,--wrap=__register_frame_table -Wl,--wrap=longjmp -Wl,--wrap=mbedtls_mpi_exp_mod -Wno-frame-address -fno-lto -fno-rtti -mlongcalls .pio\build\esp32dev\src\main.o .pio\build\esp32dev\.pio\build\esp32dev\ca_cert.pem.o -L.pio\build\esp32dev -LC:\Users\user\.platformio\packages\framework-espidf\components\esp_wifi\lib\esp32 -LC:\Users\user\.platformio\packages\framework-espidf\components\esp_rom\esp32\ld -LC:\Users\user\.platformio\packages\framework-espidf\components\esp_wifi\lib\esp32 -LC:\Users\user\.platformio\packages\framework-espidf\components\xtensa\esp32 -L.pio\build\esp32dev\esp-idf\esp32 -L.pio\build\esp32dev\esp-idf\esp32\ld -LC:\Users\user\.platformio\packages\framework-espidf\components\esp32\ld -Wl,--start-group .pio\build\esp32dev\lib7f7\libesp_modem_netif.a .pio\build\esp32dev\libe33\libesp_modem.a .pio\build\esp32dev\lib187\libbg96.a .pio\build\esp32dev\lib1f2\libi2c_bus.a .pio\build\esp32dev\lib1ad\libhts221.a .pio\build\esp32dev\lib542\liblis2dh12.a .pio\build\esp32dev\lib38c\libLED.a .pio\build\esp32dev\libe4e\libLTE_Tracker.a .pio\build\esp32dev\lib3d3\libsim7600.a .pio\build\esp32dev\libaa6\libsim800.a .pio\build\esp32dev\esp-idf\esp_pm\libesp_pm.a .pio\build\esp32dev\esp-idf\cmock\libcmock.a .pio\build\esp32dev\esp-idf\cbor\libcbor.a .pio\build\esp32dev\esp-idf\asio\libasio.a .pio\build\esp32dev\esp-idf\unity\libunity.a .pio\build\esp32dev\esp-idf\coap\libcoap.a .pio\build\esp32dev\esp-idf\console\libconsole.a .pio\build\esp32dev\esp-idf\esp_hid\libesp_hid.a .pio\build\esp32dev\esp-idf\esp_adc_cal\libesp_adc_cal.a .pio\build\esp32dev\esp-idf\protobuf-c\libprotobuf-c.a .pio\build\esp32dev\esp-idf\mdns\libmdns.a .pio\build\esp32dev\esp-idf\protocomm\libprotocomm.a .pio\build\esp32dev\esp-idf\esp_local_ctrl\libesp_local_ctrl.a .pio\build\esp32dev\esp-idf\esp_websocket_client\libesp_websocket_client.a .pio\build\esp32dev\esp-idf\expat\libexpat.a .pio\build\esp32dev\esp-idf\wear_levelling\libwear_levelling.a .pio\build\esp32dev\esp-idf\freemodbus\libfreemodbus.a .pio\build\esp32dev\esp-idf\jsmn\libjsmn.a .pio\build\esp32dev\esp-idf\fatfs\libfatfs.a .pio\build\esp32dev\esp-idf\json\libjson.a .pio\build\esp32dev\esp-idf\spiffs\libspiffs.a .pio\build\esp32dev\esp-idf\openssl\libopenssl.a .pio\build\esp32dev\esp-idf\libsodium\liblibsodium.a .pio\build\esp32dev\esp-idf\mqtt\libmqtt.a .pio\build\esp32dev\esp-idf\wifi_provisioning\libwifi_provisioning.a .pio\build\esp32dev\esp-idf\app_trace\libapp_trace.a .pio\build\esp32dev\esp-idf\app_update\libapp_update.a .pio\build\esp32dev\esp-idf\bootloader_support\libbootloader_support.a .pio\build\esp32dev\esp-idf\cxx\libcxx.a .pio\build\esp32dev\esp-idf\driver\libdriver.a .pio\build\esp32dev\esp-idf\efuse\libefuse.a .pio\build\esp32dev\esp-idf\esp-tls\libesp-tls.a .pio\build\esp32dev\esp-idf\esp32\libesp32.a .pio\build\esp32dev\esp-idf\esp_common\libesp_common.a .pio\build\esp32dev\esp-idf\esp_eth\libesp_eth.a .pio\build\esp32dev\esp-idf\esp_event\libesp_event.a .pio\build\esp32dev\esp-idf\esp_gdbstub\libesp_gdbstub.a .pio\build\esp32dev\esp-idf\esp_http_client\libesp_http_client.a .pio\build\esp32dev\esp-idf\esp_http_server\libesp_http_server.a .pio\build\esp32dev\esp-idf\esp_https_ota\libesp_https_ota.a 
[... cut ...] \libwpa_supplicant.a .pio\build\esp32dev\esp-idf\xtensa\libxtensa.a .pio\build\esp32dev\esp-idf\mbedtls\mbedtls\library\libmbedcrypto.a .pio\build\esp32dev\esp-idf\mbedtls\mbedtls\library\libmbedtls.a .pio\build\esp32dev\esp-idf\mbedtls\mbedtls\library\libmbedx509.a -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lphy -lrtc -lxt_hal -lc -lm -lstdc++ -lgcc -Wl,--end-group`

I get the exact same error message on compiling however. Thanks again for the help figuring this out.

david-cermak, 2021-11-23:
Thanks again for testing it, I can see that the linker flag is correctly propagated to the command line, but I don't understand why it could not link or find the symbol. I think I would need to know more about the PlatformIO build process...

Thought I would just test how that example compiles, this is what I tried (Apologies if some of the steps below wasn't according to the best practices for PlatformIO)
* Created a new project based on ESP32-pico-kit
* Configure the project (enabled lwip->PPP options -- I can see this part was correct in your project)
* Replaced the main file
    * Copy pasted the IDF example to `PlatformIO/Projects/my_project/src/main.c`
    * Removed all irrelevant `#ifdef`'s
    * Replaced all `CONFIG_...` options to hardcoded values (as you did in your example above)
* Created new library in  `PlatformIO/Projects/my_project/lib` called `modem`
    * Copied the entire modem component from https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client/components/modem
    * Moved private include to lib-includes ([this file](https://github.com/espressif/esp-idf/tree/release/v4.3/examples/protocols/pppos_client/components/modem/private_include) )
    * Replace the modem configuration with the actual values (setting the `APN`)
* Build the project

And surprisingly, everything compiled and linked
```bash
david@Davids-MBP:~/Documents/PlatformIO/Projects/test32$ pio run -v
Processing pico32 (platform: espressif32; board: pico32; framework: espidf)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/pico32.html
PLATFORM: Espressif 32 (3.4.0) > ESP32 Pico Kit
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40301.0 (4.3.1)
...
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <modem> (/Users/david/Documents/PlatformIO/Projects/test32/lib/modem)
Building in release mode
<lambda>(["checkprogsize"], [".pio/build/pico32/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.5% (used 17896 bytes from 327680 bytes)
Flash: [======    ]  58.4% (used 612341 bytes from 1048576 bytes)
...
======================================================================================= [SUCCESS] Took 15.32 seconds =======================================================================================
```

There must be something you have done differently, could you please share the step how you setup the project?

closed 2021-11-23 by JsMailhot