espressif/esp-idf #7920: USB host has difficulty detecting connections

leemagnusson, 2021-11-17:
## Environment

- Module or chip used:  ESP32S2
- IDF version v5.0-dev-346-g7af719e426
- Build System:      idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

I have some difficulty with the esp32s2 usb host being able to detect when a usb device is connected. There are two parts to the problem. First, it sometimes has difficulty detecting the initial connection of a device. Experimentally I found that applying a pull-down before initializing the usb host seems to fix that problem. Second, after the device is disconnected and reconnected it seems even less likely to detect the reconnection (although sometimes it does). My solution for that is to do an `esp_restart` whenever the device is disconnected.

### Code to reproduce this issue

```cpp
   gpio_config_t gpio_conf = {
      .pin_bit_mask = 3 << 19,
      .pull_down_en = GPIO_PULLDOWN_ENABLE
   };
   gpio_config(&gpio_conf);

   usb_host_init(); // usb_host_install, etc.

   while(1) {
      if (usb_device_disconnected())
          esp_restart()
   }
```

chegewara, 2021-11-17:
I dont seem to have this problem. Here is full test code:
```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "usb/usb_host.h"

usb_host_client_handle_t client_hdl;
uint8_t dev_addr = 2; 
usb_device_handle_t dev_hdl;
TaskHandle_t async_task_handle;

void client_event_callback(const usb_host_client_event_msg_t *event_msg, void *arg)
{
    if (event_msg->event == USB_HOST_CLIENT_EVENT_NEW_DEV)
    {
        ESP_LOGI("", "client event: %d, address: %d", event_msg->event, event_msg->new_dev.address);
    } else {
        ESP_LOGI("", "USB_HOST_CLIENT_EVENT_DEV_GONE");
        usb_host_device_close(client_hdl, dev_hdl);
    }
}

void app_main(void)
{
    do{

        const usb_host_config_t config = {
            .intr_flags = ESP_INTR_FLAG_LEVEL1,
        };
        esp_err_t err = usb_host_install(&config);
        ESP_LOGI("", "install status: %d", err);

        const usb_host_client_config_t client_config = {
            .async = {
                .client_event_callback = client_event_callback,
                .callback_arg = NULL,
            },
            .max_num_event_msg = 5
        };

        err = usb_host_client_register(&client_config, &client_hdl);
        ESP_LOGI("", "client register status: %d", err);

        while (1)
        {
            uint32_t event_flags;
            if (ESP_OK == usb_host_lib_handle_events(1, &event_flags))
            {
                printf("usb_host_lib_handle_events event_flags: %d\n", event_flags & 0xff);
                usb_host_client_handle_events(client_hdl, 0);
                if (event_flags & USB_HOST_LIB_EVENT_FLAGS_NO_CLIENTS)
                {
                    printf("No more clients\n");
                    do{
                        if(usb_host_device_free_all() != ESP_ERR_NOT_FINISHED) break;
                    }while(1);
                    break;
                }
                if (event_flags & USB_HOST_LIB_EVENT_FLAGS_ALL_FREE)
                {
                    printf("USB_HOST_LIB_EVENT_FLAGS_ALL_FREE\n");
                    usb_host_client_deregister(client_hdl);
                }
            }
            
        }
        usb_host_uninstall();
    }while(1);
}
```

leemagnusson, 2021-11-17:
My problem may be specific just to certain usb devices. Part of my inspiration for a pull down was here: https://www.beyondlogic.org/usbnutshell/usb2.shtml

Maybe my device expects a pull down in order to get through its initialization.

Dazza0, 2021-11-25:
@leemagnusson The ESP32-S2 internal USB PHY should have already applied those pull downs so there shouldn't be a need to call `gpio_config()` if everything is working correctly. Any chance you get capture the bus on an oscilloscope to give us a better idea about what's happening?

Dazza0, 2022-01-06:
Please try the latest USB Host update (d95b15c55740b417d1a935ac006dba4cfaeef3cf) and see if the issue is resolved?

Alvin1Zhang, 2022-01-24:
@leemagnusson Thanks for reporting, would you please help share if any updates for the issue? Thanks.

Dazza0, 2022-03-21:
@leemagnusson Fixes should be in both master and v4.4. Closing this issue. If the errors persist, please feel free to reopen.

closed 2022-03-21 by Dazza0