espressif/esp-idf #7921: ADC2 gives incorrect reading after light sleep of more than 90 seconds.

nilesh-dryad, 2021-11-17:
## Environment

- Development Kit: ESP32-S2-Kaluga-1
- Kit version (for WroverKit/PicoKit/DevKitC): V1.3
- Module or chip used: ESP32-S2-Wrover
- IDF version: v4.3.1 (Stable)
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 5V-2A

## Problem Description

ADC2 Reading is incorrect after light sleep of more than 90 seconds. I am giving stable 0.55V on channel 3 of ADC2. It gives me raw value of 559 when board is not in light sleep but if board enters into light sleep for more than 90 seconds then after wake up ADC2 channel 3 reading 1490 and then it always remains the same. If I use 60 seconds light sleep then issue does not occur.

### Expected Behavior
It should give correct value as it is without light sleep.

### Actual Behavior
It is giving incorrect value after waking up from light sleep.

### Steps to reproduce

1. Take the sample ADC2 example.
2. GIve 0.55V or any other voltage you want to channel 3 of ADC2.
3. Measure RAW value from ADC2. It should be correct. In my case, it was 559 (after voltage conversion ) which is almost equal to 0.55V.
4. now put esp32s2 into light sleep for more than 90 second.
5. after wake up, again take raw reading. now it is 1490 (after voltage conversion ) , while input voltage at channel is still 0.55V.
6. I also checked the input voltage on oscilloscope, it was always 0.55V.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "driver/adc.h"
#include "esp_adc_cal.h"
#include "esp_sleep.h"
#include "esp_log.h"
#include "driver/uart.h"

#define TAG    "ADC2" 
#define DEFAULT_VREF    1100 
#define NO_OF_SAMPLES   64          //Multisampling

static const adc_channel_t channel = ADC_CHANNEL_3;
static const adc_bits_width_t width = ADC_WIDTH_BIT_13;
static esp_adc_cal_characteristics_t *adc_chars;
static const adc_atten_t atten = ADC_ATTEN_DB_11;
static const adc_unit_t unit = ADC_UNIT_2;

static void check_efuse(void)
{
    if (esp_adc_cal_check_efuse(ESP_ADC_CAL_VAL_EFUSE_TP) == ESP_OK) {
        ESP_LOGI(TAG, "eFuse Two Point: Supported");
    } else {
        ESP_LOGE(TAG, "Cannot retrieve eFuse Two Point calibration values. Default calibration values will be used.");
    }
}

uint32_t drdGetBatVoltage(void)
{
    uint32_t voltage = 0;
    uint32_t adc_reading = 0;
    //Multisampling
    for (int i = 0; i < NO_OF_SAMPLES; i++) {
        int raw;
        adc2_get_raw((adc2_channel_t)channel, width, &raw);
        adc_reading += raw;
    }
    adc_reading /= NO_OF_SAMPLES;
    //Convert adc_reading to voltage in mV
    voltage = esp_adc_cal_raw_to_voltage(adc_reading, adc_chars);
    return voltage;
}

void drdAdc2Init(void)
{
    check_efuse();
    adc2_config_channel_atten((adc2_channel_t)channel, atten);
    vTaskDelay(20);
    adc_chars = heap_caps_calloc(1, sizeof(esp_adc_cal_characteristics_t), MALLOC_CAP_INTERNAL);
	if (adc_chars == NULL) {
		ESP_LOGE(TAG, "ADC INIT: Mem error");
		return;
	}
    esp_adc_cal_characterize(unit, atten, width, DEFAULT_VREF, adc_chars);
}

void app_main()
{
    drdAdc2Init();
    printf( "Voltage: %d\n", drdGetBatVoltage());
    printf( "Voltage: %d\n", drdGetBatVoltage());
    printf( "Voltage: %d\n", drdGetBatVoltage());
    printf( "Voltage: %d\n", drdGetBatVoltage());
    uint32_t batVoltage = 0;
    while (true) {
        batVoltage = drdGetBatVoltage();
        if (batVoltage < 875) {
            esp_sleep_enable_timer_wakeup(180000000); /* 180 seconds */
            printf("Entering light sleep: %d\n", batVoltage);
            uart_wait_tx_idle_polling(0);
            esp_light_sleep_start();
        } else {
            printf("Voltage become high %d\n", batVoltage);
        }
    }
}

```

## Debug Logs

```
Voltage: 559
Voltage: 558
Voltage: 559
Voltage: 559
Entering light sleep: 559
Voltage become high 1490
Voltage become high 1491
Voltage become high 1490
...
```

nilesh-dryad, 2021-11-18:
Note that I am not using WiFi or BLE in my project.

nilesh-dryad, 2021-11-23:
I did some more digging and found that it does not happen on each ESP32S2 wrover. I have 6 Kaluga board. Only two of them is having this issue? Is this something related to calibration?

nilesh-dryad, 2022-01-12:
Any update on this? I am still facing this issue on some of the ESP32S2. It just give increased value. I am using automatic light sleep. Without automatic light sleep, it just works fine.

nilesh-dryad, 2022-01-25:
This issue is now critical for us in production phase. Is there any solutions for this? It is simply not working. Does it mean that we can not use ESP32S2 ADC in light sleep mode?

Icarus113, 2022-01-26:
Hi @nilesh-dryad 
Please do this, in `hal/adc_hal.c`.
```
void adc_hal_set_calibration_param(adc_ll_num_t adc_n, uint32_t param)
{
    // if (param != s_previous_init_code[adc_n]) {
        adc_ll_calibration_init(adc_n);
        adc_ll_set_calibration_param(adc_n, param);
    //     s_previous_init_code[adc_n] = param;
    // }
}
```
This is just a temp fix. Final fix will be merged to Master then v4.3

nilesh-dryad, 2022-01-28:
@Icarus113 Your fix works in my initial test. I will test this with all of my boards which were giving wrong ADC results before. I will also test it with different input voltages

closed 2022-03-07 by espressif-bot