espressif/esp-idf #7923: Errors and audio-outages in example a2dp_source

mihovilkolaric, 2021-11-17:
## Environment

- Development Kit:      [ESP32-LyraT]
- Kit version (for WroverKit/PicoKit/DevKitC): ESP32-LyraT v4.3
- IDF version (run ``git describe --tags`` to find it):
v5.0-dev-346-g7af719e426
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [external 5V]

## Problem Description
Use https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_source , and only modified the name of the bluetooth-sink as seen in this diff:
```

     if (eir) {
         get_name_from_eir(eir, s_peer_bdname, NULL);
-        if (strcmp((char *)s_peer_bdname, "ESP_SPEAKER") != 0) {
+        if (strcmp((char *)s_peer_bdname, "bose") != 0) {
             return;
         }
```
Device connects successfully, but the audio-transmission is very intermittent (more silence than audio - see attached recording [esp32_a2dp_source.mp3.tar.gz](https://github.com/espressif/esp-idf/files/7557768/esp32_a2dp_source.mp3.tar.gz) ), and many lines of the error
`BT_L2CAP: l2cab is_cong_cback_context`
are logged.

### Expected Behavior
Audio-Samples (noise) is heard in headphones as described in example's README.md.

### Actual Behavior
Audio is not useable.

### Steps to reproduce

1. idf.py build flash monitor
2. connect Bluetooth headset _(used a Bose QC 35)_

### Code to reproduce this issue

https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_source

## Debug Logs

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6840
load:0x40078000,len:14836
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v5.0-dev-346-g7af719e426-dirty 2nd stage bootloader
I (27) boot: compile time 20:20:43
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1bb70h (113520) map
I (153) esp_image: segment 1: paddr=0002bb98 vaddr=3ffbdb60 size=04480h ( 17536) load
I (160) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=77660h (489056) map
I (337) esp_image: segment 3: paddr=000a7688 vaddr=3ffc1fe0 size=00320h (   800) load
I (338) esp_image: segment 4: paddr=000a79b0 vaddr=40080000 size=17094h ( 94356) load
I (382) esp_image: segment 5: paddr=000bea4c vaddr=50000000 size=00010h (    16) load
I (393) boot: Loaded app from partition at offset 0x10000
I (393) boot: Disabling RNG early entropy source...
I (405) cpu_start: Pro cpu up.
I (405) cpu_start: Starting app cpu, entry point is 0x40081204
0x40081204: call_start_cpu1 at /home/esp/esp/esp-idf/components/esp_system/port/cpu_start.c:164

I (0) cpu_start: App cpu up.
I (421) cpu_start: Pro cpu start user code
I (421) cpu_start: cpu freq: 160000000
I (421) cpu_start: Application information:
I (426) cpu_start: Project name:     a2dp_source
I (431) cpu_start: App version:      v5.0-dev-346-g7af719e426-dirty
I (438) cpu_start: Compile time:     Nov 17 2021 20:21:48
I (444) cpu_start: ELF file SHA256:  7f130321945038e7...
I (450) cpu_start: ESP-IDF:          v5.0-dev-346-g7af719e426-dirty
I (457) heap_init: Initializing. RAM available for dynamic allocation:
I (464) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (470) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (477) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (483) heap_init: At 3FFC9FD8 len 00016028 (88 KiB): DRAM
I (489) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (495) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (502) heap_init: At 40097094 len 00008F6C (35 KiB): IRAM
I (509) spi_flash: detected chip: generic
I (513) spi_flash: flash io: dio
W (516) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (531) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (623) BTDM_INIT: BT controller compile version [899ab01]
I (623) system_api: Base MAC address is not set
I (623) system_api: read default base MAC address from EFUSE
I (633) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1283) BT_AV: Starting device discovery...
W (1283) BT_BTC: btc_avrc_tg_set_rn_supported_evt failed: AVRC TG not yet initialized

W (1293) BT_BTC: A2DP Enable with AVRC
I (1303) BT_AV: bt_app_av_sm_hdlr state 1, evt 0x4
I (1313) BT_AV: event: 10
I (1313) BT_AV: event: 10
I (1313) BT_AV: event: 10
I (1313) BT_AV: event: 10
I (1313) BT_AV: Discovery started.
I (11283) BT_AV: bt_app_av_sm_hdlr state 1, evt 0xff00
I (14123) BT_AV: Device discovery failed, continue to discover...
I (14133) BT_AV: Discovery started.
I (19413) BT_AV: Scanned device: <anonymized>
I (19413) BT_AV: --Class of Device: 0x240418
I (19413) BT_AV: --RSSI: -35
I (19423) BT_AV: Found a target device, address <anonymized>, name bose
I (19423) BT_AV: Cancel device discovery ...
I (19433) BT_AV: Device discovery stopped.
I (19433) BT_AV: a2dp connecting to peer: bose
W (19443) BT_APPL: reset flags
I (19453) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
I (21283) BT_AV: bt_app_av_sm_hdlr state 4, evt 0xff00
E (21583) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (21883) BT_APPL: new conn_srvc id:18, app_id:0
I (21893) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
I (21893) BT_AV: a2dp connected
E (22253) BT_BTM: tBTM_SEC_DEV:0x3ffd1974 rs_disc_pending=0

I (23853) RCCT: AVRC conn_state evt: state 1, [<anonymized>]
I (23853) RCCT: AVRC remote features 24b, TG features 23
I (23883) RCCT: remote rn_cap: count 1, bitmask 0x2000
E (29093) BT_APPL: bta_dm_pm_btm_status hci_status=36
I (29093) BT_AV: ESP_BT_GAP_MODE_CHG_EVT mode:0
I (31283) BT_AV: bt_app_av_sm_hdlr state 5, evt 0xff00
I (31283) BT_AV: a2dp media ready checking ...
I (31283) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x3
I (31283) BT_AV: a2dp media ready, starting ...
W (31293) BT_APPL: new conn_srvc id:18, app_id:1
I (31343) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x824b

I (31343) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x3
I (31343) BT_AV: a2dp media start successfully.
I (31343) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x1
E (31603) BT_L2CAP: l2cab is_cong_cback_context
E (31723) BT_L2CAP: l2cab is_cong_cback_context
E (31833) BT_L2CAP: l2cab is_cong_cback_context
E (31943) BT_L2CAP: l2cab is_cong_cback_context
E (32073) BT_L2CAP: l2cab is_cong_cback_context
E (32173) BT_L2CAP: l2cab is_cong_cback_context
E (32283) BT_L2CAP: l2cab is_cong_cback_context
E (32403) BT_L2CAP: l2cab is_cong_cback_context
E (32513) BT_L2CAP: l2cab is_cong_cback_context
E (32623) BT_L2CAP: l2cab is_cong_cback_context
E (32743) BT_L2CAP: l2cab is_cong_cback_context
E (32853) BT_L2CAP: l2cab is_cong_cback_context
E (32963) BT_L2CAP: l2cab is_cong_cback_context
E (33093) BT_L2CAP: l2cab is_cong_cback_context
E (33203) BT_L2CAP: l2cab is_cong_cback_context
E (33303) BT_L2CAP: l2cab is_cong_cback_context
E (33433) BT_L2CAP: l2cab is_cong_cback_context
E (33543) BT_L2CAP: l2cab is_cong_cback_context
...

```
and many more, full log attached:
[esp32_a2dp_source.log](https://github.com/espressif/esp-idf/files/7557755/esp32_a2dp_source.log)

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7557759/sdkconfig.txt)
[a2dp_source.elf.tar.gz](https://github.com/espressif/esp-idf/files/7557765/a2dp_source.elf.tar.gz)

Vxs14, 2022-01-28:
Hy,

i have the same issue, with esp-idf v4.4 in exactly the same condition.

did you find a solution ?

mihovilkolaric, 2022-01-28:
no, unfortunately no progress.
Does anyone have ony hints how to debug?

ShaoMingya, 2022-02-08:
我测试JBL Pulse 3这个音箱会这样，其他音箱正常。idf-v4.4.