espressif/esp-idf #7926: ESP32 hangs on "cpu_start: App cpu up."

SMinin, 2021-11-18:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v4.3.1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [Yes] Eclipse 4.20.0
- Power Supply:         [external 3.3V]

## Problem Description

If I set "Stack smashing protection mode: Overal" in sdkconfig, then in some cases the application freezes immediately after starting the firmware at the "cpu_start: App cpu up" stage. I'll use the "blink" sample to reproduce the problem.

### Expected Behavior
Program execution

### Actual Behavior
The program freezes at "I (0) cpu_start: App cpu up."

### Steps to reproduce

1. Compile the application using sdkconfig in the attachment
2. Run application on ESP32
3. Check what is displayed on the UART interface

### Code to reproduce this issue
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

#define BLINK_GPIO 14

void app_main(void)
{
    gpio_reset_pin(BLINK_GPIO);
    gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT);
    while(1) {
        printf("Turning off the LED\n");
        gpio_set_level(BLINK_GPIO, 0);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        printf("Turning on the LED\n");
        gpio_set_level(BLINK_GPIO, 1);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

## Debug Logs
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6964
load:0x40078000,len:14308
load:0x40080400,len:3716
0x40080400: _init at ??:?

entry 0x40080680
I (27) boot: ESP-IDF v4.3.1 2nd stage bootloader
I (27) boot: compile time 10:38:10
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=07798h ( 30616) map
I (120) esp_image: segment 1: paddr=000177c0 vaddr=3ffb0000 size=028f4h ( 10484) load
I (125) esp_image: segment 2: paddr=0001a0bc vaddr=40080000 size=05f5ch ( 24412) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=16430h ( 91184) map
I (172) esp_image: segment 4: paddr=00036458 vaddr=40085f5c size=08178h ( 33144) load
I (186) esp_image: segment 5: paddr=0003e5d8 vaddr=50000000 size=00010h (    16) load
I (193) boot: Loaded app from partition at offset 0x10000
I (193) boot: Disabling RNG early entropy source...
I (206) cpu_start: Pro cpu up.
I (206) cpu_start: Starting app cpu, entry point is 0x40081050
I (0) cpu_start: App cpu up.

## Other items if possible
sdkconfig
blink.elf
[blink.zip](https://github.com/espressif/esp-idf/files/7561152/blink.zip)

o-marshmallow, 2021-11-19:
Hi @SMinin ,
I was able to reproduce the issue on my side. The problem does not appear on `4.3` dev branch (`release/v4.3`) nor on `master`.
It seems to be linked to UART buffer critical sections as the commit edb80a1965c827d15d92925b0351831f223e2f26 fixed the issue for me. So the solution is to either update your project to a newer IDF version or cherry-pick the commit:
```
git cherry-pick edb80a1965c827d15d92925b0351831f223e2f26
```

Please have a try and tell me if it fixed the issue for you.

AxelLin, 2021-11-19:
> Hi @SMinin , I was able to reproduce the issue on my side. The problem does not appear on `4.3` dev branch (`release/v4.3`) nor on `master`.

@o-marshmallow 
Since you can reproduce it I suggest bisect it to find out the bad commit.
You can also just revert edb80a1 on master tree to verify if it indeed cause the issue.
IMHO, edb80a1 is unlikely to be the bad commit.

o-marshmallow, 2021-11-22:
@AxelLin I agree, this needs further investigation. I imagine this commit has a side-effect that makes it work. But at least it's the simplest way for people who meets this issue to get something working and continue their work, until the root cause is found.

SMinin, 2021-11-22:
UART has nothing to do with it with all due respect. Attached is a screenshot of the debug window. The attachment also contains a binary image with an elf file to reproduce the error. It seems that the error occurs when the bootloader tries to access flash memory. I only get this error when "Compiler options -> Stack smashing protection mode" is set to "Overall". I am ready to provide additional information if necessary.

[image]
[blink_stack_protection_Overall.zip](https://github.com/espressif/esp-idf/files/7584567/blink_stack_protection_Overall.zip)

AxelLin, 2021-12-01:
> @AxelLin I agree, this needs further investigation. I imagine this commit has a side-effect that makes it work. But at least it's the simplest way for people who meets this issue to get something working and continue their work, until the root cause is found.

Once git bisect found the bad commit you can easily figure out the root cause.
It will only take you a few minutes, no need to guess.

AxelLin, 2021-12-01:
@SMinin 
Check if release/v4.3 branch works for you. (I don't hit this issue with v4.3.1-480-g233dc30fb1)

SunnyRuss, 2022-08-06:
I seem to have this problem.  I am using 4.4.1.  It happens every time from a power up.  The watchdog kicks in and restarts the processor and processor comes up clean (both CPUs running).  The compiler stack smashing setting is NONE
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9275426/sdkconfig.txt)
[appcpu.boot.log.txt](https://github.com/espressif/esp-idf/files/9275427/appcpu.boot.log.txt)
.

AxelLin, 2022-08-10:
> I seem to have this problem. I am using 4.4.1. It happens every time from a power up.

@SunnyRuss 
Can you try v4.4.2? https://github.com/espressif/esp-idf/releases/tag/v4.4.2

esp-wzh, 2022-08-10:
Hi, @SunnyRuss, any updates on v4.4.2 or other newer versions?

**On tag v4.3.1**: The bug was caused by commit https://github.com/espressif/esp-idf/commit/0e6f4ba2cf620572f4240f862c37de8d1abee928 and resolved in commit https://github.com/espressif/esp-idf/commit/d622bcfd46bc38e4d45fb1fecaf8dbfac4eab041. Only the version between these two commits will have the problem of no printing after "App cpu up".

**On tag v4.4.2**:I compiled some common apps based on the sdkconfig you provided (I changed back to the default partition table)  and couldn't reproduce the problem, can you provide more information for reproduction? From the log you provided I have been unable to extract more information for debugging

AxelLin, 2022-08-10:
> Hi, @SunnyRuss, any updates on v4.4.2 or other newer versions?
> 
> **On tag v4.3.1**: The bug was caused by commit [0e6f4ba](https://github.com/espressif/esp-idf/commit/0e6f4ba2cf620572f4240f862c37de8d1abee928) and resolved in commit [d622bcf](https://github.com/espressif/esp-idf/commit/d622bcfd46bc38e4d45fb1fecaf8dbfac4eab041). Only the version between these two commits will have the problem of no printing after "App cpu up".

@esp-wzh 
0e6f4ba is related to uart issue
d622bcf is related to malloc(0) and it's not in v4.3 branch.
Could you double check if above commit id is correct or not? I don't understand why these commits are related to this issue.

esp-wzh, 2022-08-10:
@AxelLin Sorry, I copied the wrong commit id, https://github.com/espressif/esp-idf/commit/edb80a1965c827d15d92925b0351831f223e2f26 is the commit for fix

closed 2023-04-24 by Alvin1Zhang