espressif/esp-idf #7927: export.bat IDF_PATH

pius-braun, 2021-11-18:
In export.bat, I had to move the statements:

:: Infer IDF_PATH from script location
set IDF_PATH=%~dp0
set IDF_PATH=%IDF_PATH:~0,-1%

from line 24 to line 17, i.e. above the statement "set PREFIX=python.exe %IDF_PATH%"
This avoids an error, when running "idf.py build".
The error message was "/tools/idf.py not found".

dobairoland, 2021-11-18:
I believe that this is the same issue as https://github.com/espressif/esp-idf/issues/7605. It was fixed together with a larger fix in the master branch but it hasn't been backported. So thank you @pius-braun for raising the issue. I assume that you use v4.3.x or an earlier version. We will fix this ASAP.

dobairoland, 2021-11-26:
This has been fixed in v4.3 branch as well. The fix for v4.2 and v4.1 will come soon. The fixing commits will be linked here.

closed 2021-11-26 by dobairoland