espressif/esp-idf #7929: Does I can send stereo audio data via ESP-IDF's I2S API functions

Augtons, 2021-11-18:
# Description

I'm always frustrated when I send stereo data by ESP-IDF 's API functions. There's an API function in ESP-IDF called "i2s_write()" but I can't find any docs and examples indicating that it can send stereo signals.

So is there a solution to use the I2S library of ESP-IDF to output stereo signals to GPIO?

If possible, could you tell me what should I do.

If not, is adding some API functions ( like "`i2s_write_to_left_channel()`, `i2s_write_to_right_channel()`, `i2s_write_stereo()`" ) feasible?

thanks!!

L-KAYA, 2021-11-19:
Yes, the current I2S driver supports stereo, you can apply stereo mode by setting `channel_format` field in `i2s_driver_config_t` to `I2S_CHANNEL_FMT_RIGHT_LEFT`. But if you want to only send data to a specific channel meanwhile keep another channel silent, the current driver in v4.4 is not supported yet.  Since I2S driver in v5.0 will be updated to a new version, it is possible to realize one single channel in stereo mode then.
Thanks for your advice Sincerely!

egnor, 2022-07-27:
[this seems obsolete and should probably be closed?]

I'm not sure if the original author understood that when using `i2s_write()`, the data for each sample includes separate values for left and right. For example, if using 16-bit sound, each sample is a 32 bit value, with 16 bits for the left channel and 16 bits for the right channel.

Augtons, 2022-07-27:
> [this seems obsolete and should probably be closed?]
>
> I'm not sure if the original author understood that when using `i2s_write()`, the data for each sample includes separate values for left and [...]

Thanks for your reply. I found the answer under your guidance!  That's in page 289 of Technical Reference Manual (version 4.4). Thanks a lot!!

closed 2025-01-13 by Alvin1Zhang