espressif/esp-idf #7930: LAN8720 DHCP issue

SimonPVS, 2021-11-18:
## Environment

Development Kit: ESP-32 WROOM
Kit version V3
Module or chip used: ESP32-WROOM-32
IDF version: esp-idf-v4.3.1
Build System: idf.py
Compiler version: esp-2020r3-8.4.0
Operating System: Windows 10
Using an IDE?: Eclipse V4.19.0.v20210303-1800
Power Supply: External 3.3

## Problem Description
DHCP does not work anymore when using esp-idf V4.3.1

### Expected Behavior
It should work as in esp-idf V4.2.1

### Actual Behavior
There is a dhcp error message when using esp-idf V4.3.1

### Steps to reproduce
Run an ethernet program with DHCP enabled, with esp idf version 4.3.1.

## Debug Logs
esp_netif_lwip: dhcp client start failed

david-cermak, 2021-11-19:
Hi @SimonPVS 

Could you please share some part of your initialisation code?

> Run an ethernet program with DHCP enabled, with esp idf version 4.3.1.

Most of default ethernet examples have DHCP enabled and work correctly on `v4.3.1`

> esp_netif_lwip: dhcp client start failed

It seems like you're starting the dhcp explicitly from your code, could you please tell us more? When exactly you run
```cpp
esp_netif_dhcpc_start(eth_netif);
```
and how is your `eth_netif` configured? using the default `ESP_NETIF_DEFAULT_ETH()`, or any changes?

Thank you

SimonPVS, 2021-11-19:
Hi, 

Today i did a re-install of the esp-idf V4.3.1 .
Weirdly enough, this had fixed the problem. I noticed there was a change in the negotiate function in the "esp_eth_phy_lan8720" file.

Does espressif apply patches to their released versions?

david-cermak, 2021-11-19:
@SimonPVS Please see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#versioning-scheme

> Does espressif apply patches to their released versions?

You are only expected to see changes if you're on a release branch. If you used `v4.3.1` release **tag**, nothing has or will be changed.

> I noticed there was a change in the negotiate function in the "esp_eth_phy_lan8720" file.

There's been a fix related to auto-negotiation, backported as 03a6c4975c7, but that was between `v4.3` and `v4.3.1`

closed 2021-11-25 by SimonPVS