espressif/esp-idf #7931: Ethernet Link down automatically after some random time when using W5500.

nilesh-dryad, 2021-11-18:
## Environment

- Development Kit: ESP32-S2-Kaluga-1
- Kit version (for WroverKit/PicoKit/DevKitC): V1.3
- Module or chip used: ESP32-S2-Wrover
- IDF version: v4.3 (Stable)
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 5V-2A

## Problem Description

I am using WIZ850io with ESP32S2 and using sample basic application provided with ESP-IDF with W5500.

### Expected Behavior
Ethernet Link should be active all the time. Even if it get down, it should automatically gets connected.

### Actual Behavior
Ethernet link is down after some random time. It also get Link Up but it does not get IP. After some debug, I come to know that Ethernet module does not reply to SPI command. I checked that by reading Ethernet MAC address from PHY register. When link was OK, it was giving correct MAC address. When link is down automatically, it stops giving correct MAC address.

### Steps to reproduce

1. Take basic ethernet example and configure it for W5500 using menuconfig
2. Run the example and keep it running for while. (approx 1-2 hours, if possible overnight)

### Code to reproduce this issue

```cpp
 Same code, no change
```
## Debug Logs

```
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x18d8
load:0x4004c000,len:0xa04
load:0x40050000,len:0x2b80
entry 0x4004c1f4
I (21) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (21) boot: compile time 14:08:49
I (21) boot: chip revision: 0
I (25) boot.esp32s2: SPI Speed      : 80MHz
I (29) boot.esp32s2: SPI Mode       : DIO
I (34) boot.esp32s2: SPI Flash Size : 2MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (70) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=106a0h ( 67232) map
I (103) esp_image: segment 1: paddr=000206c8 vaddr=3ffbed50 size=0298ch ( 10636) load
I (106) esp_image: segment 2: paddr=0002305c vaddr=40022000 size=0cd44h ( 52548) load
I (122) esp_image: segment 3: paddr=0002fda8 vaddr=50000000 size=00010h (    16) load
I (122) esp_image: segment 4: paddr=0002fdc0 vaddr=00000000 size=00258h (   600) 
I (128) esp_image: segment 5: paddr=00030020 vaddr=40080020 size=2e0d8h (188632) map
I (180) boot: Loaded app from partition at offset 0x10000
I (181) boot: Disabling RNG early entropy source...
I (192) cac��f<��3?�.���,W��
                            �V�,ZY�B��size 8KB, 4Ways, cache line size 32Byte
I (192) cpu_start: Pro cpu up.
I (204) cpu_start: Pro cpu start user code
I (204) cpu_start: cpu freq: 160000000
I (204) cpu_start: Application information:
I (208) cpu_start: Project name:     ethernet_basic
I (214) cpu_start: App version:      v4.3.1-dirty
I (219) cpu_start: Compile time:     Nov 18 2021 14:08:45
I (226) cpu_start: ELF file SHA256:  59c995876fe73212...
I (231) cpu_start: ESP-IDF:          v4.3.1-dirty
I (237) heap_init: Initializing. RAM available for dynamic allocation:
I (244) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (250) heap_init: At 3FFC2FD8 len 00039028 (228 KiB): DRAM
I (257) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (263) spi_flash: detected chip: generic
I (268) spi_flash: flash io: dio
W (271) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (288) cpu_start: Starting scheduler on PRO CPU.
I (290) w5500-mac: version=4
I (300) esp_eth.netif.glue: 64:1a:ba:00:00:02
I (300) esp_eth.netif.glue: ethernet attached to netif
I (300) eth_example: Ethernet Started
I (2300) eth_example: Ethernet Link Up
I (2300) eth_example: Ethernet HW Addr 64:1a:ba:00:00:02
I (3290) esp_netif_handlers: eth ip: 192.168.178.116, mask: 255.255.255.0, gw: 192.168.178.1
I (3290) eth_example: Ethernet Got IP Address
I (3290) eth_example: ~~~~~~~~~~~
I (3290) eth_example: ETHIP:192.168.178.116
I (3300) eth_example: ETHMASK:255.255.255.0
I (3300) eth_example: ETHGW:192.168.178.1
I (3310) eth_example: ~~~~~~~~~~~
I (1226300) eth_example: Ethernet Link Down
I (1228300) eth_example: Ethernet Link Up
I (1228300) eth_example: Ethernet HW Addr 64:1a:ba:00:00:02 <This is not read using SPI >
E (1346300) eth_example: Ethernet Lost its IP Address

```

david-cermak, 2021-11-19:
Hi @nilesh-dryad 

> IDF version: v4.3 (Stable)

Which commit id are you on? Is it the release branch or `v4.3` tag?
Note, that the important fixes to w5500 hadn't been included in `v4.3` tag, but were added to `v4.3.1`

Please see the relevant commit history: https://github.com/espressif/esp-idf/commits/release/v4.3/components/esp_eth and make sure these are included d09f6cac247, 1f2af241188.

nilesh-dryad, 2021-11-19:
@david-cermak I apologies for mentioning incorrect version in Environment section. I was using 4.3 then I switched to 4.3.1 as soon as it was released.

Still issue persists. I am able to reconnect it via following workaround but issue still persists. There is a setting of 120 seconds for getting lost ip handler callback. So I have to wait till 120 seconds to reconnect.

Note: Ethernet Lost IP handler logic I took it from the master branch.

> static void lost_ip_event_handler(void *arg, esp_event_base_t event_base,
                                 int32_t event_id, void *event_data)
{
    ESP_LOGE(TAG, "Ethernet Lost its IP Address");
    esp_eth_stop(eth_handle);
    esp_eth_del_netif_glue(ptr);
    esp_eth_driver_uninstall(eth_handle);
    vTaskDelay(100);
    ESP_ERROR_CHECK(esp_eth_driver_install(&config, &eth_handle));
    ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
    esp_eth_start(eth_handle);
}

david-cermak, 2021-11-19:
@nilesh-dryad May I ask some questions so that I could easily reproduce your problem? I've been running ESP32-S2 + w5500 setup for 7 hours already, the link is up, IP address valid and responding.
* How do you connect to w5500, wires? or a custom PCB using the kaluga board?
* Any other device on the SPI? What is the frequency?
* How do you test it, flood it with packets or just leave it respond to some periodic packets, such as pings or keep it completely silent?

> It also get Link Up but it does not get IP
> Ethernet module does not reply to SPI command. 

So you got the Link up event, but the module didn't respond to SPI?

Anyway, I'll keep running my test over the weekend to see if I'm about to see any link-downs.

nilesh-dryad, 2021-11-19:
- I connect w5500 using wires on Kaluga board.
- No other device on SPI but we are planning to add other device. Earlier I was testing with other SPI device but issue was there only with Ethernet SPI, other SPI device was working even Ethernet was not working.
- 20 Mhz Frequency I am using.
- I continuously pinging Kaluga board from my PC. But even if I don't ping, it came down after some time.

I am using wires to connect w5500 module in my setup. Can it be a wire problem? I used brand new wires and Ethernet also work for some hours using those wires. See below image.
[image]

I also have my custom board with ESP32-S2 and W5500 which I just got yesterday from fabrication. When I started this board I got link down two times but since then Ethernet is running since 15 hours without link break. Ethernet module is mounted on PCB so no wires on this board. Still I am going to put my PCB over the weekend to test the connectivity. See below image.

[image]

david-cermak, 2021-11-19:
I think that those wires could easily cause issues in a noisy environment, it not only about frequency, but mainly how the signal (edges) really look like.
I'm running a similar test, continuous pinging, using 36MHz, but connected similarly to your PCB (have a simple breakout board stacking a SAOLA kit to w5500 EVB, running for > 12 hours with no issues)

nilesh-dryad, 2021-11-22:
@david-cermak Ethernet on my new PCB worked without any link down successfully without a single break. It seems like it was cable issue with Kaluga board. I hope yours also worked over the weekend. I am closing this issue as it appears to be cable issue when using module with Kaluga board even though cables were brand new but we can not ignore the fact of noisy environment while using cables.

closed 2021-11-22 by nilesh-dryad