espressif/esp-idf #7934: wifi_prov_mgr: WIFI_PROV_END not raised

BradSaracik-WEL, 2021-11-19:
## Environment
ESP32-WROVER-IB
v4.2.2 - 7ab8f793ca5b026f37ae812bcc103e3aa698d164 
CMake
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Linux
3.3V

## Problem Description
WIFI_PROV_EVENT event handler is not picking up WIFI_PROV_END as expected as it is not successfully raised after WIFI_PROV_CRED_SUCCESS

I have not found issues raised around this behaviour. A colleague confirmed the same issue on a separate project

It appears from wifi_provisioning/manager.c that prov_stop_task() is not called / completing successfully. but wifi_prov_mgr_stop_service() is called

### Expected Behavior
After "Provisioning successful" WIFI_PROV_END event should occur

### Actual Behavior
After "Provisioning successful" WIFI_PROV_END does not occur.

Attempting to call wifi_prov_mgr_deinit(); at our end after success prevents the app from exiting the provision cleanly and caused firmware problems (sorry I don't seem to have kept that trace)

### Steps to reproduce

1. menuconfig for BLE wifi provisioning. 
2. setup wifi provisioning for BLE as per [example]( esp-idf/examples/provisioning/wifi_prov_mgr/main/app_main.c)
3. I used the android provisioning  [app](https://play.google.com/store/apps/details?id=com.espressif.provble&hl=en_AU&gl=US) 

### Code:
static void event_handler(void* arg, esp_event_base_t event_base,
                          int event_id, void* event_data)
{
    if (event_base == WIFI_PROV_EVENT) {
        switch (event_id) {
            case WIFI_PROV_START:
                ESP_LOGI(TAG, "Provisioning started");
                break;
            case WIFI_PROV_CRED_RECV: {
                //...
                break;
            }
            case WIFI_PROV_CRED_FAIL: {
                //...
                break;
            }
            case WIFI_PROV_CRED_SUCCESS:
                ESP_LOGI(TAG, "Provisioning successful");
                break;
            case WIFI_PROV_END:
                /* De-initialize manager once provisioning is finished */
                ESP_LOGI(TAG, "Provisioning ENDED");
                wifi_prov_mgr_deinit();
                break;
            default:
                break;
        }  //else ....
}

    
## Info Logs
output:
wifi_prov_mgr: STA Got IP
wifi: Provisioning successful
.
.
//do not get wifi_prov_mgr: Provisioning stopped in prov_stop_task()
//do not get Provisioning ENDED

## Debug Logs
.
.
wifi_prov_mgr: Stopping provisioning //wifi_prov_mgr_stop_service()

## Other
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7567659/sdkconfig.txt)

BradSaracik-WEL, 2021-11-22:
I'm not sure how reliable this is, but its a workaround in my build

`wifi_prov_mgr_disable_auto_stop(500); //before wifi_prov_mgr_start_provisioning()`

and
 ```
if (provision_succeeded) { //ie after WIFI_PROV_CRED_SUCCESS
            wifi_prov_mgr_stop_provisioning();
 }
```

WIFI_PROV_END now gets triggered as expected

laukik-hase, 2021-11-22:
Hello, @BradSaracik-WEL 

I tried reproducing this issue on IDF `v4.2.2` but was unable to do it. The `wifi_prov_mgr` example (with your configurations) ran as it was supposed to (without the fix you attached).

1. Do you see the failure with some specific configurations? Does the stock example work as it is supposed to on your side?
2. Could you attach detailed logs of your example (from the start of provisioning with demo credentials) and the android app version?

Following are the example logs attached from my testing:

```
I (887) wifi_prov_scheme_ble: BT memory released
I (897) app: Starting provisioning
I (897) phy_init: phy_version 4660,0162888,Dec 23 2020
W (907) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (1067) wifi:mode : sta (4c:11:ae:71:2f:b4)
I (1067) BTDM_INIT: BT controller compile version [ba56601]
I (1067) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1307) protocomm_nimble: BLE Host Task Started
I (1317) wifi_prov_mgr: Provisioning started with service name : PROV_712FB4 
I (1317) app: Provisioning started
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
W (18927) wifi:Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!

I (35067) app: Received Wi-Fi credentials
	SSID     : myssid
	Password : mypassword
I (37277) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (37877) wifi:state: init -> auth (b0)
I (37887) wifi:state: auth -> assoc (0)
I (37897) wifi:state: assoc -> run (10)
I (37927) wifi:connected with myssid, aid = 5, channel 6, BW20, bssid = 5a:9f:9c:86:91:71
I (37927) wifi:security: WPA2-PSK, phy: bgn, rssi: -38
I (37927) wifi:pm start, type: 1

I (38007) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (38877) app: Connected with IP Address:192.168.64.95
I (38877) esp_netif_handlers: sta ip: 192.168.64.95, mask: 255.255.255.0, gw: 192.168.64.106
I (38877) wifi_prov_mgr: STA Got IP
I (38877) app: Provisioning successful
I (38887) app: Hello World!
I (39887) app: Hello World!
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19
I (40137) wifi_prov_mgr: Provisioning stopped
I (40137) app: Provisioning ENDED
I (40137) wifi_prov_scheme_ble: BTDM memory released
I (40887) app: Hello World!
I (41887) app: Hello World!
I (42887) app: Hello World!
I (43887) app: Hello World!
```
```

BradSaracik-WEL, 2021-11-23:
thanks. I see that uses NimBLE, rather than Bluedroid that we're configured for. do you get the same result?

we also get:
>  wifi:Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!

BradSaracik-WEL, 2021-11-23:
I tried NimBLE and got the same result (minus workaround)
So I've seen this on different wifi ssid/routers, bluedroid and nimble and a colleague had it in a different project.
not sure... you dont have sntp?

the end is waiting  for this line that never comes
`wifi_prov_mgr: Provisioning stopped `
```I (1077) wifi:wifi driver task: 3ffd3d74, prio:23, stack:6144, core=0
I (1077) system_api: Base MAC address is not set
I (1077) system_api: read default base MAC address from EFUSE
I (1087) wifi:wifi firmware version: bb6888c
I (1087) wifi:wifi certification version: v7.0
I (1087) wifi:config NVS flash: enabled
I (1097) wifi:config nano formating: enabled
I (1097) wifi:Init data frame dynamic rx buffer num: 32
I (1097) wifi:Init management frame dynamic rx buffer num: 32
I (1107) wifi:Init management short buffer num: 32
I (1107) wifi:Init static tx buffer num: 16
I (1117) wifi:Init static rx buffer size: 1600
I (1117) wifi:Init static rx buffer num: 10
I (1127) wifi:Init dynamic rx buffer num: 32
I (1127) wifi_init: rx ba win: 16
I (1127) wifi_init: tcpip mbox: 32
I (1137) wifi_init: udp mbox: 6
I (1137) wifi_init: tcp mbox: 6
I (1147) wifi_init: tcp tx win: 5744
I (1147) wifi_init: tcp rx win: 5744
I (1147) wifi_init: tcp mss: 1436
I (1157) wifi_init: WiFi IRAM OP enabled
I (1157) wifi_init: WiFi RX IRAM OP enabled
I (1167) wifi: Wiping ESP WiFi credentials
I (1177) wifi: No WiFi Credentials available. Starting BLE provisioning
I (1177) wifi_prov_scheme_ble: BT memory released
I (1187) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1197) wifi:mode : sta (a8:03:2a:ff:5b:38)
I (1197) wifi: WiFi Attempting connection
I (1197) wifi: WiFi Scan disabled
W (1197) wifi: No SSID configured
I (1207) BTDM_INIT: BT controller compile version [ba56601]
I (1217) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1457) protocomm_nimble: BLE Host Task Started
I (1457) wifi_prov_mgr: Provisioning started with service name : prov_mything
I (1457) wifi: Provisioning started
I (1467) wifi: Joining wifi ssid=
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
W (1967) wifi: provisioning .
W (6967) wifi: provisioning .
W (11967) wifi: provisioning .
W (16967) wifi: provisioning .
W (19707) wifi:Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!

W (21967) wifi: provisioning .
W (26967) wifi: provisioning .
I (31897) wifi: Received Wi-Fi credentials
        SSID     : myssid
        Password : mypasswd
W (31967) wifi: provisioning .
I (32907) wifi:new:<9,1>, old:<1,0>, ap:<255,255>, sta:<9,1>, prof:1
I (32917) wifi:state: init -> auth (b0)
I (32927) wifi:state: auth -> assoc (0)
I (32927) wifi:state: assoc -> run (10)
I (32967) wifi:connected with myssid, aid = 1, channel 9, 40U, bssid = 46:94:ed:d2:7a:cd
I (32967) wifi:security: WPA2-PSK, phy: bgn, rssi: -62
I (32967) wifi:pm start, type: 1

I (32967) wifi:  !! WiFi connected !!
I (33017) wifi:AP's beacon interval = 204800 us, DTIM period = 4
I (35057) SNTP: Initializing SNTP with 'time.google.com' and ''
I (35057) SNTP: Setting POSIX timezone to AWST-8
I (35057) wifi: WiFi Connected, IP: 192.168.1.47, GW: 192.168.1.1, RSSI: -60
I (35067) wifi: Main DNS IP:192.168.1.1
I (35067) wifi: Fallback DNS IP:0.0.0.0
I (35067) wifi: Backup DNS IP:0.0.0.0
I (35077) esp_netif_handlers: sta ip: 192.168.1.47, mask: 255.255.255.0, gw: 192.168.1.1
I (35087) wifi_prov_mgr: STA Got IP
I (35087) wifi: Provisioning successful
W (36967) wifi: provisioning .
I (37187) SNTP: Got time sync, current time is Tue Nov 23 09:21:37 2021
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
W (41967) wifi: provisioning .
W (46967) wifi: provisioning .
W (51967) wifi: provisioning .
W (56967) wifi: provisioning .
W (61967) wifi: provisioning .
W (66967) wifi: provisioning .
W (71967) wifi: provisioning .
W (76967) wifi: provisioning .
E (81967) wifi: NO WiFi provisioned

BradSaracik-WEL, 2021-11-23:
attached the alternate sdkconfig for the above
[sdkconfig_nimble.txt](https://github.com/espressif/esp-idf/files/7585376/sdkconfig_nimble.txt)

laukik-hase, 2021-11-23:
Hello, @BradSaracik-WEL

I tried the example with `Bluedroid` but to no avail. Could you please post the logs for the example without any of your changes? Also, share the Android app version you are using. 

The message `W (29853) wifi:Error! Should use default active scan time parameter for Wi-Fi scan when Bluetooth is enabled!!!!!!` can be ignored since that is a warning printed from the Wi-Fi library, but it does not affect any functionality. The commit [`d96dd2c`](https://github.com/espressif/esp-idf/commit/d96dd2c002af9c5e850aa4a9c4392d4773e01cb3) fixes this, you can cherry-pick it.

BradSaracik-WEL, 2021-11-23:
Android app version 2.0.10 - 780389c

laukik-hase, 2021-11-23:
Hello, @BradSaracik-WEL

Thank you! Could you also post the logs for the example without any of your added features and changes? This will help us understand the issue further.

laukik-hase, 2022-01-10:
Hello, @BradSaracik-WEL

Any updates on this issue?

Alvin1Zhang, 2022-01-17:
@BradSaracik-WEL Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2022-01-17 by Alvin1Zhang