espressif/esp-idf #7937: peer_dev_rec table entries get overwritten

flonas955, 2021-11-19:
## Environment

- Development Kit:  none
- Module or chip used:  ESP32
- IDF version: v4.3.1
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:  Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes (VS Code withe ESP IDF extension)

## Problem Description
I am having problems with Bluetooth bonding on iPhones, especially on an iPhone Xs device. I have previously posted more info on those issues [here](https://github.com/espressif/esp-nimble/issues/33) and [here](https://github.com/apache/mynewt-nimble/issues/1052).

Basically, what happens is that, after bonding two phones to the same ESP32 (one iPhone Xs, iOS 14.1 and one Android 11 phone), the iPhone will, eventually, report "Peer removed pairing information" error and fail to connect. The only way to reconnect is to delete the old bond and reconnect. Still, the error will appear again very soon. 

The maximum number of bonds is set to 3.

I have found something that doesn't look good with the peer device records returned by **ble_rpa_get_peer_dev_records()**. It seems that the records are overwritten and I think this is directly related to the issue. I will show the logs below.

### Expected Behavior
After bonding was done, all connections should work at each attempt. 

### Actual Behavior
After bonding at least 2 phones, eventually the iPhone will report that the peer removed pairing information. 

### Steps to reproduce

This is not easily reproducible, but the steps are the follwing:

1. Bond the iPhone
2. Do several connects and disconnects -> those work.
2. Bond the Android device (do not have other phones to try with)
3. Do several connects and disconnects with the Android -> those work.
4. Delete the bond from the iPhone and then rebond with the board and try to connect.

After this step 4, the problem will usually appear. If it doesn't, repeat step 4 several times. 

### Code to reproduce this issue

I managed to reproduce the problem using the [bleprph ](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/bleprph) example. I have only added RPA address resolution to that example and increased the number of stored CCCD from 8 to 32. I have attached the project files:

[bleprph.zip](https://github.com/espressif/esp-idf/files/7568854/bleprph.zip)

## Debug Logs

I have printed the _peer_dev_rec_ array from _ble_hs_resolv.c_ in the disconnect event (code is attached above), using the _print_rpa_peers()_ function, defined in _main.c_.

E (232581) NimBLE_BLE_PRPH: ble_rpa_get_num_peer_dev_records() returned 2
E (232581) NimBLE_BLE_PRPH: peer_dev_record index 0: rec_used = 1
E (232591) NimBLE_BLE_PRPH: peer_dev_record index 0: rand_addr_type = 1
E (232601) NimBLE_BLE_PRPH: peer_dev_record index 0: pseudo_addr = 4e:86:39:a1:66:d2
E (232601) NimBLE_BLE_PRPH: peer_dev_record index 0: rand_addr = 4e:86:39:a1:66:d2
E (232611) NimBLE_BLE_PRPH: peer_dev_record index 0: identity_addr = 50:7a:c5:5f:c3:38 <- The iPhone's ID address
E (232621) NimBLE_BLE_PRPH: peer_dev_record index 0: peer_addr = 50:7a:c5:5f:c3:38
E (232631) NimBLE_BLE_PRPH: peer_dev_record index 0: irk_present = 1

E (232651) NimBLE_BLE_PRPH: peer_dev_record index 1: rec_used = 1
E (232651) NimBLE_BLE_PRPH: peer_dev_record index 1: rand_addr_type = 0
E (232661) NimBLE_BLE_PRPH: peer_dev_record index 1: pseudo_addr = 49:92:d7:12:69:21
E (232671) NimBLE_BLE_PRPH: peer_dev_record index 1: rand_addr = 49:92:d7:12:69:21
E (232681) NimBLE_BLE_PRPH: peer_dev_record index 1: identity_addr = ac:5f:ea:e2:08:c1
E (232691) NimBLE_BLE_PRPH: peer_dev_record index 1: peer_addr = ac:5f:ea:e2:08:c1   <- The Android's ID address
E (232691) NimBLE_BLE_PRPH: peer_dev_record index 1: irk_present = 1
E (232701) NimBLE_BLE_PRPH: 

/****** Up until this point, both phones are connecting properly. Next, I deleted the bond from the iPhone and created a new one *******/

GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
advertise complete; reason=13GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
connection established; status=0 handle=0 our_ota_addr_type=1 our_ota_addr=52:f1:8d:15:04:21 our_id_addr_type=0 our_id_addr=7c:9e:bd:c9:68:5e peer_ota_addr_type=0 peer_ota_addr=50:7a:c5:5f:c3:38 peer_id_addr_type=0 peer_id_addr=50:7a:c5:5f:c3:38 conn_itvl=24 conn_latency=0 supervision_timeout=72 encrypted=0 authenticated=0 bonded=0

mtu update event; conn_handle=0 cid=4 mtu=256
subscribe event; conn_handle=0 attr_handle=8 reason=1 prevn=0 curn=0 previ=0 curi=1
encryption change event; status=0 handle=0 our_ota_addr_type=1 our_ota_addr=52:f1:8d:15:04:21 our_id_addr_type=1 our_id_addr=52:f1:8d:15:04:21 peer_ota_addr_type=1 peer_ota_addr=4e:86:39:a1:66:d2 peer_id_addr_type=1 peer_id_addr=4e:86:39:a1:66:d2 conn_itvl=24 conn_latency=0 supervision_timeout=72 encrypted=1 authenticated=0 bonded=1

subscribe event; conn_handle=0 attr_handle=8 reason=2 prevn=0 curn=0 previ=1 curi=0
disconnect; reason=531 handle=0 our_ota_addr_type=1 our_ota_addr=52:f1:8d:15:04:21 our_id_addr_type=0 our_id_addr=7c:9e:bd:c9:68:5e peer_ota_addr_type=0 peer_ota_addr=50:7a:c5:5f:c3:38 peer_id_addr_type=0 peer_id_addr=50:7a:c5:5f:c3:38 conn_itvl=24 conn_latency=0 supervision_timeout=72 encrypted=1 authenticated=0 bonded=1

E (256421) NimBLE_BLE_PRPH: ble_rpa_get_num_peer_dev_records() returned 2
E (256431) NimBLE_BLE_PRPH: peer_dev_record index 0: rec_used = 1
E (256431) NimBLE_BLE_PRPH: peer_dev_record index 0: rand_addr_type = 0
E (256441) NimBLE_BLE_PRPH: peer_dev_record index 0: pseudo_addr = 49:92:d7:12:69:21
E (256451) NimBLE_BLE_PRPH: peer_dev_record index 0: rand_addr = 49:92:d7:12:69:21
E (256461) NimBLE_BLE_PRPH: peer_dev_record index 0: identity_addr = ac:5f:ea:e2:08:c1 <- Android ID
E (256461) NimBLE_BLE_PRPH: peer_dev_record index 0: peer_addr = ac:5f:ea:e2:08:c1
E (256471) NimBLE_BLE_PRPH: peer_dev_record index 0: irk_present = 1
E (256481) NimBLE_BLE_PRPH: =============================================================================
E (256491) NimBLE_BLE_PRPH: peer_dev_record index 1: rec_used = 1
E (256501) NimBLE_BLE_PRPH: peer_dev_record index 1: rand_addr_type = 0
E (256501) NimBLE_BLE_PRPH: peer_dev_record index 1: pseudo_addr = 49:92:d7:12:69:21
E (256511) NimBLE_BLE_PRPH: peer_dev_record index 1: rand_addr = 49:92:d7:12:69:21
E (256521) NimBLE_BLE_PRPH: peer_dev_record index 1: identity_addr = ac:5f:ea:e2:08:c1 <- Android ID
E (256531) NimBLE_BLE_PRPH: peer_dev_record index 1: peer_addr = ac:5f:ea:e2:08:c1
E (256541) NimBLE_BLE_PRPH: peer_dev_record index 1: irk_present = 1
E (256541) NimBLE_BLE_PRPH: =============================================================================

As you can see, the record for the iPhone disappeared and it was replaced by the other phone's record. At this step, the "Peer removed pairing information" error will be shown on the iPhone.

Are there any known issues with the RPA device record table? 

Thank you,
Florin

veneno-529, 2023-01-08:
I'm using Nimble 1.3 and esp idf 4.4.2 version and Platform io extension in Vs code.
When I enable RPA my bonding information is not stored in rpa peer records. My android device gets bonded,paired successfully but when i disconnect and connect it again it fails to connect.
In the logs I have found this line which says it was not able to find records of peer device.
**E (83478) NimBLE_BLE_PRPH: ble_rpa_get_num_peer_dev_records() returned 0**

I have posted more details regarding logs and my code in the following issue number.
https://github.com/espressif/esp-nimble/issues/33#issuecomment-1374715896

SumeetSingh19, 2023-01-11:
Hi @flonas955 ,
I checked and it does seem that records can get overwritten sometimes. I have provided a patch that I hope will avoid these duplicate entries, could you apply it at your end and see if you still get the errors?

Patch: [Fix_incomplete_clearing_of_peer_dev_rec.zip](https://github.com/espressif/esp-idf/files/10391217/Fix_incomplete_clearing_of_peer_dev_rec.zip)

veneno-529, 2023-01-14:
> @flonas955 I'm using Nimble 1.3 and esp idf 4.4.2 version and Platform io extension in Vs code. When I enable RPA my bonding information is not stored in rpa peer records. My android device gets bonded,paired successfully but when i disconnect and connect it again it fails to connect. In the logs I have found this line which says it was not able to find records of peer device. **E (83478) NimBLE_BLE_PRPH: ble_rpa_get_num_peer_dev_records() returned 0**
> 
> I have posted more details regarding logs and my code in the following issue number. [#5105 (comment)](https://github.com/espressif/esp-idf/issues/5105#issuecomment-1374696851)

@SumeetSingh19 can you help out with my issue?

rahult-github, 2023-04-25:
Changes are committed to github 5c8263a08bb8a892604dcd5988846a49b5a71bd7 . Closing this issue. Feel free to reopen in case of any further comments.

closed 2023-04-25 by rahult-github