espressif/esp-idf #7940: Could we please get support for pyparsing > 2.3.1?

leres, 2021-11-20:
**Is your feature request related to a problem? Please describe.**

requirements.txt pins pyparsing to ">=2.0.3,<2.4.0" which means the newest version allowed is 2.3.1 and that was released way back in January of 2019. While many newer version appear to work, e.g. 2.4.7, 3.0.4 and 3.0.6 do not; the symptoms are numerous LD errors, e.g. "...dangerous relocation: l32r: literal placed after use: ..." 

**Describe the solution you'd like**

I was surprised to find that even recent releases of esp-idf do not support current  versions of pyparsing; could ldgen be updated to work with pyparsing 3.X?

dobairoland, 2021-11-21:
Thanks. We are completely aware of this and tracking it. There were several changes in pyparsing regarding this. None of them was sufficient for us. We are waiting for a new release and we will resume supporting the newest versions again.

The question "Is your feature request related to a problem?" is related to your personal issues why you need this. This is not clear to me. Please share if there is any beside that you think always the newest version should be supported. ESP-IDF is run in a custom virtual environment which should not interfere with the "outside world". For other applications you still can have the latest version of pyparsing. And there is no use case for you in the ESP-IDF environment I'm aware of where you should use pyparsing directly.

dobairoland, 2021-11-22:
https://github.com/pyparsing/pyparsing/issues/87

I'm linking the related issue here which is the root cause for using still `pyparsing<2.4.0`. It is closed but not all related issues are fixed for us and we need to address those.

BTW, I think very soon it will be possible to bypass all version requirements. I think this might happen much sooner than fixing pyparsing compatibility.