espressif/esp-idf #7942: ESP32C3 will crash when the log debug level lightsleep is enabled

dreamcmi, 2021-11-20:
I use ESP-IDF 4.3.1 to program esp32c3. When I set the level of log output to debug, the error of `abort() was called at PC 0x4038191f on core 0 ` appears when entering lightsleep mode.
The development board uses ESP32-C3-Devkitm-1, and the system is windows.
```shell
D (361) cpu_start: calling init function: 0x4200002a
D (366) intr_alloc: Connected src 39 to int 2 (cpu 0)
I (370) sleep: Configure to isolate all GPIO pins in sleep state
I (376) sleep: Enable automatic switching of GPIO sleep configuration
D (382) intr_alloc: Connected src 50 to int 3 (cpu 0)
I (387) cpu_start: Starting scheduler.
D (391) intr_alloc: Connected src 37 to int 4 (cpu 0)
D (391) heap_init: New heap initialised at 0x3fcc0000
D (401) intr_alloc: Connected src 33 to int 7 (cpu 0)
I (401) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
start lightsleep

abort() was called at PC 0x4038191f on core 0
Core  0 register dump:
MEPC    : 0x403840d0  RA      : 0x40384796  SP      : 0x3fc8fa30  GP      : 0x3fc8b200  
TP      : 0x3fc89df4  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x3fc8fa94  A0      : 0x3fc8fa5c  A1      : 0x3fc8fa92  
A2      : 0x00000000  A3      : 0x3fc8fa89  A4      : 0x00000001  A5      : 0x3fc8d000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000004  S3      : 0xffffffff  
S4      : 0x3fc90148  S5      : 0x3fc8ff9c  S6      : 0x00000000  S7      : 0x00000001  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fc8fa30: 0x00000000 0x00000000 0x3fc8fa90 0x40389820 0xa5a5a5a5 0x00000004 0x3fc8fa8c 0x3fc8bf38
3fc8fa50: 0x3fc8fa94 0x3fc8bf54 0x3fc8fa90 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fc8fa70: 0x43502074 0x34783020 0x31383330 0x20663139 0x63206e6f 0x2065726f 0x00000030 0x00000000
3fc8fa90: 0x3fc90030 0x38333034 0x66313931 0x42011f00 0xffffffff 0x00000001 0x3fc8dd58 0x40381922
3fc8fab0: 0x3c021510 0x3fc900ec 0x3fc8dd58 0x4038500e 0xa5a5a5a5 0x00000004 0x3fc8fb0c 0x3fc8dbc4
3fc8fad0: 0x3c020664 0x00000001 0x3fc8dd58 0x40381a50 0x3fc8fbf4 0x00000009 0x00000002 0x40381b26
3fc8faf0: 0x00000000 0x00000008 0x3fc90148 0x3fc8dd58 0x3c021510 0x00000009 0x3c02065c 0x4200d110
3fc8fb10: 0xa5a5a5a5 0xa5a5a5a5 0x00000002 0x4200e178 0xa5a5a5a5 0x00000064 0x00000003 0x3c021510
3fc8fb30: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x3c0262c8 0x00000001 0x00000001
3fc8fb50: 0x00000000 0x3c0262c8 0x00000000 0x00000000 0x0000
Data too long, check log folder for remaining data.
[2021/11/21 00:24:50.4513] → a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fc10: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fc30: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fc50: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fc70: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x303034a5
3fc8fc90: 0xffffffff 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fcb0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fcd0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fcf0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fd10: 0x00000191 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x3c021558 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fd30: 0x00000009 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000001 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fd50: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fd70: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fd90: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fdb0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fdd0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fdf0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8fe10: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 0caf53b7b69d8722

Rebooting...
```
I have provided the minimum reproduction project in the attachment for official inspection。
[lightsleep.zip](https://github.com/espressif/esp-idf/files/7574846/lightsleep.zip)

o-marshmallow, 2021-11-22:
Hi @dreamcmi ,
Thanks for reporting, I am going to manage this

o-marshmallow, 2021-11-22:
Hi @dreamcmi ,

I just pushed a fix internally, it will be synchronized to Github once the tests pass.

closed 2021-11-22 by espressif-bot