espressif/esp-idf #7943: I use fish shell to set my idf path ,but it run failed.

Yecgaa1, 2021-11-21:
/home/xtx/esp/esp-idf/export.fish                                                                                                                    02:36:57
Detecting the Python interpreter
source: Error encountered while sourcing file “~/esp/esp-idf/tools/detect_python.fish”:
source: No such file or directory
Adding ESP-IDF tools to PATH...
~/esp/esp-idf/export.fish (line 1): The expanded command was empty.
"$ESP_PYTHON" "$IDF_PATH"/tools/idf_tools.py export
^
in command substitution
        called on line 18 of file ~/esp/esp-idf/export.fish
in function '__main'
        called on line 68 of file ~/esp/esp-idf/export.fish
from sourcing file ~/esp/esp-idf/export.fish
        called on line 185 of file /usr/share/fish/config.fish
in function '.' with arguments '/home/xtx/esp/esp-idf/export.fish'
~/esp/esp-idf/export.fish: Unknown error while evaluating command substitution
in function '__main'
        called on line 68 of file ~/esp/esp-idf/export.fish
from sourcing file ~/esp/esp-idf/export.fish
        called on line 185 of file /usr/share/fish/config.fish
in function '.' with arguments '/home/xtx/esp/esp-idf/export.fish'
env: ‘idf.py’: No such file or directory

dobairoland, 2021-11-21:
Hi @Yecgaa1. Did you set IDF_PATH to be `~/esp/esp-idf`? Could you please try to set it to `/home/xtx/esp/esp-idf` to see if that helps?

Yecgaa1, 2021-11-22:
Hi，However I have ever set it 
[image]

This is how I set
[image]

dobairoland, 2021-11-22:
Yes, please try to change the first line to:
```sh
export IDF_PATH=/home/xtx/esp/esp-idf
```
or 
```sh
export IDF_PATH=$HOME/esp/esp-idf
```

I think the issue is that `~` is not replaced with `/home/xtx`.

What is your operating system, terminal and IDE? Are you using Vscode in WSL or on Linux?

Yecgaa1, 2021-11-22:
`export IDF_PATH=/home/xtx/esp/esp-idf`
Oh，It works！
System is ubuntu20.04.Is that a bug?

dobairoland, 2021-11-22:
Fish shell seems to prefer `set` instead of `export`: https://fishshell.com/docs/current/fish_for_bash_users.html#variables

`export` works with full path or `$HOME` but not with `~`. `set` works with `~` as well.

```sh
export TEST=~
echo $TEST
```
it prints:
```
~
```

```sh
 set TEST ~
 echo $TEST
```

it prints:
```
/home/<user>
```

So I don't think this would be a bug but a limitation of `fish`.

closed 2021-11-22 by dobairoland