espressif/esp-idf #7946: [esp32 C3 rev1] wifi build error

chegewara, 2021-11-22:
esp-idf: 4a011f318377ab717b9dc459019c1055602c93ff
board: esp32 C3 rev1

```
/home/chegewara/.espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/chegewara/esp/master/components/esp_phy/lib/esp32c3/libphy.a(phy_chip_v7.o): in function `phy_get_romfunc_addr':
(.text.phy_get_romfunc_addr+0x1a): undefined reference to `rom_temp_to_power'
```
For minimum `rev1` it fail to build, for `rev3` it is ok (i didnt test other rev versions).

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7577603/sdkconfig.txt)
Thanks

xueyunfei998, 2021-11-22:
hi @chegewara 

Which example do you use to make compilation errors? I'll try to reproduce this problem locally

chegewara, 2021-11-22:
It is my custom code with C++ class. Code works good with S3, can be build with C3 rev3, but fail to build with C3 rev1.
I am guessing it should fail with any espressif example.

It was my "faul", because i thought i have C3 rev1, when in fact i do have C3 rev2, but that is not changing the fact i cant build app.

xueyunfei998, 2021-11-22:
hi @chegewara 
If the problem is solved, you can close the problem. Thanks

chegewara, 2021-11-22:
I dont know if problem is solved, wifi app still does not build for C3 rev1.

xueyunfei998, 2021-11-22:
hi @chegewara 
I suggest you use it directly C3 rev3.
There may be some problems with C3 Rev1. C3 Rev3 has been fixed

chegewara, 2021-11-22:
i can do that, but what will do someone with C3 rev1?

I am not asking for suggestion, but reporting problem with build wifi app.

chegewara, 2021-11-22:
To make it easier to reproduce:
https://github.com/espressif/esp-idf/tree/master/examples/wifi/scan
esp-idf: App "scan" version: v5.0-dev-329-g4a011f3183
```
/home/chegewara/.espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/chegewara/esp/master/components/esp_phy/lib/esp32c3/libphy.a(phy_chip_v7.o): in function `phy_get_romfunc_addr':
(.text.phy_get_romfunc_addr+0x1a): undefined reference to `rom_temp_to_power'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
[sdkconfig.rev1.txt](https://github.com/espressif/esp-idf/files/7583904/sdkconfig.txt)
[sdkconfig.rev2.txt](https://github.com/espressif/esp-idf/files/7583923/sdkconfig.rev2.txt)

Just to make sure i tried to build for esp32C3 rev2 and also failed with the same error, so:
- severity: BUG
- importance: high 
- additional steps: better test case maybe??

xueyunfei998, 2021-11-23:
hi @chegewara 
I compile this example with that commit id you provide, esp-idf: App "scan" version: v5.0-dev-329-g4a011f3183

https://github.com/espressif/esp-idf/tree/master/examples/wifi/scan

My compilation here can pass.

You can follow the following steps:

idf.py set-target esp32c3

idf.py build

[image]

chegewara, 2021-11-23:
Yes, it pass for C3 rev3, which is default settings. Like i said, it fail for C3 rev1 and rev2.

0xjakob, 2021-11-24:
@chegewara confirmed, with **Chip revision 2** there's a linker error. I'll take a look at it.

0xjakob, 2021-11-24:
`rom_temp_to_power` is provided in the [revision 3 linker script](https://github.com/espressif/esp-idf/blob/master/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld#L30). But for some reason, it's not provided in the [revision <3 linker script](https://github.com/espressif/esp-idf/blob/master/components/esp_rom/esp32c3/ld/esp32c3.rom.ld#L1899). 

`rom_temp_to_power` is required by `libphy.a` but not provided on revision < 3, because of the linker script.
In revision 3, `rom_temp_to_power` is provided by the rom code.

I'll pass it to the relevant colleagues, since I'm not familiar with libphy.a.

chegewara, 2021-11-24:
Not a problem, as long as it can be solved.

Thanks

xueyunfei998, 2021-11-25:
hi @chegewara 

After our internal discussion, at present, rev1 rev2 and idf have stopped maintenance, so it is recommended to use rev3 directly.

AxelLin, 2021-11-25:
> hi @chegewara
> 
> After our internal discussion, at present, rev1 rev2 and idf have stopped maintenance, so it is recommended to use rev3 directly.

So you are asking people to throw away the rev1,rev2 devices and buy a new one with rev3?

chegewara, 2021-11-25:
Thanks a lot for letting us know we are not important to get fix, again, and by "us" i mean all people who were stupid enough to buy espressif boards earlier.
Now we have wifi featured boards without wifi support. 
Since its not an issue to espressif i am closing it.

chegewara, 2021-11-25:
PS in that case i am suggesting to remove minimum revision select option from menuconfig, to not make us confused.

TianaESP, 2021-11-25:
Hi @AxelLin @chegewara, sorry for letting you have such a bad experience on this. Please let me have a brief clarification. 
As far as I know, we sent rev1 and rev2 to customers mainly for pre-study and evaluation, even we know we will have further updates on silicons(like rev3) at that time. We value customers' feedback, so we want to share with you although they are “uncompleted” or "limited". However, we recommend customers build MP products with the rev3, which is better and shares the largest marketing. from my knowledge, rev3 is the only version we produce now. 
If you are using rev1 and rev2 for MP, of course, we will keep supporting you with your questions. But, if it is possible for you to switch to rev3, we would be grateful. 
Thanks for your suggestions wrt removing minimum revision select option from menuconfig. If you have any questions or concerns. Please feel free to ask.
Tiana

chegewara, 2021-11-25:
That sounds reasonable, but is not actually full story.
I bought my 5 C3 from mouser and i believe there are hundreds or even thousands C3 boards sold on mouser, maybe also digkey and/or amazon. Sure, i can switch to rev3 as i am "lucky" enough and i have 2 C3 as a sample from espressif, but i still have 5 rev2 which are crap now. 
I can live with that, because i am earning enough on esp32 programming to buy new boards, but this is bug in closed wifi driver and we dont even cant fix it in forked repo.

Jason2866, 2021-11-25:
@chegewara have you tried to uncomment in [file](https://github.com/espressif/esp-idf/blob/master/components/esp_rom/esp32c3/ld/esp32c3.rom.ld#L1899)

AxelLin, 2021-12-23:
> hi @chegewara
> 
> After our internal discussion, at present, rev1 rev2 and idf have stopped maintenance, so it is recommended to use rev3 directly.

@xueyunfei998 @0xjakob 

I'm surprised that I hit this issue on v4.3.2.
If you don't want to support older h/w rev in master tree, that's fine.
But this is a **breaking** change, AFAICT, it should not happen on stable branches.

denruss, 2021-12-28:
I have the same problem, already have a solution?

NHellFire, 2022-01-06:
> @chegewara have you tried to uncomment in [file](https://github.com/espressif/esp-idf/blob/master/components/esp_rom/esp32c3/ld/esp32c3.rom.ld#L1899) [27ddbf5#diff-56ec23fe68028a0ce77e0eb47e28216ebb0c9f87cd1fbffe1df4e1c21abdf5ab](https://github.com/espressif/esp-idf/commit/27ddbf5fccb6122238c84e333fa5f50d25b6dd89#diff-56ec23fe68028a0ce77e0eb47e28216ebb0c9f87cd1fbffe1df4e1c21abdf5ab) Looks like not an dramatic issue when back again. Btw. I have a effected board too, which was NOT sold as development board...

I can confirm that reverting the edits to esp32c3.rom.ld makes WiFi functional on my rev2 sample.

Seenov, 2022-01-10:
The suggestion of using rev 3 is only practical if Espressif identifies the revisions of the chips in the part numbers and replaces the rev2 that were sold to customers. I have 2 Dev kits that have rev 3 parts ordered at the same time as qty 78 ESP32-C3-WROOM-02-N4 that are revision 2 (unknown to me at the time). I have another full reel of 500 pieces that have the same part number.  The Espressif factory label on the reel does not seem to identify the revision number. There is no information on the revision of the chips on the Mouser website
[image]
.

StanzaW, 2022-01-12:
Hi Seeno, sorry. Please get in touch with our sales team sales@espressif.com. Our staff will help check the version of this whole reel and provide after-sales service, including changing them to v3 if they are v2.

Seenov, 2022-01-12:
Hi, this resolves the issue for the 500 pieces, but I have 2 other orders, one of 78 that are rev 2 and one of 178 that I will test shortly. Please keep the issue open. Thank you

StanzaW, 2022-01-13:
Hi Seenov, please get in touch with our sales team at sales@espressif.com and provide your contact address. Our staff will arrange rev 3 modules for you. There are a small amount of rev 2 modules on the market. We have decided not to provide further support for them.   Sorry.

Seenov, 2022-01-13:
Thanks, great customer service, I am impressed!

pic16f887, 2022-01-26:
With such "good" support for you products I am not sure that should use ESP32 C3 because my dev board is useless with up to date toolchain and usually sellers can't answer which revision of SoC is on the board.

Seenov, 2022-01-26:
Hello pic16f887, I think that reading the QRCode on the SoC, may produce info to determine the rev or the manufacture date. Ask Espressif about it. Here is a picture of the SoC on my rev 3 dev board. It was shipped from Mouser July 16th, 2021.
Good luck.
[image]
k

AxelLin, 2022-02-23:
https://docs.espressif.com/projects/espressif-esp-faq/en/latest/commercial-faq/index.html#how-can-i-tell-if-an-espressif-module-is-in-mass-production-or-an-npi-product

vikeyfox, 2022-03-18:
遇到了同样的问题，至少得给条活路吧，要不然买的几十片样片就报废了

Jason2866, 2022-03-18:
@vikeyfox as workaround you can use this [file](https://github.com/tasmota/esp-idf/blob/v4.4_dev/components/esp_rom/esp32c3/ld/esp32c3.rom.ld) and replace the orig. one.

vikeyfox, 2022-03-19:
> 

> @vikeyfox作为解决方法，您可以使用此[文件](https://github.com/tasmota/esp-idf/blob/v4.4_dev/components/esp_rom/esp32c3/ld/esp32c3.rom.ld)并替换原始文件。一。

OH,thankyou,I'll try

brianlinuxing, 2022-11-29:
I am very late to this, but I've got these errors with my Rev. 1 too.

My understanding is that is to fix this, in the short term, it is to adjust the esp32c3.rom.ld file?

Does anyone have a copy of the file that will work with ESP-IDF v4.4?

I tried following then link from @Jason2866 but it’s vanished, is there another one that is good?

Thanks in advance :)

Jason2866, 2022-11-29:
It is [here](https://github.com/tasmota/esp-idf/blob/Tasmota/v4.4/components/esp_rom/esp32c3/ld/esp32c3.rom.ld)

brianlinuxing, 2022-12-03:
Thank you so much @Jason2866 :)

0xjakob, 2023-02-28:
We are closing this issue now. If you happen to have this very problem (old ESP32-C3 HW revision that is not supported by current IDF version), please contact sales as indicated above.

closed 2023-02-28 by 0xjakob