espressif/esp-idf #7947: How do I set 1 nano second timer in esp32 ?

knucklehead-nilu, 2021-11-22:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

Development Kit: ESP32-Mini-1
Kit version
Module or chip used: ESP32
IDF version :v4.2.1
Build System: Make|CMake|idf.py
Compiler version : xtensa-esp32-elf-gcc (crosstool NCG esp-2020r3)8.4.0
Operating System: Windows
(Windows only) environment type: Command Prompt
Using an IDE?: Yes (Eclipse)

Power Supply: USB
## Problem Description

//Detailed problem description goes here.
I am trying to implement 1 nano second timer for my esp32 board.
This I am looking for, to check time taken by my function/handler.
 
So, could you please help me with this. 

Thanks!!

o-marshmallow, 2021-11-22:
Hi @enthunilu ,

The simplest way to get the time a function took to execute is to use `int64_t esp_timer_get_time(void)`. This returns the elapsed time, in microseconds, since boot. so you can do something like:
```
void app_main () {
    int64_t first = esp_timer_get_time();
    function_to_bench();
    int64_t second = esp_timer_get_time();
    /* Calculate elapsed time within function_to_bench */
    int64_t elapsed = second - first;
    printf("Elapsed: %lld\n", elapsed);
   [...]
}
```

However, I not sure what you mean by "1 nano second timer". `esp_timer_get_time()` returns microseconds, so even if this function takes less than 1us to execute, you won't have a precise nanosecond value.
Moreover, ESP32 CPU is running at 240MHz, which is 1 clock cycle every 4ns. It means that even the simplest instruction, `nop`, would take at least 4ns.

negativekelvin, 2021-11-22:
esp_cpu_get_ccount and esp_clk_cpu_freq

enthunilu, 2021-11-25:
Thank you so much @o-marshmallow and @negativekelvin for your quick response. 
For me both options will work. Thank you again!!

o-marshmallow, 2021-11-26:
You're welcome :+1:
Closing this issue.

closed 2021-11-26 by o-marshmallow