espressif/esp-idf #7951: a2dp example output only noise

wushifu32, 2021-11-23:
## Environment

- Development Kit:      [ESP32-DevKit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-270-gcaa9e323cc
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
I  run the a2dp example. And all I can here only noise.
I captured the D25 pin output waveform as following:
[image]

This seems the data write to I2S buffer is not match the internal DAC format

nopnop2002, 2021-11-25:
@wushifu32 

Yes.
this example sends random noise data, not music.
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/classic_bt/a2dp_source/main/main.c#L387

Try this.
https://github.com/nopnop2002/esp-idf-a2dp-source

wushifu32, 2021-11-25:
It's a2dp sink example. The music source is from my phone

BetterJincheng, 2021-11-26:
Hi @wushifu32,

It seems you have some misunderstandings about the a2dp_sink_demo. The sink just counts the data received from source and print the number, not actually play the audio.

Turn to us if any questions.
Thanks

wushifu32, 2021-11-26:
Hi @BetterJincheng ,

I see the sink demo call `write_ringbuf` to send the data received to task `bt_i2s_task_handler`. And in this task, the data is written to I2S buffer by calling `i2s_write`. In this demon, the I2S module is configured as internal DAC. So I think it can play the audio. Am I misunderstand the demo or the audio format not right?

wushifu32, 2021-11-26:
The codec capabilities of a2dp source from my phone is:
```
I (15297) BT_AV: A2DP audio stream configuration, codec type 0
I (15297) BT_AV: Configure audio player 21-15-2-35
```
According to [a2dp sepc](https://tsquarex.files.wordpress.com/2014/10/a2dp_spec_v12.pdf), the audio format is Joint Stereo, not DUAL CHANNEL.

BetterJincheng, 2021-11-29:
@wushifu32 Which demo do you refer? Please link here.

wushifu32, 2021-11-30:
@BetterJincheng 
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_sink

moefear85, 2022-01-23:
a2dp_sink_demo can infact play audio from a source, I just tested it. You can also use the internal DAC. But there is a simple error in esp-idf that you have to correct. See this:

[](https://www.esp32.com/viewtopic.php?t=6025)

wushifu32, 2022-06-05:
> a2dp_sink_demo can infact play audio from a source, I just tested it. You can also use the internal DAC. But there is a simple error in esp-idf that you have to correct. See this:
> 
> https://www.esp32.com/viewtopic.php?t=6025

This patch works

Alvin1Zhang, 2023-09-11:
Thanks for sharing the updates, sorry for the very slow turnaround, feel free to reopen.

closed 2023-09-11 by Alvin1Zhang