espressif/esp-idf #7952: esp32 s2 tiny usb driver lost complete interrupt  during concurrent IN & OUT  test

chuanjinpang, 2021-11-23:
**Environment** 
verion:esp-idf-4.3-beta3
EPS32 S2  as a usb device 
host : centos7
2. Describe your problem.
esp-idf-4.3-beta3\esp-idf-4.3-beta3\components\tinyusb\tinyusb\src\portable\espressif\esp32s2\dcd_esp32s2.c
when concurrent issue USB IN & OUT transcation in stress test. it seem lost complete interrupt  that cause USB can't work.
 if (USB0.out_ep_reg[n].doepint & USB_XFERCOMPL0_M) can ocurr in bad case

I suspect the esp-idf tiny usb have known limitation beacuse the IOT project use a Semphore for switch IN or OUT transfer.
refer:
esp-iot-solution\components\usb\tinyusb\additions\tusb\src\class\net\net_device.c
3.  my queston:
 is that it's a limitation for usb transfer? or it's a bug ?

chuanjinpang, 2021-11-25:
let me update status .
my diy project is a usb 802.11 wireless network card with esp32s2, 
the  root cause maybe the wifi EMI affect the USB transfer that cause esp32s2 usb transfer failed.  and the esp32s2 usb tinyusb seem doesn't have any error recovery flow, then the USB endpoint work abnormal. 
the USB recieve the data from host, then issue the data to wifi interface, then wifi start transmit, and the wifi also input data to host, then the EMI affect the USB transfer. 
I doesn't have the USB sniffer now, maybe 202201 got one . i will check the bus status .

tore-espressif, 2022-03-31:
Hello @chuanjinpang sorry for such a long delay.
this may be the same case as https://github.com/espressif/esp-idf/issues/7404 and https://github.com/espressif/esp-idf/issues/8241

Could you please try your code on latest `master` branch of esp-idf?

closed 2023-02-22 by Alvin1Zhang