espressif/esp-idf #7953: aligenie_demo无法同时调颜色和亮度

zsjlovestudy, 2021-11-23:
用天猫精灵APP控制这个设备的时候，先在固定亮度不变时，再改变颜色颜色都是正确的；亮度不变的时候，去调完颜色，再去调大亮度，灯的颜色就不是对应的颜色了。但是调小了以后颜色又是正确的了。

InfiniteYuan, 2021-11-24:
@zsjlovestudy  这是因为 天猫精灵这里用的是 HSL 色彩空间，当改变亮度的时候，实际修改是的 HSL 中的 L 分量，可以查看 HSL 色彩空间，当 L 为 100 时，颜色为白色。

如果这种不适合你的应用，你可以自定义修改。

[image]

[image]

zsjlovestudy, 2021-11-24:
大佬，能给点思路吗，我暂时没什么思路。
1.当亮度不变时，去改变颜色，亮度不能改变。
2.当颜色不变时，去调大亮度，不会变成白色。

zsjlovestudy, 2021-11-24:
> @zsjlovestudy 这是因为 天猫精灵这里用的是 HSL 色彩空间，当改变亮度的时候，实际修改是的 HSL 中的 L 分量，可以查看 HSL 色彩空间，当 L 为 100 时，颜色为白色。
> 
> 如果这种不适合你的应用，你可以自定义修改。
> 
> [image]
> 
> [image]

大佬，能给点思路吗，我暂时没什么思路。
1.当亮度不变时，去改变颜色，亮度不能改变。
2.当颜色不变时，去调大亮度，不会变成白色。

InfiniteYuan, 2021-11-25:
可以试试 限制 HSL 中的 L 分量 最大为 50。

具体要你根据色彩空间自己调的。

closed 2025-03-13 by Alvin1Zhang