espressif/esp-idf #7954: When ESP32 is used as STA, it cannot connect to another ESP32 which is used as AP:Disconnect reason : 201

YXW-123, 2021-11-23:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version : [v4]
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version : v4.3.1
- Build System:         [idf.py]
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
Using the code of /esp/esp-idf/examples/wifi/getting_started,and flash to 2 ESP32s,and run in two terminal windows of the virtual machine.But STA cann't connect to AP.My phone can find the ap and connect successfully.
### Expected Behavior
 Two ESP32s which used as STA and AP can connect to each other.
### Actual Behavior
E (13079) wifi station: Disconnect reason : 201
### Steps to reproduce

1. Copy the example code to ~/esp
2.cd ~/esp/softAP ,then idf.py build,then idf.py -p /dev/ttyUSB0 flash monitor
[image]
3.cd  ~/esp/station ,then idf.py build,then idf.py -p /dev/ttyUSB1 flash monitor
[image]

4.Let my mobilephone connect to AP,it succeeds.
[image]

### Code to reproduce this issue

```cpp
static void event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        esp_wifi_connect();
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        if (s_retry_num < EXAMPLE_ESP_MAXIMUM_RETRY) {
            esp_wifi_connect();
            s_retry_num++;
            ESP_LOGI(TAG, "retry to connect to the AP");
        } else {
            xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);
        }
        ESP_LOGI(TAG,"connect to the AP fail");
    } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI(TAG, "got ip:" IPSTR, IP2STR(&event->ip_info.ip));
        s_retry_num = 0;
        xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
    }
}

```

## Debug Logs
```
I (779) wifi:mode : sta (94:b9:7e:d4:9d:cc)
I (779) wifi:enable tsf
I (779) wifi station: wifi_init_sta finished.
E (2829) wifi station: Disconnect reason : 201
I (2829) wifi station: retry to connect to the AP
I (2829) wifi station: connect to the AP fail
E (4879) wifi station: Disconnect reason : 201
I (4879) wifi station: retry to connect to the AP
I (4879) wifi station: connect to the AP fail
E (6929) wifi station: Disconnect reason : 201
I (6929) wifi station: retry to connect to the AP
I (6929) wifi station: connect to the AP fail
E (8979) wifi station: Disconnect reason : 201
I (8979) wifi station: retry to connect to the AP
I (8979) wifi station: connect to the AP fail
E (11029) wifi station: Disconnect reason : 201
I (11029) wifi station: retry to connect to the AP
I (11029) wifi station: connect to the AP fail
E (13079) wifi station: Disconnect reason : 201
I (13079) wifi station: connect to the AP fail
I (13079) wifi station: Failed to connect to SSID:myssid, password:mypassword

```

nishanth-radja, 2021-11-25:
@YXW-123 I see that the AP is having credentials ssid : "myssid", password: "mypassword" where as the STA is trying to connect with ssid:"myssid",password: " " . Can you pls verify in the menuconfig on why is the password blank on the STA side.
Pls share the SDK config of both AP and STA, if it still fails to connect.

xueyunfei998, 2021-11-25:
hi @YXW-123 

**E (13079) wifi station: Disconnect reason : 201**

This 201 error is a wifi password error.

Please check whether the passwords of the next two devices are always the same.

AxelLin, 2021-11-25:
> hi @YXW-123
> 
> **E (13079) wifi station: Disconnect reason : 201**
> 
> This 201 error is a wifi password error.

@xueyunfei998 
How to look up the meaning of disconnect reason code?
If I check wifi_err_reason_t, 201 means WIFI_REASON_NO_AP_FOUND rather than "wifi password error".

xueyunfei998, 2021-11-25:
@AxelLin 

This error code can be found here:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html,

but there is no detailed explanation.

There will be 201 error in the following two cases：

1  The device carries the password to connect the ap without the password.

2  The device does not have a password to connect to the ap with the password.

YXW-123, 2021-11-25:
It's  my fault to forget I have  changed the password.But I find that the connection is unstable. Even both of the ssid and password  are right,it will often report error "falied to connect"and few successes.

xueyunfei998, 2021-11-25:
hi @nishanth-radja 

~/esp-idf/examples/wifi/getting_started/station

~/esp-idf/examples/wifi/getting_started/softAP

I tested it with the examples of sta and ap. I tested it 30 times, and every time it was successful.

You can retest with sta's example.

Make sure there is no ap with the same name (ssid :myssid) around,

The example of sta is wpa2 encryption by default, please set the password.

AxelLin, 2022-02-13:
> It's my fault to forget I have changed the password.Another possible reason is signal strength.I was not told to install the antenna due to the first use.I found that when the two ESPs are very close, the probability of successful connection becomes greater

@YXW-123 
Does that mean this is not an issue and can be closed?

YXW-123, 2022-03-01:
Yes!

closed 2022-03-01 by YXW-123