espressif/esp-idf #7955: v4.2.2: interaction between uart, i2c, gpio and twai driver leads to crash in esp_intr_alloc()

zevv, 2021-11-23:
## Environment

- Development Kit:     custom board
- Kit version (for WroverKit/PicoKit/DevKitC): -
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.2.2
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:  Linux, Debian testing
- Using an IDE?:  No
- Power Supply:  Battery

## Problem Description

The program below is the minimized problem that shows in our application when upgrading from v4.2 to v4.2.2: symptom is `twai_install_driver()` failing with `ESP_ERR_NOT_FOUND`:

### Expected Behavior

Above program works on both v4.2 and v4.2.2

### Actual Behavior

Program crashes with an `ESP_ERR_NOT_FOUND` on `twai_driver_install()`. see logs below

### Steps to reproduce

1. Install esp-idf v4.2
2. compile and flash the above program and notice no errors are reported
3. install esp-idf v4.2.2
4. compile and flash the above program and notice the reported errors

### Code to reproduce this issue

```cpp
#include <driver/gpio.h>                                                            
#include "driver/twai.h"                                              
#include "driver/uart.h"
#include "driver/i2c.h"            
                                
void app_main()                                  
{                                             
    twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT(GPIO_NUM_4, GPIO_NUM_5, TWAI_MODE_NORMAL);
    twai_timing_config_t t_config = TWAI_TIMING_CONFIG_500KBITS();
    twai_filter_config_t f_config = TWAI_FILTER_CONFIG_ACCEPT_ALL();
                          
    ESP_ERROR_CHECK(uart_driver_install(UART_NUM_0, 2048, 0, 0, NULL, 0));  
    ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0));  
    ESP_ERROR_CHECK(gpio_install_isr_service(0));  
    ESP_ERROR_CHECK(twai_driver_install(&g_config, &t_config, &f_config));
}          
```

## Debug Logs

```
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x400840a8
file: "/opt/toolchains/esp-idf/components/driver/twai.c" line 491
func: twai_driver_install
expression: esp_intr_alloc(ETS_TWAI_INTR_SOURCE, g_config->intr_flags, twai_intr_handler_main, NULL, &p_twai_obj->isr_handle)

abort() was called at PC 0x400840ab on core 0

Backtrace:0x40085237:0x3ffb4620 0x400858e1:0x3ffb4640 0x40088f06:0x3ffb4660 0x400840ab:0x3ffb46d0 0x400d6531:0x3ffb46f0 0x400d2fa2:0x3ffb4710 0x400d1b1e:0x3ffb4770 0x400858e9:0x3ffb47a0

```

zevv, 2021-11-23:
bisected this to commit a0bb9b2f57ccd8cec5d258a9a8bad0fef3dfefcf

negativekelvin, 2021-11-23:
https://github.com/espressif/esp-idf/issues/6709#issuecomment-801176126

Try installing the twai driver first

zevv, 2021-11-23:
Sure, that helps. Is that the recommended solution, or just a workaround for an underlying issue?

Dazza0, 2021-11-24:
@zevv  I think what's happening is that there literally aren't enough level 1 interrupts. `TWAI_GENERAL_CONFIG_DEFAULT()` will set `.intr_flags = ESP_INTR_FLAG_LEVEL1`, and since your example calls `twai_driver_install()` last, at that point there are no more available level 1 interrupts.

You can try using clearing `.intr_flags` or using a different level.

Dazza0, 2022-02-14:
@zevv This looks like an issue of actually running out of level 1 interrupts, and doesn't look like a TWAI driver. As suggested above, please try using a different interrupt level. I'm closing this issue, please feel free to reopen.

closed 2022-02-14 by Dazza0