espressif/esp-idf #7956: ASSERT_ERR(node->corrupt_check == KE_LIST_PATTERN), in ke_mem.c at line 275

Platin21, 2021-11-23:
## Environment

- Development Kit: Ublox/Nina-w106
- Kit version (for WroverKit/PicoKit/DevKitC): 106
- Module or chip used:  Ublox/Custom Board
- IDF version (run ``git describe --tags`` to find it):
   v3.4 with vTaskDelay Config Assert Disabled
- Build System:  CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    (crosstool-NG esp-2020r3) 8.4.0
- Operating System:    Linux
- Power Supply:  5V

## Problem Description

Crash in ke_mem on line 275

### Expected Behavior

To not crash?

### Actual Behavior

Crashing the system

### Steps to reproduce

No clear path

## Debug Logs

```
ASSERT_ERR(node->corrupt_check == KE_LIST_PATTERN), in ke_mem.c at line 275
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x4018ca7a  PS      : 0x00060f34  A0      : 0x800d4d3c  A1      : 0x3ffbcf70
A2      : 0x00000000  A3      : 0xb33fffff  A4      : 0x00000014  A5      : 0x000000a5
A6      : 0x000000a5  A7      : 0x00000001  A8      : 0x80113dd6  A9      : 0x3ffbcf50
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000014  A13     : 0x000000a5
A14     : 0x000000a5  A15     : 0x00000001  SAR     : 0x0000001e  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x4018ca77:0x3ffbcf700x400d4d39:0x3ffbcf90 0x400938eb:0x3ffbcfb0

Core  1 register dump:
PC      : 0x40089418  PS      : 0x00060434  A0      : 0x80019f39  A1      : 0x3ffe9210
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x3ffc21ec
A6      : 0x3ffc21ec  A7      : 0x4008933c  A8      : 0x80089418  A9      : 0x3ffe91f0
A10     : 0x0000004c  A11     : 0x0000004c  A12     : 0x00000010  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400893c0  LEND    : 0x400893c7  LCOUNT  : 0x00000000

Backtrace:0x40089415:0x3ffe92100x40019f36:0x3ffe9230 0x4001ab1f:0x3ffe9260 0x40046619:0x3ffe9280 0x4018064a:0x3ffe92b0 0x4017d916:0x3ffe9300 0x40019d11:0x3ffe9340 0x40055b4d:0x3ffe9360 0x40174b7f:0x3ffe9380 0x40175181:0x3ffe93a0

ELF file SHA256: 52e7b7e4e767838c
``` 
The backtrace is sadly not useful as it just returns missing addresses..

Platin21, 2021-11-23:
@Yulong-espressif  any comments? Do you guys need more infos?

negativekelvin, 2021-11-24:
I would guess it is in the low level Bluetooth stack failing a heap check due to memory corruption

Platin21, 2021-11-24:
@negativekelvin yes what i suspected too. Meanwhile that doesn't help at all.

negativekelvin, 2021-11-25:
You should state how often it happens, whether you are using stack/heap checking, overview of your app, what is the simplest version of your app that reproduces it, if you tried a newer branch, etc...

Platin21, 2021-11-25:
@negativekelvin if there would be a easy repro i would have added it here.
How often it happens once in two hours if no other assert gets triggert.

Platin21, 2021-11-25:
If it's really a heap corruption issue. Would be pretty bad as we can't modify the code which does these alloctions as it's inside on of the IDF Blobs.

negativekelvin, 2021-11-25:
If you are getting other asserts and you can't reproduce it with simple code it suggests your app is doing something to corrupt memory.  Even if the corrupt memory is inside the Bluetooth reserve pool it doesn't mean your app can't corrupt it somehow directly or indirectly.

Platin21, 2021-11-26:
@negativekelvin That is what my thinking was too but we switched everything to just use a single poll of memory the only exception to this is exactly the ble part.

xiewenxiang, 2021-12-31:
Hi @Platin21,

How can we reproduce this issue?  What is the probability of this problem repeating?

Is there a repetitive init/deint Bluetooth stack in your application? If so， It is recommended to stop broadcasting, stop scanning, and disconnect all connections before deinit the stack.

deepinno, 2022-01-14:
@xiewenxiang 
Hardware platform: esp32c3.
esp-idf version: v4.3
commit fbc8ad04177c6eaea9355f6e7b109e238fda60e1
///===========================
I find a similar bug these days that we use ble mesh for publish msg periodically.
It is strange that it run ok in our officer, but NG in another place.
I have updated your last libbtdm_app.a, but it not fixed.
//============================
The error infomation is below. Do you have any suggestions？

APP_MAIN: ---V002,system_free_size=196848---
APP_MAIN: ---V002,system_free_size=196848---
assert ke_mem.c 267, param 00000000 00000000
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
MEPC    : 0x40002d66  RA      : 0x40002d52  SP      : 0x3fca5e30  GP      : 0x3fc8fd64  
TP      : 0x3fc7701c  T0      : 0x4005890e  T1      : 0x00000000  T2      : 0x06040200  
S0/FP   : 0x00000000  S1      : 0x00000010  A0      : 0x0000002d  A1      : 0x0000000a  
A2      : 0x3fca5d80  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3fcdf9f8  
A6      : 0xa0000000  A7      : 0x0000000a  S2      : 0x00000000  S3      : 0x3fce0000  
S4      : 0x3fcdfdd0  S5      : 0x0000a55a  S6      : 0x0000000c  S7      : 0x00000000  
S8      : 0x3ff1b000  S9      : 0x00000001  S10     : 0x3fce0000  S11     : 0x0000001c  
T3      : 0x00000000  T4      : 0x26242220  T5      : 0x1e1c1a18  T6      : 0x16141210  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000018  MTVAL   : 0x00004398  
MHARTID : 0x00000000

deepinno, 2022-01-14:
@xiewenxiang 
My esp32c3 ble mesh app Crash about every two hours. And see above for error messages pls.

Platin21, 2022-01-14:
@deepinno the intial one?

Platin21, 2022-01-14:
> Hi @Platin21,
> 
> How can we reproduce this issue? What is the probability of this problem repeating?
> 
> Is there a repetitive init/deint Bluetooth stack in your application? If so， It is recommended to stop broadcasting, stop scanning, and disconnect all connections before deinit the stack.

It happens for us ever 6 hours. Hard to say how to replicate we use twai ble, wifi, and have a pretty high cpu usage by several other tasks without twai we can’t replicate it.

Platin21, 2022-01-14:
We have it not doing inti deinit repeatedly.

deepinno, 2022-01-15:
> 

Yes, it's the first time.
We have a set of Bluetooth mesh nodes that behave differently in two different places.
It can run stably in our R &amp; D office, but it will crash every few hours when we arrive at the customer site.
It seems to have something to do with the surrounding Bluetooth environment.

Platin21, 2022-01-15:
@deepinno seems to be a completely different error than the initial one though. 
So don’t want to be rude or something but might make sense to make a issue for the ble mesh stack separately? Or do i see that differently?

xenigma2019, 2022-04-13:
Our problem seems to have been solved. Solutions for reference:
Component config > Bluetooth > Bluetooth > Bluetooth controller>BLE adv report flow control number, set to 50

Component config > Bluetooth > Bluetooth > Bluetooth controller>Maximum number of devices in scan duplicate filter. set to 50

Component config > Bluetooth > Bluetooth > Bluetooth controller>BLE adv lost event threshold value. set to 80

Component config > Bluetooth > Bluetooth > Bluetooth controller>Maximum number of Mesh adv packets in scan duplicate filter. set to 50

ziaulhassan7575, 2022-05-19:
Did someone know, how to regenerate this issue ?

maenkai, 2024-03-19:
This bug appears randomly and I want to know how it is triggered so that the problem can be verified after the subsequent update version