espressif/esp-idf #7957: WiFi and tinyusb of ESP32-S3 not working at the same time

JnCrMx, 2021-11-24:
## Environment

- Development Kit: ESP32-S3-DevKitC-1-N8R8
- Module or chip used: ESP32-S3-WROOM-1
- IDF version: v5.0-dev-422-ga18cb70905
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Linux
- Using an IDE?: Yes, VSCodium, but issue also happens when using terminal only
- Power Supply: USB

## Problem Description
After adding WiFi to a project that uses tinyusb (in device mode), tinyusb stopped working. The host (my PC) is not able to communicate with the device.
My ``dmesg`` says:
```
[45283.773593] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[45295.965514] usb 1-1: USB disconnect, device number 6
[45296.444494] usb 1-1: new full-speed USB device number 7 using xhci_hcd
[45296.773461] usb 1-1: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[45296.773467] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[45296.773469] usb 1-1: Product: USB JTAG/serial debug unit
[45296.773471] usb 1-1: Manufacturer: Espressif
[45296.773472] usb 1-1: SerialNumber: 7C:DF:A1:E0:AA:2C
[45296.805117] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[45296.928997] usb 1-1: USB disconnect, device number 7
[45297.387834] usb 1-1: new full-speed USB device number 8 using xhci_hcd
[45297.544556] usb 1-1: device descriptor read/64, error -71
[45297.774550] usb 1-1: device descriptor read/64, error -71
[45298.011164] usb 1-1: new full-speed USB device number 9 using xhci_hcd
[45298.164552] usb 1-1: device descriptor read/64, error -71
[45298.404560] usb 1-1: device descriptor read/64, error -71
[45298.511201] usb usb1-port1: attempt power cycle
[45298.917834] usb 1-1: new full-speed USB device number 10 using xhci_hcd
[45298.944827] usb 1-1: Device not responding to setup address.
[45299.184853] usb 1-1: Device not responding to setup address.
[45299.397833] usb 1-1: device not accepting address 10, error -71
[45299.524502] usb 1-1: new full-speed USB device number 11 using xhci_hcd
[45299.551469] usb 1-1: Device not responding to setup address.
[45299.791498] usb 1-1: Device not responding to setup address.
[45300.004491] usb 1-1: device not accepting address 11, error -71
[45300.004566] usb usb1-port1: unable to enumerate USB device
[45306.991176] usb 1-1: new full-speed USB device number 12 using xhci_hcd
[45307.144545] usb 1-1: device descriptor read/64, error -71
[45307.374578] usb 1-1: device descriptor read/64, error -71
[45307.614504] usb 1-1: new full-speed USB device number 13 using xhci_hcd
[45307.767896] usb 1-1: device descriptor read/64, error -71
[45308.001243] usb 1-1: device descriptor read/64, error -71
[45308.111188] usb usb1-port1: attempt power cycle
[45308.521174] usb 1-1: new full-speed USB device number 14 using xhci_hcd
[45308.548520] usb 1-1: Device not responding to setup address.
[45308.784968] usb 1-1: Device not responding to setup address.
[45308.994507] usb 1-1: device not accepting address 14, error -71
[45309.117826] usb 1-1: new full-speed USB device number 15 using xhci_hcd
[45309.145010] usb 1-1: Device not responding to setup address.
[45309.384620] usb 1-1: Device not responding to setup address.
[45309.594498] usb 1-1: device not accepting address 15, error -71
[45309.594536] usb usb1-port1: unable to enumerate USB device
```
This behavior occurs only with the ``esp_wifi_start()`` call, all other WiFi-related calls seem to have no effect on tinyusb.
All log messages from tinyusb (on verbose) stop after calling ``esp_wifi_start()``.

### Expected Behavior
The host is able to communicate with the device.

### Actual Behavior
The device does not respond to the host.

### Steps to reproduce

I modified the "tusb_sample_descriptor" example, so it connect to a WiFi network after ``tinyusb_driver_install``.
See https://gist.github.com/JnCrMx/c63bff13016269986d867408f6266dee.
To reproduce, simply replace "tusb_sample_descriptor_main.c" in the example, flash it on an ESP32-S3, connect the USB port and watch the host's log (e.g. ``dmesg``) for an output (either the custom device descriptor on success or the above error messages).

## Debug Logs

``dmesg`` output for success (**without WiFi!**) and fail is also included in the Gist.
``idf.py monitor`` output (on *verbose*) is included in the Gist.
``espefuse.py summary`` output is included in the Gist.

## Additional Information
The pins of the DevKit board are all left unconnected. The only connections are the two USBs (one for UART and one for the actual USB connection).

MaxwellAlan, 2021-11-24:
Hi @JnCrMx , thanks for reporting this issue and we have fix this in internal esp-idf, please wait for the new commit sync to github esp-idf.

chegewara, 2021-11-24:
https://esp32.com/viewtopic.php?f=2&p=87641#p86456

igrr, 2021-11-25:
This should be fixed in https://github.com/espressif/esp-idf/commit/65f3bf4a8a21f6ef51649c3202ca282e9b04d124, could you please update and try again?

JnCrMx, 2021-11-25:
It works as expected now, thank you so much!

closed 2021-11-25 by JnCrMx