espressif/esp-idf #7959: Idle task constantly resetting watchdog kills interrupt performance

ztefanjo, 2021-11-24:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-S3
- IDF version: v4.4-dev-3401-gb86fe0c66c
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: No
- Power Supply:         USB
- Using Power Managment with DFS
- UART baudrate: 460800

## Problem Description
We get UART_FIFO_OVF events when sending more than 128 bytes of data to the ESP32. Setting uart_set_rx_full_threshold to a lower value (32) helps a bit but not completely. Debugging shows that the main problem is that the idle_hook_cb in task_wdt.c is constantly calling esp_task_wdt_reset which in turn is locking level 1-3 interrupts via portENTER_CRITICAL. By applying the [attached patch](https://github.com/espressif/esp-idf/files/7593830/4-0003-Hqv-relaxed-idle-task-watchdog-reset.patch.txt), the problem seems to be solved, also without changing the uart_set_rx_full_threshold. There are two suggested changes:
- Call esp_task_wdt_reset more seldom, in the [attached patch](https://github.com/espressif/esp-idf/files/7593830/4-0003-Hqv-relaxed-idle-task-watchdog-reset.patch.txt), it is called once at every OS tick.
- Lock critical section shorter time and only when necessary. In the [attached patch](https://github.com/espressif/esp-idf/files/7593830/4-0003-Hqv-relaxed-idle-task-watchdog-reset.patch.txt), it is also unlocked between find_task_in_twdt_list and reset_hw_timer to give interrupts execution time.

It can also be questioned if portENTER_CRITICAL is the best way to protect the watchdog list, or if a mutex could be used instead (still allowing interrupts and other tasks to execute).

The same issue has been encountered also by others, see e.g. https://esp32.com/viewtopic.php?t=14078.

### Expected Behavior
There should not be any problem to receive on UART, especially not when the rest of the system is idle.

### Actual Behavior
UART FIFO overflow

### Steps to reproduce

1. send data packages that are larger than 128 bytes to the ESP32, I used 220 byte in my test.

igrr, 2021-11-25:
Hi @ztefanjo, thank you for your analysis. 

Normally `esp_vApplicationIdleHook` function which dispatches the idle hooks should be calling `esp_pm_impl_waiti`, which in turn should put the CPU into a "wait for interrupt" state or enter light sleep mode. If instead the idle hooks get called over and over again, this probably indicates a bug in esp_pm_impl_waiti.

You also mention ESP32 in the description of the issue, but at the same time say `Module or chip used: ESP32-S3`. Which chip are you actually using?

Are there any tasks running on the APP CPU? Perhaps the situation you are seeing could be reproduced with some simple code example; if so, could you share it with us?

Finally, could you also attach your sdkconfig file, please? This will help us reproduce the issue.

ztefanjo, 2021-11-25:
Hi @igrr,
We have CONFIG_FREERTOS_USE_TICKLESS_IDLE enabled as we are using Power Management.
I tested setting a breakpoint at cpu_hal_waiti, but I never get there, because s_skipped_light_sleep is always true. s_mode pends between CPU_MAX and APB_MIN but never reaches LIGHT_SLEEP, probably because I have a light sleep lock active (to keep the UART responsive). Therefore esp_pm_impl_waiti always returns directly in my case. 

I am actually testing both on ESP32 and ESP32-S3, the problem seems to exist for both of them. Attached is our [sdkconfig](https://github.com/espressif/esp-idf/files/7602711/sdkconfig.zip) for ESP32-S3.

esp-lis, 2021-11-26:
> Hi @igrr, We have CONFIG_FREERTOS_USE_TICKLESS_IDLE enabled as we are using Power Management. I tested setting a breakpoint at cpu_hal_waiti, but I never get there, because s_skipped_light_sleep is [...]

[image]

I think your CPU has entered waiti mode, **CONFIG_FREERTOS_USE_TICKLESS_IDLE** is enabled and `s_skipped_light_sleep` is always true.

ztefanjo, 2021-11-26:
> I think your CPU has entered waiti mode, **CONFIG_FREERTOS_USE_TICKLESS_IDLE** is enabled and `s_skipped_light_sleep` is always true.
> 

You are quite right, I must have encountered some sort of compiler optimization when trying to set my breakpoint at `cpu_hal_waiti()`. It is called. Or at least, it is called today, see below....

> Whether to disable the task watchdog (`CONFIG_ESP_TASK_WDT`) can solve your problem? It's only a test.

Unfortunately, I am not able to reproduce the UART problem today. When writing the bug, it was very reproducible without my patch and not reproducible with my patch. Perhaps differences in the number of BLE/Wifi devices in the vicinity is the explanation, causing different load on the CPU:s?!?

I did however measure the number of times `esp_task_wdt_reset` is called using the code attached below. As my added code also takes some time to execute, it can be seen that `esp_task_wdt_reset` is called more than 562323 - 1026666 times every second (i.e. it is called more or less every microsecond). This cannot be good.

```
static bool idle_hook_cb(void)
{
    static TickType_t lastTick = 0;
    static unsigned int counter  = 0;
    static unsigned int max      = 0;
    static unsigned int min      = 0xFFFFFFFF;
    ++counter;
    TickType_t newTick =  xTaskGetTickCount();
    if ( ( newTick - lastTick ) > 1000 )
    {
        lastTick = newTick;
        if ( counter > max )
        {
            max = counter;
        }
        if ( counter < min )
        {
            min = counter;
        }
        counter = 0;
    }
    esp_task_wdt_reset();
    return true;
}
```

ztefanjo, 2021-11-30:
Must have closed this by mistake. The problem is not solved.

igrr, 2021-11-30:
I think the underlying issue is that `s_skipped_light_sleep` is always true — this isn't the expected behavior, and it is probably the reason why the idle hook is called so often. I ran a simpler test application based on wifi/power_save example, but didn't manage to reproduce this behavior. Do you think you could find a way to reduce your application to something which you could share, and which would still reproduce the issue?

ztefanjo, 2022-01-25:
After updating to latest 4.4 branch, we run into some problems with the patch previously attached. With the latest changes, one of the CPU:s seem to be more busy than the other one, leading to that the watchdog was not always updated with the previous patch. The new patch attached here solves that problem:
[4-0003-Hqv-relaxed-idle-task-watchdog-reset-v2.patch.txt](https://github.com/espressif/esp-idf/files/7933421/4-0003-Hqv-relaxed-idle-task-watchdog-reset-v2.patch.txt).

xiaojie76, 2022-04-11:
@ztefanjo Thank you for your proposal, we will consider it carefully. But this doesn't seem to be a perfect solution, it just reduces the probability of killing interrupts. Why not try to use a higher interrupt priority (>3) , so it won't be affected by portENTER_CRITICAL.

ztefanjo, 2022-04-11:
Hi @xiaojie76 , We are already using the highest allowed interrupt priority for our UART, which is ESP_INTR_FLAG_LEVEL3. Trying an even higher priority will lead to an error in the `esp_intr_alloc_intrstatus` function:
`if ((flags&ESP_INTR_FLAG_HIGH) && (handler)) return ESP_ERR_INVALID_ARG;` 

According to the comment in esp_intr_alloc.h, higher level interrupts need to be handled in assembly, but we want of course to use the UART driver provided by the IDF.

As you say, ESP_INTR_FLAG_LEVEL3 is also affected by portENTER_CRITICAL as it seems.

xiaojie76, 2022-04-11:
@ztefanjo Here is an example of using level 5 interrupts, you can request high-priority interrupts and modify highint5_c_handler to suit your application scenario.
[esp32s3_highint5_example.zip](https://github.com/espressif/esp-idf/files/8461399/esp32s3_highint5_example.zip)
.

ztefanjo, 2022-04-11:
@xiaojie76 : Is this really safe? There must be a reason that someone introduced the rule that priorities > 3 must not be implemented in C. If it is safe, I propose that you remove the limitation from `esp_intr_alloc_intrstatus` so that high priority interrupts to C are allowed.

igrr, 2022-04-11:
Using C interrupt handler functions with high level interrupts is indeed not possible. However in the example which @xiaojie76 has shared, the interrupt handler function is written in assembly. It does some work to prepare for C code execution (save the registers, set up the stack, and so on). You can find it in `highint5.S`. With that in place, you can have a high-level interrupt handler body implemented in C. Note that you still can't call FreeRTOS functions (such xQueueSendFromISR) in high-level interrupt handlers, so the usefulness of that interrupt handler is somewhat limited.

ztefanjo, 2022-04-11:
OK, as we want to use the existing IDF UART driver and that uses FreeRTOS features, I guess this is not an option then. Even if we wrote an own handler, we would need to have some sort of synchronization to signal e.g. when there is new data that has arrived.

xiaojie76, 2022-04-11:
@igrr Do you think the patch provided by @ztefanjo can be adopted? Or do you have any better suggestion?

xiaojie76, 2022-04-13:
`	esp_vApplicationIdleHook
                idle_hook_cb
		    esp_task_wdt_reset	
		esp_pm_impl_waiti
		    if (s_skipped_light_sleep[core_id]) {
		        cpu_hal_waiti();
		    }
`
@ztefanjo You mentioned CONFIG_FREERTOS_USE_TICKLESS_IDLE is enabled and s_skipped_light_sleep is always true.
So the CPU will enter the "wait for interrupt" state. But now continues to call esp_task_wdt_reset, is it because there is a continuous and frequent interrupt that causes the CPU to exit the waiti state? In addition to UART_RXFIFO_FULL_INT, are there any frequent interrupts in your application?

ztefanjo, 2022-04-13:
@xiaojie76 : This is a good question. cpu_hal_waiti is never blocking for very long times if I recall correctly and it is hard to know why it returns, as the source code is hidden.
Is there a way to tell what made the function stopping to wait? Except for the UART, we use a few GPIO:s but they should be very stable. We also use both BLE and Wifi, perhaps they are generating frequent interrupts?
Could the debugger itself cause interrupts?

xiaojie76, 2022-04-13:
@ztefanjo  I don't have a good way to see what interrupt happen, like you said, maybe from WIFI or BLE.  This also doesn't seem to be the direction to solve the problem.

jack0c, 2022-04-14:
@ztefanjo  uart_set_rx_full_threshold = 32, baud rate = 460800,  it takes 560us per interrupt, 128 is the FIFO depth of UART hardware,  the hardware can keep receiving data without software process for about 1.68ms(560us*3). IMO, there may be some code disable interrupt for long:

1. ESP32 have a live lock bug in cache, which is possible to disable interrupt for a while and use watch dog to save the CPUs
2. Some functions from heap may disable interrupt for long, for instance, realloc will allocate a new buffer and copy the original value into it in the enter critical context.

Can you confirm 2 things:
1. does ESP32S3 has the same issue
2. does lower baud rate help

xiaojie76, 2022-04-18:
@ztefanjo, the execution time of `esp_task_wdt_reset` is about 300 cpu cycles (about 4us according to your configuration cpu_freq 80MHz). But in your statistics `esp_task_wdt_reset `is called more than 562323 - 1026666 times every second , which is obviously impossible and seems to be an order of magnitude higher. Can you recount this parameter (need to distinguish the number of calls from core0 and core1)?

ztefanjo, 2022-04-19:
Hi,
I have re-measured how often esp_task_wdt_reset is called from idle_hook_cb. Now on ESP32S3 and for each CPU with the following idle_hook_cb function:
```
    static TickType_t lastTick[2] = { 0 };
    static unsigned int counter[2]  = { 0 };
    static unsigned int max[2]      = { 0 };
    static unsigned int min[2]      = { 0xFFFFFFFF, 0xFFFFFFFF };
    const uint32_t    core          = xPortGetCoreID();
    ++counter[ core ];
    TickType_t newTick =  xTaskGetTickCount();
    if ( ( newTick - lastTick[core] ) > 1000 )
    {
        lastTick[core] = newTick;
        if ( counter[core] > max[core] )
        {
            max[core] = counter[core];
        }
        if ( counter[core] < min[core] )
        {
            min[core] = counter[core];
        }
        counter[core] = 0;
    }
    esp_task_wdt_reset();
    return true;
```

max[0] is 186304 and max[1] is 184936 in my test-run. It is far from 562323 - 1026666 that I measured last time on ESP32, but still very high.

Yes, we are using portMAX_DELAY as timeout to UART xQueueReceive, but there are many other things that we also run and that may prevent xExpectedIdleTime to become large enough for sleep.

Re-attaching our [sdkconfig.defaults](https://github.com/espressif/esp-idf/files/8512541/sdkconfig.defaults.txt), as things have changed a bit since I first created this issue.

xiaojie76, 2022-04-19:
@ztefanjo Ok, I guess the problem is here, when`(xExpectedIdleTime < configEXPECTED_IDLE_TIME_BEFORE_SLEEP) `, s_skipped_light_sleep should be true but now false, and `cpu_hal_waiti `should be executed, but in fact neither enter light sleep nor wait for interrupt state.
You can temporarily change the code,
```
if( xExpectedIdleTime >= configEXPECTED_IDLE_TIME_BEFORE_SLEEP )
{
    ...
}else{
    ...
    s_skipped_light_sleep[core_id]=true; //Temporarily remove the static attribute
}
```
Just as a test to see if that solves your problem first.

ztefanjo, 2022-04-19:
@xiaojie76 : I tested the change you suggested (making s_skipped_light_sleep non-static and setting it to true in portTASK_FUNCTION if xExpectedIdleTime is < configEXPECTED_IDLE_TIME_BEFORE_SLEEP). This results in that idle_hook_cb is called max 2433 times for CPU 0 and 2012 times for CPU 1 per second, so this solution also seems to work well :-). It do makes sense to consider light sleep to be skipped also in this case.

xiaojie76, 2022-04-19:
@ztefanjo Thanks a lot for your feedback, this is indeed a bug in the esp_pm_impl_waiti implementation. I'll submit a fix on esp-idf soon.

xiaojie76, 2022-04-21:
@ztefanjo  You can use this patch [bugfix_esp_pm_impl_waiti_v4.4.patch.txt](https://github.com/espressif/esp-idf/files/8526735/bugfix_esp_pm_impl_waiti_v4.4.patch.txt),  just modify esp_pm_impl_waiti().  This modification will be added in the next version,  please close this issue if  no problem,.

Alvin1Zhang, 2022-05-16:
Thanks for reporting and sharing the updates, fix on master branch is available https://github.com/espressif/esp-idf/commit/1b81efc320bb6d73fe8ad513ab8367c168526f6f and fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/dcc77b0bed232cf549b48cb4e81528e894edbb72, feel free to reopen if the issue still happens, thanks.

closed 2022-05-16 by Alvin1Zhang

AxelLin, 2022-05-16:
@Alvin1Zhang 

Could you check if v4.3 also needs this fix? (The fix is in pm_impl.c, I think the impact is not just esp32s3).
PS. The patch can be cleanly apply to v4.3.

Alvin1Zhang, 2022-05-16:
> @Alvin1Zhang @xiaojie76
> 
> Could you check if v4.3 also needs this fix? (The fix is in pm_impl.c, I think the impact is not just esp32s3). PS. The patch can be cleanly apply to v4.3.

@AxelLin Thanks for the note, we are back porting the fix to release/4.3 now, currently under inter reviewing. Will update once the fix on release/4.3 is available, thanks.

AxelLin, 2022-05-28:
v4.3 fix: 30013ca66f51a68a89fa4db4d5030d47e24eaf7b

Alvin1Zhang, 2022-05-30:
@AxelLin  Thanks for the notes, this fix will be included in the forth coming v4.3.3 release, thanks.